coolsnowwolf/lede #2311: trojan以及UnblockNeteaseMusic下载失败

qd201211, 2019-12-06:
make download下，trojan-1.13.0-842ad5bb07eb8bce035fb274571e586629a97c99.tar.gz以及UnblockNeteaseMusic-0.22.1.tar.gz两个文件一直显示下载错误。换过上网方法重试问题依旧
trojan下载报错如下，网易云报错基本一致

trojan-1.13.0-842ad5bb07eb8bce035fb274571e586629a97c99.tar.gz: Download from https://github.com/trojan-gfw/trojan.git failed
trojan-1.13.0-842ad5bb07eb8bce035fb274571e586629a97c99.tar.gz: Requires sha256sum for verification
Checking out files from the git repository...
Cloning into 'trojan-1.13.0'...
remote: Enumerating objects: 2365, done.
remote: Total 2365 (delta 0), reused 0 (delta 0), pack-reused 2365
Receiving objects: 100% (2365/2365), 833.69 KiB | 540.00 KiB/s, done.
Resolving deltas: 100% (1774/1774), done.
Checking connectivity... done.
Note: checking out '842ad5bb07eb8bce035fb274571e586629a97c99'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by performing another checkout.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -b with the checkout command again. Example:

  git checkout -b new_branch_name

HEAD is now at 842ad5b... Update contributors
Packing checkout...
tar: unrecognized option '--sort=name'
Try 'tar --help' or 'tar --usage' for more information.

EnnawYang, 2019-12-06:
手动下载或者换梯子 我的没问题

qd201211, 2019-12-06:
> *

> 手动下载或者换梯子 我的没问题

我今天凌晨三点下载的也没这个问题，但是今天下午重新git clone之后就不行了，，我也不知道为什么，梯子换过了也一样

qd201211, 2019-12-06:
编译没问题，单独make download不行，，，神奇。现编译成功

closed 2019-12-07 by coolsnowwolf