coolsnowwolf/lede #2320: 关于Trojan

chchia, 2019-12-07:
服务器和windows客户端都设置后了，能使用没问题。

但是把相同配置输入ssr+里的trojan，死活启动不了，不知道有没有配置教程？

chchia, 2019-12-07:
https://github.com/coolsnowwolf/lede/commit/b0bca4dce08285142d1888e31a5e16f53ba20767

这个可以了，就是 游戏模式UDP中继 未运行

chchia, 2019-12-07:
发现新问题，每个几十秒trojan自动从启一次

[2019-12-08 01:44:40] [FATAL] exiting. . . 
[2019-12-08 01:44:40] [FATAL] fatal: bind: Address in use
Welcome to trojan 1.13.0
2019-12-08 01:44:39 Welcome to trojan 1.13.0 Started!
Welcome to trojan 1.13.0
[2019-12-08 01:43:57] [FATAL] exiting. . . 
[2019-12-08 01:43:57] [FATAL] fatal: bind: Address in use
Welcome to trojan 1.13.0
2019-12-08 01:43:56 Welcome to trojan 1.13.0 Started!
Welcome to trojan 1.13.0
[2019-12-08 01:43:14] [FATAL] exiting. . . 
[2019-12-08 01:43:14] [FATAL] fatal: bind: Address in use
Welcome to trojan 1.13.0
2019-12-08 01:43:13 Welcome to trojan 1.13.0 Started!
Welcome to trojan 1.13.0
[2019-12-08 01:42:31] [FATAL] exiting. . . 
[2019-12-08 01:42:31] [FATAL] fatal: bind: Address in use

coolsnowwolf, 2019-12-08:
Trojan 本身并不支持 UDP

closed 2019-12-08 by coolsnowwolf