coolsnowwolf/lede #2324: Github Actions编译中失败

jeremy125, 2019-12-07:
执行到Makedownload阶段时提示失败，日志如下 ：
```
Make download14s
##[error]Process completed with exit code 2.
.config:60:warning: unexpected data:  CONFIG_PACKAGE_luci-app-v2ray-server=y
.config:61:warning: unexpected data:  CONFIG_PACKAGE_luci-app-verysync=y
.config:62:warning: unexpected data:  CONFIG_PACKAGE_luci-app-webadmin=y
.config:63:warning: unexpected data:  CONFIG_PACKAGE_luci-app-wireguard=y
.config:64:warning: unexpected data:  CONFIG_PACKAGE_luci-app-wrtbwmon=y
.config:65:warning: unexpected data:  EOF
.config:68:warning: unexpected data:  cat >> .config <<EOF
.config:69:warning: unexpected data:  CONFIG_PACKAGE_luci-theme-argon=y
.config:70:warning: unexpected data:  CONFIG_PACKAGE_luci-theme-netgear=y
.config:71:warning: unexpected data:  EOF
.config:74:warning: unexpected data:  cat >> .config <<EOF
.config:75:warning: unexpected data:  CONFIG_PACKAGE_curl=y
.config:76:warning: unexpected data:  CONFIG_PACKAGE_htop=y
.config:77:warning: unexpected data:  CONFIG_PACKAGE_nano=y
.config:78:warning: unexpected data:  CONFIG_PACKAGE_screen=y
.config:79:warning: unexpected data:  CONFIG_PACKAGE_tree=y
.config:80:warning: unexpected data:  CONFIG_PACKAGE_vim-fuller=y
.config:81:warning: unexpected data:  CONFIG_PACKAGE_wget=y
.config:82:warning: unexpected data:  EOF
.config:94:warning: unexpected data: sed -i 's/^[ \t]*//g' ./.config
.config:95:warning: unexpected data: make defconfig
Error opening terminal: unknown.
/home/runner/work/lede/lede/include/toplevel.mk:136: recipe for target 'menuconfig' failed
make[1]: Leaving directory '/home/runner/work/lede/lede'
/home/runner/work/lede/lede/include/toplevel.mk:97: recipe for target '.config' failed
make[1]: *** [menuconfig] Error 1
make: *** [.config] Error 2
##[error]Process completed with exit code 2.
```

openwrt-ci.yml 脚本如下：

#
# This is free software, lisence use MIT.
# 
# Copyright (C) 2019 P3TERX <https://p3terx.com>
# Copyright (C) 2019 KFERMercer <KFER.Mercer@gmail.com>
# 
# <https://github.com/KFERMercer/OpenWrt-CI>
# 

name: OpenWrt-CI

on:
  schedule:
    - cron: 0 20 * * *
  push:
    branches: 
      - master

jobs:
  build:
    runs-on: ubuntu-latest

    steps:
      - name: Checkout
        uses: actions/checkout@master
        with:
          ref: master

      - name: Initialization environment
        env:
          DEBIAN_FRONTEND: noninteractive
        run: |
          docker rmi `docker images -q`
          echo "Deleting files, please wait ..."
          sudo rm -rf \
            /usr/share/dotnet \
            /etc/mysql \
            /etc/php
          sudo -E apt-get -y purge \
            azure-cli \
            ghc* \
            zulu* \
            hhvm \
            llvm* \
            firefox \
            google* \
            dotnet* \
            powershell \
            openjdk* \
            mysql* \
            php*
          sudo -E apt-get update
          sudo -E apt-get -y install build-essential asciidoc binutils bzip2 gawk gettext git libncurses5-dev libz-dev patch unzip zlib1g-dev lib32gcc1 libc6-dev-i386 subversion flex uglifyjs git-core gcc-multilib p7zip p7zip-full msmtp libssl-dev texinfo libglib2.0-dev xmlto qemu-utils upx libelf-dev autoconf automake libtool autopoint device-tree-compiler
          sudo -E apt-get -y autoremove --purge
          sudo -E apt-get clean

      - name: Update feeds
        run: |
          ./scripts/feeds update -a
          ./scripts/feeds install -a

      - name: Costom configure file
        run: |
          rm -f ./.config*
          touch ./.config

          #
          # ========================固件定制部分========================
          # 

          # 
          # 如果不对本区块做出任何编辑, 则生成默认配置固件. 
          # 

          # 以下为定制化固件选项和说明:
          #

          #
          # 有些插件/选项是默认开启的, 如果想要关闭, 请参照以下示例进行编写:
          # 
          #          =========================================
          #         |  # 取消编译VMware镜像:                   |
          #         |  cat >> .config <<EOF                   |
          #         |  # CONFIG_VMDK_IMAGES is not set        |
          #         |  EOF                                    |
          #          =========================================
          #

          # 
          # 以下是一些提前准备好的一些插件选项.
          # 直接取消注释相应代码块即可应用. 不要取消注释代码块上的汉字说明.
          # 如果不需要代码块里的某一项配置, 只需要删除相应行.
          #
          # 如果需要其他插件, 请按照示例自行添加.
          # 注意, 只需添加依赖链顶端的包. 如果你需要插件 A, 同时 A 依赖 B, 即只需要添加 A.
          # 
          # 无论你想要对固件进行怎样的定制, 都需要且只需要修改 EOF 回环内的内容.
          # 

          # 编译x64固件:
           cat >> .config <<EOF
           CONFIG_TARGET_x86=y
           CONFIG_TARGET_x86_64=y
           CONFIG_TARGET_x86_64_Generic=y
           EOF

          # 固件压缩:
           cat >> .config <<EOF
           CONFIG_TARGET_IMAGES_GZIP=y
           EOF

          # 编译UEFI固件:
           cat >> .config <<EOF
           CONFIG_EFI_IMAGES=y
           EOF

          # IPv6支持:
           cat >> .config <<EOF
           CONFIG_PACKAGE_dnsmasq_full_dhcpv6=y
           CONFIG_PACKAGE_ipv6helper=y
           EOF

          # 多文件系统支持:
           cat >> .config <<EOF
           CONFIG_PACKAGE_kmod-fs-nfs=y
           CONFIG_PACKAGE_kmod-fs-nfs-common=y
           CONFIG_PACKAGE_kmod-fs-nfs-v3=y
           CONFIG_PACKAGE_kmod-fs-nfs-v4=y
           CONFIG_PACKAGE_kmod-fs-ntfs=y
           CONFIG_PACKAGE_kmod-fs-squashfs=y
           EOF

          # USB3.0支持:
           cat >> .config <<EOF
           CONFIG_PACKAGE_kmod-usb-ohci=y
           CONFIG_PACKAGE_kmod-usb-ohci-pci=y
           CONFIG_PACKAGE_kmod-usb2=y
           CONFIG_PACKAGE_kmod-usb2-pci=y
           CONFIG_PACKAGE_kmod-usb3=y
           EOF

          # 常用LuCI插件选择:
           cat >> .config <<EOF
           CONFIG_PACKAGE_luci-app-adbyby-plus=y
           CONFIG_PACKAGE_luci-app-aria2=y
           CONFIG_PACKAGE_luci-app-baidupcs-web=y
           CONFIG_PACKAGE_luci-app-docker=y
           CONFIG_PACKAGE_luci-app-frpc=y
           CONFIG_PACKAGE_luci-app-hd-idle=y
           CONFIG_PACKAGE_luci-app-kodexplorer=y
           CONFIG_PACKAGE_luci-app-minidlna=y
           CONFIG_PACKAGE_luci-app-openvpn=y
           CONFIG_PACKAGE_luci-app-openvpn-server=y
           CONFIG_PACKAGE_luci-app-qbittorrent=y
           CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_Kcptun=y
           CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_Shadowsocks=y
           CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_ShadowsocksR_Server=y
           CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_ShadowsocksR_Socks=y
           CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_V2ray=y
           CONFIG_PACKAGE_luci-app-ttyd=y
           CONFIG_PACKAGE_luci-app-v2ray-server=y
           CONFIG_PACKAGE_luci-app-verysync=y
           CONFIG_PACKAGE_luci-app-webadmin=y
           CONFIG_PACKAGE_luci-app-wireguard=y
           CONFIG_PACKAGE_luci-app-wrtbwmon=y
           EOF

          # LuCI主题:
           cat >> .config <<EOF
           CONFIG_PACKAGE_luci-theme-argon=y
           CONFIG_PACKAGE_luci-theme-netgear=y
           EOF

          # 常用软件包:
           cat >> .config <<EOF
           CONFIG_PACKAGE_curl=y
           CONFIG_PACKAGE_htop=y
           CONFIG_PACKAGE_nano=y
           CONFIG_PACKAGE_screen=y
           CONFIG_PACKAGE_tree=y
           CONFIG_PACKAGE_vim-fuller=y
           CONFIG_PACKAGE_wget=y
           EOF

          # 取消编译VMware镜像以及镜像填充 (不要删除被缩进的注释符号):
          # cat >> .config <<EOF
          # # CONFIG_TARGET_IMAGES_PAD is not set
          # # CONFIG_VMDK_IMAGES is not set
          # EOF

          # 
          # ========================固件定制部分结束========================
          # 

          sed -i 's/^[ \t]*//g' ./.config
          make defconfig

      - name: Make download
        run: |
          make download -j8
          find dl -size -1024c -exec ls -l {} \;
          find dl -size -1024c -exec rm -f {} \;

      - name: Compile firmware
        run: |
          echo -e "$(nproc) thread build."
          make -j$(nproc) V=s

      - name: Assemble artifact
        run: |
          rm -rf ./artifact/
          mkdir -p ./artifact/
          find ./bin/targets/ -name "*combined*img*" | xargs -i mv -f {} ./artifact/
          find ./bin/targets/ -name "*sysupgrade*bin*" | xargs -i mv -f {} ./artifact/

      - name: Upload artifact
        uses: actions/upload-artifact@master
        with:
          name: OpenWrt firmware
          path: ./artifact/

不知道原因是什么，请帮忙看下，谢谢！

M1A6I3L, 2019-12-07:
少删了空格，删除#后的首字母要在原来#的位置，参考作者的示例。

# 编译x64固件:
cat >> .config <<EOF
CONFIG_TARGET_x86=y
CONFIG_TARGET_x86_64=y
CONFIG_TARGET_x86_64_Generic=y
EOF

https://github.com/KFERMercer/OpenWrt-by-lean/blob/CI-demo/.github/workflows/openwrt-ci.yml

jeremy125, 2019-12-07:
的确如楼上朋友所说，脚本文本中需要对齐，现在正常编译中了，谢谢！

closed 2019-12-07 by jeremy125

baokeding, 2020-06-04:
OpenWrt-CI / Build OpenWrt and release
failed 8 hours ago in 4h 5m 3s
Search logs
  Set up job4s
  Checkout6s
  Space cleanup10m 42s
  自定义插件主题。7s
  Update feeds37s
  Costom configuration file13s
  Make download3m 41s
  Compile firmware3h 49m 32s
##[error]Process completed with exit code 2.
Collected errors:
Configuring luci-i18n-mwan3helper-zh-cn.
Configuring libustream-openssl20150806.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-app-samba4:
Configuring kmod-button-hotplug.
Configuring kmod-fs-squashfs.
Configuring luci-app-docker.
 * 	samba4-libs
Configuring kmod-igb.
 * opkg_install_cmd: Cannot install package luci-app-samba4.
Configuring kmod-bnx2.
 * check_conflicts_for: The following packages conflict with samba4-server:
Configuring urngd.
 * check_conflicts_for: 	samba36-server * 
Configuring luci-i18n-firewall-zh-cn.
 * opkg_install_cmd: Cannot install package samba4-server.
Configuring luci-i18n-docker-zh-cn.
make[2]: *** [package/install] Error 255
Configuring ppp-mod-pppoe.
Configuring luci-i18n-nlbwmon-zh-cn.
package/Makefile:65: recipe for target 'package/install' failed
make[2]: Leaving directory '/home/runner/work/lede/lede'
make[1]: *** [/home/runner/work/lede/lede/staging_dir/target-x86_64_musl/stamp/.package_install] Error 2
package/Makefile:108: recipe for target '/home/runner/work/lede/lede/staging_dir/target-x86_64_musl/stamp/.package_install' failed
make[1]: Leaving directory '/home/runner/work/lede/lede'
/home/runner/work/lede/lede/include/toplevel.mk:225: recipe for target 'world' failed
make: *** [world] Error 2
##[error]Process completed with exit code 2.
 Prepare artifact
请问我这是哪里错了