coolsnowwolf/lede #2333: 为什么删除了这个ramips: delete MTK 762x support

bonnyandsky, 2019-12-07:
一觉睡醒，发现编译失败，查了更新内容发现762x系列被删除了，为啥啊

jxph1123, 2019-12-08:
谁惹大佬生气了？

bonnyandsky, 2019-12-08:
这下尴尬了，没得玩了。

hcym, 2019-12-08:
呵呵，mtk用老固件足够

bonnyandsky, 2019-12-08:
> 呵呵，mtk用老固件足够

意思是说这个删除的东西并不影响编译mt7621的固件？

EnnawYang, 2019-12-08:
> > 呵呵，mtk用老固件足够
> 
> 意思是说这个删除的东西并不影响编译mt7621的固件？

.... 影响 他这是讽刺你听不出来吗

M1A6I3L, 2019-12-08:
好气啊。。。虽然我主路由是x86，但我还有d2和k2p啊。。。

删除事件源于某斐讯用户自己不学编译，甚至云编译都不用，还整天催更新，结果把大雕气着了。。。
虽然他只是导火索，大雕早就看MTK不爽。

哎~~~

bonnyandsky, 2019-12-08:
> 好气啊。。。虽然我主路由是x86，但我还有d2和k2p啊。。。
> 
> 删除事件源于某斐讯用户自己不学编译，甚至云编译都不用，还整天催更新，结果把大雕气着了。。。
> 虽然他只是导火索，大雕早就看MTK不爽。
> 
> 哎~~~
这下尴尬了，没得用了。还好本地我还有源代码。增加删除软件可以本地编译。不过希望还是能把mtk加回来。

M1A6I3L, 2019-12-08:
> 这下尴尬了，没得用了。还好本地我还有源代码。增加删除软件可以本地编译。不过希望还是能把mtk加回来。
本地确实不影响，就算不会改代码，直接下载旧版本的代码就好，而且目前的功能MTK够用了，甚至多开几个MTK都带不起来。
但是被删了还是各种怨念🙃🙃🙃

bonnyandsky, 2019-12-08:
> > 这下尴尬了，没得用了。还好本地我还有源代码。增加删除软件可以本地编译。不过希望还是能把mtk加回来。
> 
> 本地确实不影响，就算不会改代码，直接下载旧版本的代码就好，而且目前的功能MTK够用了，甚至多开几个MTK都带不起来。
> 但是被删了还是各种怨念

请问怎么下载旧版本的代码？我自己修改的脚本我都不知道怎么恢复，原来版本只能做二次修改。

bonnyandsky, 2019-12-08:
我刚学会云编译，然后就没得玩了。

luk24, 2019-12-08:
回退这条提交就行了

hanandjun, 2019-12-08:
> > > 这下尴尬了，没得用了。还好本地我还有源代码。增加删除软件可以本地编译。不过希望还是能把mtk加回来。
> > 
> > 
> > 本地确实不影响，就算不会改代码，直接下载旧版本的代码就好，而且目前的功能MTK够用了，甚至多开几个MTK都带不起来。
> > 但是被删了还是各种怨念
> 
> 请问怎么下载旧版本的代码？我自己修改的脚本我都不知道怎么恢复原来版本。只能做二次修改。

检出指定commit id的提交，之后的提交没问题 cherry-pick 过来，这样就能继续苟着(或者拉取最新的代码，把这条提交revert下，手动滑稽)

RealKiro, 2019-12-08:
- 回退：
```
git reset --hard b0bca4dce08285142d1888e31a5e16f53ba20767
```

- 如果是fork的仓库还要再加一条：
```
git push --force
```

[image]

M1A6I3L, 2019-12-08:
> 请问怎么下载旧版本的代码？我自己修改的脚本我都不知道怎么恢复原来版本。只能做二次修改。

本地的说个最简单的，找到旧版本的commits，比如https://github.com/coolsnowwolf/lede/tree/b0bca4dce08285142d1888e31a5e16f53ba20767 然后然后点右边clone or download，选download zip，最后解压再放到本地编译环境，cd进去，之后就老步骤了。

M1A6I3L, 2019-12-08:
额大雕回滚了？？？？？

bonnyandsky, 2019-12-08:
我本地的没更新，是fork这边的自动更新了，然后fork这边怎么回退？github有这个功能吗？

bonnyandsky, 2019-12-08:
> > > > 这下尴尬了，没得用了。还好本地我还有源代码。增加删除软件可以本地编译。不过希望还是能把mtk加回来。
> > > 
> > > 
> > > 本地确实不影响，就算不会改代码，直接下载旧版本的代码就好，而且目前的功能MTK够用了，甚至多开几个MTK都带不起来。
> > > 但是被删了还是各种怨念
> > 
> > 
> > 请问怎么下载旧版本的代码？我自己修改的脚本我都不知道怎么恢复原来版本。只能做二次修改。
> 
> 检出指定commit id的提交，之后的提交没问题 cherry-pick 过来，这样就能继续苟着(或者拉取最新的代码，把这条提交revert下，手动滑稽)

GIt我才用两天，完全不明白你说的，能否说的详细一点。谢谢

bonnyandsky, 2019-12-08:
> * 回退：
> 
> ```
> git reset --hard b0bca4dce08285142d1888e31a5e16f53ba20767
> ```
> 
> * 如果是fork的仓库还要再加一条：
> 
> ```
> git push --force
> ```
> 
> [image]
是在本地的命令行执行下面两条命令就可以了吗？
git reset --hard b0bca4dce08285142d1888e31a5e16f53ba20767
git push --force // 强制覆盖线上

isforerunner, 2019-12-08:
傻逼，还真TM以为自己了不起，删MTK Support。

luk24, 2019-12-08:
> 傻逼，还真TM以为自己了不起，删MTK Support。

哇，你有本事自己搞啊，在这说个屁啊，人家的仓库想干嘛干嘛，不服转openwrt官方repo

luk24, 2019-12-08:
同步了OpenWrt 19.07-rc2的代码，大家不用担心了

bonnyandsky, 2019-12-08:
> 同步了OpenWrt 19.07-rc2的代码，大家不用担心了

所以说危机解除了？支持mtp了？

hyird, 2019-12-08:
> 傻逼，还真TM以为自己了不起，删MTK Support。

哇，你有本事自己搞啊，在这说个屁啊，人家的仓库想干嘛干嘛，不服转openwrt官方repo

outemancut, 2019-12-08:
删就删呗 自己找回来就好了 这东西借鉴就好了 没人让你完全克隆别人的仓库 直接用OPENWRT官方就好了 感觉还可以

LGA1150, 2019-12-08:
> 傻逼，还真TM以为自己了不起，删MTK Support。

本菜鸡强势围观ing

hyird, 2019-12-08:
> 删就删呗 自己找回来就好了 这东西借鉴就好了 没人让你完全克隆别人的仓库 直接用OPENWRT官方就好了 感觉还可以

删是大雕的自由 没理由bb  真的受不了 一群白嫖还有理的傻逼

phc1003, 2019-12-08:
> 傻逼，还真TM以为自己了不起，删MTK Support。

就是了不起，有本事自己搞

coolsnowwolf, 2019-12-08:
姑且不论我是同步到 19.07-rc2了，就算我删除自己代码的自由，都要你批准吗？

jxph1123, 2019-12-08:
支持大雕，不要理那些傻逼，白用还BB，不要一颗老鼠屎误了一锅饭，那种人直接无视。

zcluo, 2019-12-08:
大雕别动怒，林子大了什么鸟都有

gbaoye, 2019-12-08:
> > 傻逼，还真TM以为自己了不起，删MTK Support。
> 
> 哇，你有本事自己搞啊，在这说个屁啊，人家的仓库想干嘛干嘛，不服转openwrt官方repo

舔的不错，开源应该保持中立性，这是基本原则

ForgotFun, 2019-12-08:
不要再跪舔了！！！跟着佐大学OpenWrt开发，让你充满勇气和技术，再也不用看lean的脸色了，你也可以自己开发固件。

报名地址：http://forgotfun.org/2018/04/openwrt-training-2018.html

adore595, 2019-12-08:
不要再跪舔了！！！跟着佐大学OpenWrt开发，让你充满勇气和技术，再也不用看lean的脸色了，你也可以自己开发固件。

报名地址：http://forgotfun.org/2018/04/openwrt-training-2018.html

bonnyandsky, 2019-12-08:
> 姑且不论我是同步到 19.07-rc2了，就算我删除自己代码的自由，都要你批准吗？

大雕，我刚才删除了仓库，重新fork了你的仓库。然后我按照教程只是修改了修改后编译还是失败了，是什么问题啊。修改内容如下
  push:
     branches: 
       - master
只是删除了这3句话前面的“# ”

dolphinpaopao, 2019-12-08:
> > > 傻逼，还真TM以为自己了不起，删MTK Support。
> > 
> > 
> > 哇，你有本事自己搞啊，在这说个屁啊，人家的仓库想干嘛干嘛，不服转openwrt官方repo
> 
> 舔的不错，开源应该保持中立性，这是基本原则

站在道德的制高点把所有都批判一番，开源是心怀分享共同进步美好愿景的所有人一点一点积累起来的，并不是依靠某一个圣人或者团体一己之力建立，对所有有贡献的人都应该有一份感激之心。认为世界都应该围着自己转？严于律人宽以待己和逼捐，大抵如此。

closed 2019-12-08 by bonnyandsky

MxCen, 2019-12-08:
> 傻逼，还真TM以为自己了不起，删MTK Support。

大家让一让，我憋了一天了。我尿黄，让我滋滋他。

luk24, 2019-12-09:
> > > 傻逼，还真TM以为自己了不起，删MTK Support。
> > 
> > 
> > 哇，你有本事自己搞啊，在这说个屁啊，人家的仓库想干嘛干嘛，不服转openwrt官方repo
> 
> 舔的不错，开源应该保持中立性，这是基本原则

你指点江山的时候扪心自问下对开源社区有啥贡献，还学人家指点江山，笑skr人。开源是自由的，你说什么都行，这也是你还能在这条主题下蹦跶的原因。