coolsnowwolf/lede #2338: 求教如何在云编译的时候加入魔改版的SSR

sunlanhao, 2019-12-08:
已经按照糖炒栗子大神的教程添加了 git clone https://github.com/sunlanhao/luci-app-ssr-plus-jo.git 
请问L大 固件定制部分下面默认的SSR 还需要做什么修改吗？ 
[image]
[image]

EnnawYang, 2019-12-08:
包名都不一样好吗 当然要改

sunlanhao, 2019-12-08:
> 包名都不一样好吗 当然要改
我把名字替换成魔改版的名字可以吗？   这个还需要添加依赖吗
[image]

EnnawYang, 2019-12-08:
> > 包名都不一样好吗 当然要改
> > 我把名字替换成魔改版的名字可以吗？   这个还需要添加依赖吗
> > [image]

依赖不清楚，魔改版的我没用过

kang-mk, 2019-12-08:
我也想知道

sunlanhao, 2019-12-08:
> 我也想知道

等大佬来指点迷津

saoba1982, 2019-12-09:
我也一直没成功加入过软件，不知道怎么回事.

closed 2019-12-09 by coolsnowwolf