coolsnowwolf/lede #2339: overlay空间如何修改

xiangfeigao, 2019-12-08:
HC5962 我修改了空间大小后，生成的固件系统无法启动，感觉改的没有问题啊，请大家教教我。
修改文件如下：
partition@0 {
			label = "u-boot";
			reg = <0x0 0x80000>;
			read-only;
		};

		partition@80000 {
			label = "debug";
			reg = <0x80000 0x80000>;
			read-only;
		};

		factory: partition@100000 {
			label = "factory";
			reg = <0x100000 0x40000>;
			read-only;
		};

		partition@140000 {
			label = "kernel";
			reg = <0x140000 0x200000>;
		};

		partition@340000 {
			label = "ubi";
#reg = <0x340000 0x1E00000>; 修改为64M
	        reg = <0x340000 0x4000000>;
		};

		partition@4340000 {
			label = "hw_panic";
			reg = <0x4340000 0x80000>;
			read-only;
		};

		partition@43c0000 {
			label = "bdinfo";
			reg = <0x43c0000 0x80000>;
			read-only;
		};

		partition@4440000 {
			label = "backup";
			reg = <0x4440000 0x80000>;
			read-only;
		};

		partition@44c0000 {
			label = "overly";
			reg = <0x44c0000 0x3c00000>;
		};

		partition@80c0000 {
			label = "firmware_backup";
			reg = <0x80c0000 0x2000000>;
		};

		partition@82c0000 {
			label = "oem";
			reg = <0x82c0000 0x200000>;
		};

		partition@84c0000 {
			label = "opt";
			reg = <0x54c0000 0x2ac0000>;
		};
	};
};

coolsnowwolf, 2019-12-08:
uboot 限制了，不能这么简单的改

xiangfeigao, 2019-12-08:
有相关文章参考吗，我想修改128M的存储

xiangfeigao, 2019-12-08:
通过mtd修改分区应该很容易，我现在就想如何通过编译修改mtd分区，请指教

closed 2019-12-08 by xiangfeigao