coolsnowwolf/lede #2343: Trojan 的是否去除的讨论

coolsnowwolf, 2019-12-08:
SSR Plus: 为 Trojan 提供官方并不支持的 UDP 中继功能

我感觉被那些 Up 主骗了，这玩意 ：

1.本质上就是 V2 的 WS + TLS 简化版本 
2. 并未内部实现多线程 
3.并不支持 UDP 透明代理

大家觉得这个功能是否应该保留？

EnnawYang, 2019-12-08:
暂时没有这个需求，我使用的机场都不支持这个（我个人觉得这个好像没那么火啊

wentijianihao, 2019-12-08:
效果好的话，有必要保留吧

chchia, 2019-12-08:
我建议继续提供支持，谢谢。我是外国人常到中国出差。
我有个小米nano，出差带着用，内存太小不支持v2ray的ws+tls

昨天编译你的固件后，trojan运行正常至今，非常稳定 （除了没支持udp一切都好）。
不过今天最新用itp2sock后目前变成 ShadowsocksR Plus+ 未运行 了。

我现在从新clone一次再编译看看。

hcym, 2019-12-08:
技术上的事不懂，喜欢简单粗暴的tcp，编译官方源码时要么v2要么ssr，艰难时期再走ipv6，从来不用混淆什么的。

gpeditmsc1986, 2019-12-08:
多余类似叠加的功能必须剔除，系统臃肿

snakwu, 2019-12-08:
老老实实的，把ssr-plus弄些实用功能吧，例如支持自定义dns，和支持dnsforwarder解释吧，dnsforwarder是pdns的优秀代替方案，还有支持v2ray定时自动更新二进制，这些都是多么实用和好用的功能。。。。。。。。。。。

coolsnowwolf, 2019-12-08:
> 老老实实的，把ssr-plus弄些实用功能吧，例如支持自定义dns，和支持dnsforwarder解释吧，dnsforwarder是pdns的优秀代替方案，还有支持v2ray定时自动更新二进制，这些都是多么实用和好用的功能。。。。。。。。。。。

恰恰相反，不应该有更多选项，只会让人更迷惑。
做减法比做加法更难。

It just works 是这个固件的原则

snakwu, 2019-12-08:
dnsforwarder是pdns的优秀代替方案, 明显的比pdns解释要快一些,怕迷惑直接dnsforwarder替代掉pdns吧,pdns有些过时了,自定义dns就是提供一些默认的dns外,可自已填ip,这个算不上什么加法和减法,我不懂你的想法,只是建议,最后还是看你

snakwu, 2019-12-08:
固件,我觉得,道要原则不是保守,是使到它更好用

snakwu, 2019-12-08:
不过,话说回来,也非常感谢你,一直在默默无闻的更新这个固件,是一个非常好的项目

redchenjs, 2019-12-08:
然而好用是没有标准的，真正想要好用唯有自己实现，毕竟一个人一个瓜

snakwu, 2019-12-08:
> 然而好用是没有标准的，真正想要好用唯有自己实现，毕竟一个人一个瓜
之前就有人讨论过这个话题
https://github.com/coolsnowwolf/lede/issues/725

keke1023, 2019-12-08:
首先就是要感谢lean大不断的付出，至于功能，ss和v2今年两三次大事下来看都属于容易被阻断的形式，v2能上ws+tls当然好，但是就今年的情况来说，trojan的评价还是一致的认为隐蔽性强，个人实际使用下来应该比v2的硬件要求低，连接速度上也貌似快一些，希望能保留，如果一个选择足够好，那么做减法比较好做，但是目前眼看前途未卜，还是希望能先做加法再做减法
然后目前的编译出来，不知道别人是否遇到这个问题，我的只要用trojan服务器+开udp，就会卡在应用画面一直转，然后pdnsd运行不起来，关掉udp的话就正常运行
再次致谢，辛苦lean大了

jh0072110, 2019-12-08:
从最近小蔷蔷的动向来看，有些减法不能做，trojan从现阶段来看没必要加，速度也不太行，可以友好的期待它的发展，可是正规军ss现在添加的一些混淆之类的，比较成熟，我认为是可以添加的唉（obfs和v2ray plugin），也不知道大佬对ss阵营有什么偏见一直不添加，ss比较起来还是最轻量化和最快的，个人反倒希望能加入这些的支持…

nameless3721, 2019-12-08:
[image]
udp服务端应该是支持的 可能插件未跟进  trojan的确和v2 ws+tls类似不过效率高点对于机场的硬件开销少点吧 速度嘛都一样工具只是提供手段 能跑多少完全看用的什么服务器 世界加钱可及 我觉得如果不是有啥太严重的冲突作为一个附加支持保留吧

keivenliao, 2019-12-09:
个人觉得看个人需要，我软路由，加谷歌云最低配，目前 V2+ws+tls  速度和稳定都OK。
从大众使用的角度来说，固件支持是可以的，但是要不要用就看个人了。 

另外， 我也想要 SS 的混淆控件！

hzjnet, 2019-12-09:
既然已经进来了，也有需求，建议保留吧

kfen, 2019-12-09:
同意snakwu和keivenliao的观点，$$P+添加能自定义dns和$$混淆插件(obfs) ,挺实用的两个功能~

norikohxb, 2019-12-09:
保留啊，trojan在你看youtube的时候使用youtube上面的下载的插件，在下载的时候可以达到20MB/s的下载速度，但是很快有失败网络错误，我估计是给流媒体服务器察觉到速度太快了给你切断连接，
https://r1---sn-i3belnez.googlevideo.com/videoplayback?expire=1575878112&ei=gKntXenlAofG4wKggr_4CA&ip=35.220.154.2&id=o-ADfD6MUj7S0UYAB6iEyz8eBwQ2-后面一大堆

norikohxb, 2019-12-09:
在chrome那边有youtube的下载插件，就是当你打开youtube的时候会嗅探出视频下载地址，我下载8K的视频，文件名是webm（youtube的8k视频视频+音频是分开的）用同一个网段的相同谷歌云的不同ip的v2ray和trojan对比， v2ray的速度就是持续6MBps（大约60Mbps）不会断流，可以一个900MB的视频从头下载到尾，但是用v2ray下载，速度很快可以20MB（200mbps）下载，但是会断流，就是每次下载了比如10分之一，chrome报网络错误，然后还要重试，又是快速下载，我不知道是trojan插件没有更新完善，还是谷歌的流媒体服务器感受到20MB（200Mbps）每秒的下载速度超过正常看视频的需求就给你断流了。

norikohxb, 2019-12-09:
不要一下子扼杀trojan在摇篮中，另外你可以看看我的服务器www.xmve.com是有内容的，至少他这种伪装方式会比v2ray先进，等gfw正式封杀v2ray的时候，你trojan至少伪装的比较好，还有一个网站在那边让gfw误解为http的访问流量。所以感觉还是要留。新技术出来总是要尝试到完善阶段，如果真的不行再放弃他。这个才出来多久。回帖反对的又有几个人是第一时间就去假设自己trojan的服务器。

norikohxb, 2019-12-09:
还有我之前在恩山论坛反馈的，trojan会出错，是因为所用的op的ssr的插件对于trojan没有编译好导致，换一个其他人编译的op就行了，现在就是用youtube下载视频的嗅探插件，嗅探出视频webm的地址，进行chrome的下载，trojan会因为速度超级快而断流（我的宽带就300Mbps，trojan下载流媒体webm下来的速度可以达到25MB每秒，相当于占据了我大部分带宽了）这个断流问题不知道是ssr的问题还是谷歌的流媒体服务器感觉到不寻常流量给你掐断了。

norikohxb, 2019-12-09:
traojan在youtube测速上并没有感觉会有v2ray有明显区别，区别最大的就是下载8K的 youtube视频直接从谷歌服务器下载，trojan的速度可以挖掘整个带宽，但是好像很快就断流了，然后必须重试，就又可以高速下载，一个900mbs的视频估计要重试个7，8次才能下载完毕。这个如果不是谷歌流媒体服务器屏蔽的大流量断线的策略就麻烦能在后续的ssr改进，这样就太棒了，下载8K的youtube视频，按照200Mbps的速度在跑，我是很开心的

lifetyper, 2019-12-09:
很抱歉，我当初也是被trojan自己的宣传误导了，到后来我发现这玩意确实就和lean大说的一样，就是v2 tls的简化版，并没有任何特殊的地方，而且udp转发和多线程这些v2明显做得更好。
如果非要说好处，trojan唯一的好处可能就是服务器端内存占用少一点。

至于是否移除，如果当前功能已经足够并且不会影响ssr plus其他部分的工作，建议还是保留，如果哪天trojan本身已经对ssr plus是个制约，那移除掉我也是支持的。
其实大家用的最多的还是ss/ssr/v2这几个，而且就其他平台的客户端支持来看，trojan也差得远。

hplee0120, 2019-12-09:
希望添加一个指定备用服务器功能

MxCen, 2019-12-09:
暂时没有这个需求。

norikohxb, 2019-12-09:
lean大大有空的时候能不能看看
2019-12-09 15:12:56 Welcome to trojan 1.13.0 Started!
2019-12-09 15:12:53 redir udp error.restart!
2019-12-09 15:12:21 Welcome to trojan 1.13.0 Started!
2019-12-09 15:12:19 redir udp error.restart!
目前在用的trojan一直有这个错误能否看看是什么问题导致

azgh, 2019-12-09:
@coolsnowwolf 建议先不加，毕竟与V很相似，根据硬件的差异可采取以下：

- 配置低，选择ss，加obfs，一直稳定使用；

- 配置高，选择v2ray，加ws+tls，也一直稳定使用。

T感觉就是介于S和V之间的产品，但平台的扩展性，远不如S和V，如果哪一天SS+OBFS和V2ray+WS+TLS不能正常使用了，再选择新的方案加入，没必要添加非常多重复的插件导致系统的稳定行下降以及编译的复杂程度，保持精品路线。

tufu9441, 2019-12-09:
> @coolsnowwolf 建议先不加，毕竟与V很相似，根据硬件的差异可采取以下：
> 
> * 配置低，选择ss，加obfs，一直稳定使用；
> * 配置高，选择v2ray，加ws+tls，也一直稳定使用。
> 
> T感觉就是介于S和V之间的产品，但平台的扩展性，远不如S和V，如果哪一天SS+OBFS和V2ray+WS+TLS不能正常使用了，再选择新的方案加入，没必要添加非常多重复的插件导致系统的稳定行下降以及编译的复杂程度，保持精品路线。

完全赞成 @azgh 的建议~

jzhou404, 2019-12-09:
> 我建议继续提供支持，谢谢。我是外国人常到中国出差。
> 我有个小米nano，出差带着用，内存太小不支持v2ray的ws+tls
> 
> 昨天编译你的固件后，trojan运行正常至今，非常稳定 （除了没支持udp一切都好）。
> 不过今天最新用itp2sock后目前变成 ShadowsocksR Plus+ 未运行 了。
> 
> 我现在从新clone一次再编译看看。

我的小米nano，lean OP 支持v2ray的ws+tls很正常， 一直7x24在使用啊？

chchia, 2019-12-09:
> > 我建议继续提供支持，谢谢。我是外国人常到中国出差。
> > 我有个小米nano，出差带着用，内存太小不支持v2ray的ws+tls
> > 昨天编译你的固件后，trojan运行正常至今，非常稳定 （除了没支持udp一切都好）。
> > 不过今天最新用itp2sock后目前变成 ShadowsocksR Plus+ 未运行 了。
> > 我现在从新clone一次再编译看看。
> 
> 我的小米nano，lean OP 支持v2ray的ws+tls很正常， 一直7x24在使用啊？

不知道你的nano是什么版本？我的nano只有64MB内存，每次一运行v2ray ws+tls就会出现not enough memory。。。。。

gbaoye, 2019-12-09:
> > 老老实实的，把ssr-plus弄些实用功能吧，例如支持自定义dns，和支持dnsforwarder解释吧，dnsforwarder是pdns的优秀代替方案，还有支持v2ray定时自动更新二进制，这些都是多么实用和好用的功能。。。。。。。。。。。
> 
> 恰恰相反，不应该有更多选项，只会让人更迷惑。
> 做减法比做加法更难。
> 
> It just works 是这个固件的原则

为什么你不为用户选择编译固件的默认选项，还有剥夺用户的选择权，对用户来说，仅仅多了一个选项，从成千上万的选项里多了一个实用的选项，如果你真的在乎用户的选项，最好的方式是把LUCI英文备注成简体中文，一劳而永逸，何乐而不为？

gbaoye, 2019-12-09:
每一个型号都有最佳的选项，最佳的驱动，IPV6，实用而必备功能，不是有些人用不到，是不知道，从几千个选项里打勾，还是从几个选项里划掉，我不知道哪一个更加简单便捷

eallion, 2019-12-10:
目前比较理想的方案是 shadowsocks-libev + v2ray-plugin , SSR 和 v2ray 都有明显的缺陷。
SSR 本身已经多年不更新，目前 SSR 还这么多人在用是因为机场主只提供这种落后的东西。
（PS：本人坚决不用机场，也不卖机场。）
Trojan 的理念不错，而且对 ip 的保护比较好，不容易被封，但是速度不是很理想。
结论：如果 SSR+ 不能添加 v2ray-plugin ,建议保留 Trojan 。（当时添加其他很多种玩具那也是极好的。）

ioxuy, 2019-12-10:
建议保留,在LEDE没有添加trojan的时候,我自已编译了openwrt-trojan在用,稳定性很不错!

coolsnowwolf, 2019-12-11:
> It just works

这个固件发展到现在，砍掉或者一开始就不做的功能起码占了一半
换来的是稳定和简单，还有高性能

star-o0O, 2019-12-12:
> > > 老老实实的，把ssr-plus弄些实用功能吧，例如支持自定义dns，和支持dnsforwarder解释吧，dnsforwarder是pdns的优秀代替方案，还有支持v2ray定时自动更新二进制，这些都是多么实用和好用的功能。。。。。。。。。。。
> > 
> > 
> > 恰恰相反，不应该有更多选项，只会让人更迷惑。
> > 做减法比做加法更难。
> > It just works 是这个固件的原则
> 
> 为什么你不为用户选择编译固件的默认选项，还有剥夺用户的选择权，对用户来说，仅仅多了一个选项，从成千上万的选项里多了一个实用的选项，如果你真的在乎用户的选项，最好的方式是把LUCI英文备注成简体中文，一劳而永逸，何乐而不为？

怎么说呢，这是一个开源项目，有什么需求或你认为可以的配置，完全可以pull上来，合入的也不少。你也可以提交翻译，想翻译多少/什么语言都可以。上来就是为什么你不，就像为什么你不给我钱，为什么你不给我一份955而且月入10K的工作？

jh0072110, 2019-12-12:
> 怎么说呢，这是一个开源项目，有什么需求或你认为可以的配置，完全可以pull上来，合入的也不少。你也可以提交翻译，想翻译多少/什么语言都可以。上来就是为什么你不，就像为什么你不给我钱，为什么你不给我一份955而且月入10K的工作？

哟嘿，老哥这话有点冲了啊，我倒是觉得本来也是无偿的，图的就是爱用用不爱用拉倒的状态，会编的自个儿就编了，咱不会编就安心舔jio就对了，没什么好抱怨的。
PS：继续给大佬舔jio希望加入ss的插件功能（obfs&v2ray-plugin），现在这网络环境，单纯的ss扛不住了啊，这是刚需啊大佬TAT

tufu9441, 2019-12-12:
> 哟嘿，老哥这话有点冲了啊，我倒是觉得本来也是无偿的，图的就是爱用用不爱用拉倒的状态，会编的自个儿就编了，咱不会编就安心舔jio就对了，没什么好抱怨的。
> PS：继续给大佬舔jio希望加入ss的插件功能（obfs&v2ray-plugin），现在这网络环境，单纯的ss扛不住了啊，这是刚需啊大佬TAT

SS的simple-obfs确实好用，虽然是上古时期的产物但是很稳，我每次编译时都会自己加进去。

adamw92, 2019-12-12:
不如取消然后支持SS的插件

tefiszx, 2019-12-13:
v2才是正道。其他的都是渣渣。

colingpt, 2019-12-14:
> v2才是正道。其他的都是渣渣。

什么时候解决了nat 的问题再说正道吧。。。

closed 2019-12-15 by coolsnowwolf

xorryk, 2020-02-09:
> [image]
> udp服务端应该是支持的 可能插件未跟进 trojan的确和v2 ws+tls类似不过效率高点对于机场的硬件开销少点吧 速度嘛都一样工具只是提供手段 能跑多少完全看用的什么服务器 世界加钱可及 我觉得如果不是有啥太严重的冲突作为一个附加支持保留吧

请问能告知下NAT fullcone 情况下的服务器设置吗？

bigbighill, 2020-02-18:
感觉gost最简洁，比trojan还要方便快捷

mOnkD404, 2020-03-04:
手机小火箭用trojan比v2延迟低挺多的，软路由上差不多

chacha20, 2020-05-29:
出来透个气，但一直都不敢发言