coolsnowwolf/lede #2347: 软路由编译samba4报错，请问怎么办？！

zxlhhyccc, 2019-12-08:
[image]

redchenjs, 2019-12-08:
应该立即关闭电脑然后睡觉

chchia, 2019-12-08:
你提供的信息太少了，看不出错误原因。

zxlhhyccc, 2019-12-08:
> 你提供的信息太少了，看不出错误原因。

这个报错我是已经去掉了samaba36

zxlhhyccc, 2019-12-08:
> 你提供的信息太少了，看不出错误原因。

time: package/feeds/packages/attr/compile#0.06#0.10#0.47
bash: nproc: command not found
make[2]: Entering directory `/home/lin/lede/package/lean/samba4'
rm -f /home/lin/lede/build_dir/target-x86_64_musl/samba-4.9.2/.built
touch /home/lin/lede/build_dir/target-x86_64_musl/samba-4.9.2/.built_check
(cd /home/lin/lede/build_dir/target-x86_64_musl/samba-4.9.2; ./buildtools/bin/waf install -j --targets=auth_builtin,auth_sam,auth_unix,auth_script,pdb_smbpasswd,pdb_tdbsam,vfs_default,vfs_fruit,vfs_shadow_copy2,vfs_recycle,vfs_fake_perms,vfs_readonly,vfs_cap,vfs_offline,vfs_crossrename,vfs_catia,vfs_streams_xattr,vfs_xattr_tdb,smbd/smbd,smbpasswd,pdbedit,testparm,smbstatus,smbtree,smbget,mvxattr,nmblookup,net,smbcontrol,profiles,rpcclient,smbcacls,smbcquotas,client/smbclient,client/cifsdd --destdir="/home/lin/lede/build_dir/target-x86_64_musl/samba-4.9.2/ipkg-install" )
waf [command] [options]

Main commands (example: ./waf build -j4)
  build       : build all targets
  clean       : removes the build files
  configure   : configures the project
  ctags       : build 'tags' file using ctags
  dist        : makes a tarball for distribution
  distcheck   : test that distribution tarball builds and installs
  distclean   : removes the build directory
  etags       : build TAGS file using etags
  install     : installs the build files
  pep8        : run pep8 validator
  pydoctor    : build python apidocs
  reconfigure : reconfigure if config scripts have changed
  test        : Run the test suite (see test options below)
  testonly    : run tests without doing a build first
  uninstall   : removes the installed files
  wafdocs     : build wafsamba apidocs
  wildcard_cmd: called on a unknown command

waf: error: option -j: invalid integer value: '--targets=auth_builtin,auth_sam,auth_unix,auth_script,pdb_smbpasswd,pdb_tdbsam,vfs_default,vfs_fruit,vfs_shadow_copy2,vfs_recycle,vfs_fake_perms,vfs_readonly,vfs_cap,vfs_offline,vfs_crossrename,vfs_catia,vfs_streams_xattr,vfs_xattr_tdb,smbd/smbd,smbpasswd,pdbedit,testparm,smbstatus,smbtree,smbget,mvxattr,nmblookup,net,smbcontrol,profiles,rpcclient,smbcacls,smbcquotas,client/smbclient,client/cifsdd'
make[2]: *** [/home/lin/lede/build_dir/target-x86_64_musl/samba-4.9.2/.built] Error 2
make[2]: Leaving directory `/home/lin/lede/package/lean/samba4'
time: package/lean/samba4/compile#0.84#0.44#5.24
make[1]: *** [package/lean/samba4/compile] Error 2
make[1]: Leaving directory `/home/lin/lede'
make: *** [package/lean/samba4/compile] 错误 2

caicaicai21, 2019-12-23:
autosamba 这个包也会引入 samaba36。

closed 2019-12-24 by coolsnowwolf