coolsnowwolf/lede #2360: 迅雷下载任务一多时可能断网

zhangdp, 2019-12-09:
当迅雷同时下载任务一多时比如超过10个，大概率会断网，连接这个路由的所有设备都无法上网并且无法ping通外网。但管理网页能正常访问，网络诊断那里可以ping通，关掉迅雷后等几分钟后恢复正常。
经测试不是光猫或者运营商的问题：光猫拨号，lede wan口dhcp当二级路由使用，当连接lede的电脑迅雷下载一多断网的时候，连接光猫的设备能正常访问外网。
硬件配置应该也不是瓶颈，CPU i5-6500、16G内存、nvme ssd的Esxi 6.7分配2核心、1GB内存，按理说应该够用，连接数不到1万，内存还剩900M多一点，Esxi监控页面cpu使用率10%多一点。
推测可能是跟某个插件冲突？目前选择插件如下：
LuCI ---> Applications ---> luci-app-accesscontrol  #访问时间控制
LuCI ---> Applications ---> luci-app-adbyby-plus   #广告屏蔽大师Plus +
LuCI ---> Applications ---> luci-app-arpbind  #IP/MAC绑定
LuCI ---> Applications ---> luci-app-autoreboot  #支持计划重启
LuCI ---> Applications ---> luci-app-ddns   #动态域名 DNS（集成阿里DDNS客户端）
LuCI ---> Applications ---> luci-app-filetransfer  #文件传输（可web安装ipk包）
LuCI ---> Applications ---> luci-app-firewall   #添加防火墙
LuCI ---> Applications ---> luci-app-flowoffload  #Turbo ACC网络加速（集成FLOW,BBR,NAT,DNS...
LuCI ---> Applications ---> luci-app-frpc   #内网穿透 Frp
LuCI ---> Applications ---> luci-app-ipsec-virtual**d  #virtual**服务器 IPSec
LuCI ---> Applications ---> luci-app-mwan3   #MWAN3负载均衡
LuCI ---> Applications ---> luci-app-mwan3helper   #MWAN3分流助手
LuCI ---> Applications ---> luci-app-nlbwmon   #网络带宽监视器
LuCI ---> Applications ---> luci-app-p p t p-server  #virtual**服务器 p p t p
LuCI ---> Applications ---> luci-app-ramfree  #释放内存
LuCI ---> Applications ---> luci-app-samba   #网络共享（Samba）
LuCI ---> Applications ---> luci-app-sqm  #流量智能队列管理（QOS）
-------------------------------------------------------------------------------------------
LuCI ---> Applications ---> luci-app-乳酸菌饮料-plus   #乳酸菌饮料兲朝上网Plus+
LuCI ---> Applications ---> luci-app-乳酸菌饮料-plus ---> Include s-s New Versiong  #新SS代理
LuCI ---> Applications ---> luci-app-乳酸菌饮料-plus ---> Include v2瑞  #v2瑞透明代理
LuCI ---> Applications ---> luci-app-乳酸菌饮料-plus ---> Include Kcptun  #Kcptun代理
LuCI ---> Applications ---> luci-app-乳酸菌饮料-plus ---> Include Trojan  #Trojan代理
-------------------------------------------------------------------------------------------
LuCI ---> Applications ---> luci-app-syncdial  #多拨虚拟网卡（原macvlan）
LuCI ---> Applications ---> luci-app-unblockmusic  #解锁网易云灰色歌曲
LuCI ---> Applications ---> luci-app-upnp   #通用即插即用UPnP（端口自动转发）
LuCI ---> Applications ---> luci-app-vlmcsd  #KMS服务器设置
LuCI ---> Applications ---> luci-app-vsftpd  #FTP服务器

LuCI ---> Applications ---> luci-app-wol   #WOL网络唤醒
LuCI ---> Applications ---> luci-app-wrtbwmon  #实时流量监测
LuCI ---> Applications ---> luci-app-xlnetacc  #迅雷快鸟
LuCI ---> Applications ---> luci-app-zerotier  #ZeroTier内网穿透

Extra packages  --->  ipv6helper  #支持 ipv6

Utilities  --->  open-vm-tools  #打开适用于VMware的VM Tools

请问大佬是哪个插件不兼容还是要怎么解决？

coolsnowwolf, 2019-12-09:
默认开 Fullcone ，迅雷把端口占完了

zhangdp, 2019-12-09:
感谢大佬的回复~~好像FULLCONE没有地方关闭，是只能把Turbo ACC卸载掉吗？

hcym, 2019-12-09:
防火墙里面那个full设置成0，n1单背必须要有的，不过可能影响下载p2p哦，我是不用这个功能

zhangdp, 2019-12-09:
就没有完美的解决方案吗。。

linguowei, 2019-12-10:
我是pt一旦满速下载也会断网，只能重启才恢复，这个和你是一样的原因吗

linguowei, 2019-12-10:
@zhangdp 找了一下FULLCONE NAT可以在防火墙里面关闭，我已经关了，现在再看看还会不会出现断网的情况

hcym, 2019-12-10:
我就纳闷，现在的网络流量随便看看足够吧，偶尔浏览器下载win10不过两三分钟，bt电驴迅马的有什么用

linguowei, 2019-12-11:
@hcym 讨论这个没意义，你有你的使用场景，别人有别人的

zhangdp, 2019-12-12:
防火墙里关闭FULLCONE NAT了，还是有可能断网

linguowei, 2019-12-13:
@zhangdp 那奇怪，我的关了，现在用两天了，没再断了

closed 2019-12-20 by coolsnowwolf