coolsnowwolf/lede #2365: 最新版打开编译界面出现这个警告

jxph1123, 2019-12-10:
make menuconfig
WARNING: Makefile 'package/feeds/packages/collectd/Makefile' has a dependency on 'librouteros', which does not exist

ljk4160, 2019-12-10:
WARNING: Makefile 'package/feeds/packages/collectd/Makefile' has a dependency on 'librouteros', which does not exist
同

outemancut, 2019-12-10:
WARNING: Makefile 'package/feeds/packages/collectd/Makefile' has a dependency on 'librouteros', which does not exist
一样哈哈

outemancut, 2019-12-10:
去OP官方把librouteros复制一下到相对文件夹就OK了 哈哈哈

eallion, 2019-12-10:
```
WARNING: Makefile 'package/lean/amule/Makefile' has a dependency on 'libpng', which does not exist
WARNING: Makefile 'package/lean/amule/Makefile' has a dependency on 'libupnp', which does not exist
WARNING: Makefile 'package/lean/amule/Makefile' has a build dependency on 'libgd', which does not exist
WARNING: Makefile 'package/lean/antileech/Makefile' has a build dependency on 'libgd', which does not exist
WARNING: Makefile 'package/lean/autocore/Makefile' has a dependency on 'bc', which does not exist
WARNING: Makefile 'package/lean/autocore/Makefile' has a dependency on 'lm-sensors', which does not exist
WARNING: Makefile 'package/lean/automount/Makefile' has a dependency on 'kmod-fs-exfat', which does not exist
WARNING: Makefile 'package/lean/automount/Makefile' has a dependency on 'ntfs-3g', which does not exist
WARNING: Makefile 'package/lean/autosamba/Makefile' has a dependency on 'luci-app-samba', which does not exist
WARNING: Makefile 'package/utils/busybox/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/busybox/Makefile' has a build dependency on 'libpam', which does not exist
WARNING: Makefile 'package/lean/cifsd-tools/Makefile' has a dependency on 'glib2', which does not exist
WARNING: Makefile 'package/network/utils/curl/Makefile' has a dependency on 'libgnutls', which does not exist
WARNING: Makefile 'package/network/utils/curl/Makefile' has a dependency on 'libopenldap', which does not exist
WARNING: Makefile 'package/network/utils/curl/Makefile' has a dependency on 'libssh2', which does not exist
WARNING: Makefile 'package/lean/ddns-scripts_aliyun/Makefile' has a dependency on 'ddns-scripts', which does not exist
WARNING: Makefile 'package/lean/ddns-scripts_aliyun/Makefile' has a dependency on 'wget', which does not exist
WARNING: Makefile 'package/lean/default-settings/Makefile' has a dependency on 'luci-base', which does not exist
WARNING: Makefile 'package/lean/dnsforwarder/Makefile' has a dependency on 'wget', which does not exist
WARNING: Makefile 'package/lean/dnsforwarder/Makefile' has a dependency on 'coreutils-base64', which does not exist
WARNING: Makefile 'package/lean/ipt2socks/Makefile' has a build dependency on 'libuv', which does not exist
WARNING: Makefile 'package/lean/ipv6-helper/Makefile' has a dependency on 'luci-proto-ipv6', which does not exist
WARNING: Makefile 'package/lean/ipv6-helper/Makefile' has a dependency on 'wget', which does not exist
WARNING: Makefile 'package/boot/kexec-tools/Makefile' has a dependency on 'liblzma', which does not exist
WARNING: Makefile 'package/lean/libidn2/Makefile' has a dependency on 'libunistring', which does not exist
WARNING: Makefile 'package/network/services/lldpd/Makefile' has a dependency on 'libnetsnmp', which does not exist
WARNING: Makefile 'package/lean/luci-app-accesscontrol/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-adbyby-plus/Makefile' has a dependency on 'wget', which does not exist
WARNING: Makefile 'package/lean/luci-app-adbyby-plus/Makefile' has a dependency on 'coreutils', which does not exist
WARNING: Makefile 'package/lean/luci-app-adbyby-plus/Makefile' has a dependency on 'coreutils-nohup', which does not exist
WARNING: Makefile 'package/lean/luci-app-adbyby-plus/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-amule/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-arpbind/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-autoreboot/Makefile' has a dependency on 'luci', which does not exist
WARNING: Makefile 'package/lean/luci-app-autoreboot/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-baidupcs-web/Makefile' has a dependency on 'wget', which does not exist
WARNING: Makefile 'package/lean/luci-app-baidupcs-web/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-cifsd/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-dnsforwarder/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-docker/Makefile' has a dependency on 'docker-ce', which does not exist
WARNING: Makefile 'package/lean/luci-app-docker/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-familycloud/Makefile' has a dependency on 'coreutils', which does not exist
WARNING: Makefile 'package/lean/luci-app-familycloud/Makefile' has a dependency on 'coreutils-nohup', which does not exist
WARNING: Makefile 'package/lean/luci-app-familycloud/Makefile' has a dependency on 'bash', which does not exist
WARNING: Makefile 'package/lean/luci-app-familycloud/Makefile' has a dependency on 'wget', which does not exist
WARNING: Makefile 'package/lean/luci-app-familycloud/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-filetransfer/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-flowoffload/Makefile' has a build dependency on 'luci-base/host', which does not exist
[... cut ...]
WARNING: Makefile 'package/lean/trojan/Makefile' has a dependency on 'boost-system', which does not exist
WARNING: Makefile 'package/lean/trojan/Makefile' has a dependency on 'boost-program_options', which does not exist
WARNING: Makefile 'package/lean/trojan/Makefile' has a dependency on 'boost-date_time', which does not exist
WARNING: Makefile 'package/lean/UnblockNeteaseMusic/Makefile' has a dependency on 'node', which does not exist
WARNING: Makefile 'package/lean/wxbase/Makefile' has a dependency on 'libexpat', which does not exist
```

closed 2019-12-11 by coolsnowwolf