coolsnowwolf/lede #2368: 希望SSR+加入按内网IP地址单独设置是走gfwlist还是ip白名单。

bingo235, 2019-12-10:
目前路由上设置全局走ip白名单，因为很多网站虽然没有被墙，但速度很慢，比如github。
但此时bt流量会走ssr，会被协议审计。家里的plex服务获取外网的ip时也获取的ssr ip，无法直通。
所以ssr+最好有指定某个内网IP走gfwlist的功能，可以指定内网ip，或者端口号，这样对于需要特殊功能的内网主机会方便很多。

outemancut, 2019-12-10:
越来越复杂了  这个加一下那个加一下 东西越来越大

bingo235, 2019-12-10:
我觉得先把ssr功能完善，目前最稳的还是ssr，我现在有魅影 喵帕斯，都是443端口ssr最稳。
魅影能稳定跑满200Mbps，喵帕斯90%情况能稳定跑满150Mbps。
自建的小鸡也是ssr走443端口tls混淆，也很稳定。
其他v2ray，trojan都尝试过，并没有明显优势。

bingo235, 2019-12-10:
内网IP分组，不同IP走不同代理模式，我觉得很有必要。

AmadeusGhost, 2019-12-10:
bt流量这个，可以这样设置让它不走xxr：
[image]

closed 2019-12-11 by coolsnowwolf

bingo235, 2019-12-12:
这个并不能满足需求，常用端口只有22,53,587,465,995,993,143,80,443，像我需要走ssr连接vps，vps的ssh一般都把默认端口改掉。或者先常用端口可以自定义。 @coolsnowwolf

bingo235, 2019-12-12:
我看现在代码，模式选择IP白名单或者GFW会建立相应的mangle规则表。IP分组只需要启动是把所有mangle规则表都建起来，然后把需要分组的IP加到对应的规则表里。其实实现也很方便吧。