coolsnowwolf/lede #2369: Nano Pi R1s H5编译好固件无法刷入(不知道target是哪个。。。)

i6o6i, 2019-12-10:
相关配置信息
```
cat >> .config <<EOF
CONFIG_TARGET_sunxi_cortexa53=y
CONFIG_TARGET_sunxi=y
EOF
```
更多详细的配置在这
[openwrt-ci.yml](https://github.com/i6o6i/lede/blob/master/.github/workflows/openwrt-ci.yml)

keke1023, 2019-12-10:
我是用官方提供的源码+lean的package编译出来的，暂时还不知道怎么添加这个机型，比较接近的neo2可以试着改一下，加上usb-net驱动和无线驱动

tjuyy, 2019-12-10:
NanoPi R1S-H5 可以fork我的repo，用Github Actions自动编译
[tjuyy/NanoPi-R1S-H5-Openwrt-Autobuild](https://github.com/tjuyy/NanoPi-R1S-H5-Openwrt-Autobuild)

i6o6i, 2019-12-10:
对于我这种不会硬件，又不想装ubuntu的人来说简直是福音（刚刚还想着刷好官方固件，直接opkg一个一个插件装来着），非常感谢

tjuyy, 2019-12-10:
> 对于我这种不会硬件，又不想装ubuntu的人来说简直是福音（刚刚还想着刷好官方固件，直接opkg一个一个插件装来着），非常感谢

对，由于官方还不支持这个设备，只能用FriendlyArm fork的openwrk，再导入Lean的package，

Actions 确实会方便很多

i6o6i, 2019-12-10:
额。。这里好像遇到点问题，ssh 的输出看不到。。。不能进去make menuconfig怎么解决。。

closed 2019-12-10 by i6o6i