coolsnowwolf/lede #2381: 编译K2（psg1218a）无法生成固件

lxl6125, 2019-12-10:
按教程编译完之后只有这几个文件。。。
config.seed
openwrt-ramips-mt7620-psg1218a.manifest
openwrt-ramips-mt7620-psg1218a-initramfs-kernel.bin
sha256sums

同样的方法编译小米mini就一切正常

EnnawYang, 2019-12-11:
固件太大

lxl6125, 2019-12-11:
> 固件太大

请问该怎么解决🤔

EnnawYang, 2019-12-11:
> > 固件太大
> 
> 请问该怎么解决🤔

自己精简下咯

hong26, 2019-12-11:
碰到同样问题，只能生成initramfs-kernel固件，首次编译生成的initramfs-kernel固件大概7.8m左右，刷进去后可以正常使用但重启无法保存配置。使用make clean清除后尝试着精简，编译出来的固件反而达到了9m，依然只能生成initramfs-kernel固件。有没有办法解决呢？

lxl6125, 2019-12-12:
> 碰到同样问题，只能生成initramfs-kernel固件，首次编译生成的initramfs-kernel固件大概7.8m左右，刷进去后可以正常使用但重启无法保存配置。使用make clean清除后尝试着精简，编译出来的固件反而达到了9m，依然只能生成initramfs-kernel固件。有没有办法解决呢？

主要是内核太大了，内核模块选项又锁定了，精简不了

lxl6125, 2019-12-12:
> 碰到同样问题，只能生成initramfs-kernel固件，首次编译生成的initramfs-kernel固件大概7.8m左右，刷进去后可以正常使用但重启无法保存配置。使用make clean清除后尝试着精简，编译出来的固件反而达到了9m，依然只能生成initramfs-kernel固件。有没有办法解决呢？

make menuconfig的时候把LuCi——Applications里面你不需要的东西取消掉，我就保留了adbyby-plus和ssr-plus，编译成功了

hong26, 2019-12-12:
> 碰到同样问题，只能生成initramfs-kernel固件，首次编译生成的initramfs-kernel固件大概7.8m左右，刷进去后可以正常使用但重启无法保存配置。使用make clean清除后尝试着精简，编译出来的固件反而达到了9m，依然只能生成initramfs-kernel固件。有没有办法解决呢？

已经解决了，办法是删除编译时产生的所有文件夹（包括脚本文件夹和构建文件夹）重新精简后进行编译就能编译成功啦！要省事可以删除目录后重新拉取项目代码，精简后进行编译。

hong26, 2019-12-12:
> > 碰到同样问题，只能生成initramfs-kernel固件，首次编译生成的initramfs-kernel固件大概7.8m左右，刷进去后可以正常使用但重启无法保存配置。使用make clean清除后尝试着精简，编译出来的固件反而达到了9m，依然只能生成initramfs-kernel固件。有没有办法解决呢？
> 
> make menuconfig的时候把LuCi——Applications里面你不需要的东西取消掉，我就保留了adbyby-plus和ssr-plus，编译成功了

我解决了，不需要精简太多，可以一步步试探，只要升级固件控制在8m以内就能成功编译啦！

lxl6125, 2019-12-12:
> > 碰到同样问题，只能生成initramfs-kernel固件，首次编译生成的initramfs-kernel固件大概7.8m左右，刷进去后可以正常使用但重启无法保存配置。使用make clean清除后尝试着精简，编译出来的固件反而达到了9m，依然只能生成initramfs-kernel固件。有没有办法解决呢？
> 
> 已经解决了，办法是删除编译时产生的所有文件夹（包括脚本文件夹和构建文件夹）重新精简后进行编译就能编译成功啦！要省事可以删除目录后重新拉取项目代码，精简后进行编译。

make clean

hong26, 2019-12-12:
> > > 碰到同样问题，只能生成initramfs-kernel固件，首次编译生成的initramfs-kernel固件大概7.8m左右，刷进去后可以正常使用但重启无法保存配置。使用make clean清除后尝试着精简，编译出来的固件反而达到了9m，依然只能生成initramfs-kernel固件。有没有办法解决呢？
> > 
> > 
> > 已经解决了，办法是删除编译时产生的所有文件夹（包括脚本文件夹和构建文件夹）重新精简后进行编译就能编译成功啦！要省事可以删除目录后重新拉取项目代码，精简后进行编译。
> 
> make clean

不知道为什么，我运行make clean后无效，还是自己手动删除比较妥当

closed 2019-12-15 by coolsnowwolf