coolsnowwolf/lede #2390: 编译的时，提示缺少文件。

sunlanhao, 2019-12-11:
[image]

MxCen, 2019-12-11:
git pull最新代码可以解决。

sunlanhao, 2019-12-12:
> git pull最新代码可以解决。

解决啦

txl263, 2019-12-12:
最新代码

```
Installing package 'prometheus-node-exporter-lua' from packages
WARNING: No feed for package 'bmx6' found
WARNING: No feed for package 'bmx6-json' found
WARNING: No feed for package 'bmx7' found
WARNING: No feed for package 'bmx7-json' found
Installing package 'prosody' from packages
Installing package 'psqlodbc' from packages
Installing package 'pv' from packages
Installing package 'pyodbc' from packages
Installing package 'python-appdirs' from packages
Installing package 'python-asn1crypto' from packages
Installing package 'python-awscli' from packages
Installing package 'python-pyasn1' from packages
Installing package 'python-botocore' from packages
Installing package 'python-docutils' from packages
Installing package 'python-jmespath' from packages
Installing package 'python-rsa' from packages
Installing package 'python-colorama' from packages
Installing package 'python-s3transfer' from packages
Installing package 'python-futures' from packages
Installing package 'python-boto3' from packages
Installing package 'python-cachelib' from packages
Installing package 'python-cachetools' from packages
Installing package 'python-contextlib2' from packages
Installing package 'python-crcmod' from packages
Installing package 'python-cryptodome' from packages
Installing package 'python-cryptodomex' from packages
Installing package 'python-curl' from packages
Installing package 'python-decorator' from packages
Installing package 'python-egenix-mx-base' from packages
Installing package 'python-evdev' from packages
Installing package 'python-gmpy2' from packages
Installing package 'python-gnupg' from packages
Installing package 'python-influxdb' from packages
Installing package 'python-intelhex' from packages
Installing package 'python-jsonpath-ng' from packages
Installing package 'python-ldap' from packages
Installing package 'python-markdown' from packages
Installing package 'python-mysql' from packages
Installing package 'python-oauthlib' from packages
Installing package 'python-packages' from packages
Installing package 'python-paho-mqtt' from packages
Installing package 'python-parsley' from packages
Installing package 'python-psycopg2' from packages
Installing package 'python-pyasn1-modules' from packages
Installing package 'python-pyopenssl' from packages
Installing package 'python-pyrsistent' from packages
Installing package 'python-pyserial' from packages
Installing package 'python-qrcode' from packages
Installing package 'python-requests-oauthlib' from packages
Installing package 'python-schema' from packages
Installing package 'python-sentry-sdk' from packages
Installing package 'python-service-identity' from packages
Installing package 'python-simplejson' from packages
Installing package 'python-txsocksx' from packages
Installing package 'python3-bottle' from packages
Installing package 'python3-flask-login' from packages
Installing package 'python3-maxminddb' from packages
Installing package 'python3-netifaces' from packages
Installing package 'python3-packages' from packages
Installing package 'python3-pyroute2' from packages
Installing package 'python3-unidecode' from packages
Installing package 'qemu' from packages
Installing package 'spice-protocol' from packages
Installing package 'spice' from packages
Installing package 'qrencode' from packages
Installing package 'quassel-irssi' from packages
[... 368 lines cut ...]
WARNING: Makefile 'package/feeds/packages/bmx7-dnsupdate/Makefile' has a dependency on 'bmx7-json', which does not exist
WARNING: Makefile 'package/feeds/luci/freifunk-gwcheck/Makefile' has a dependency on 'olsrd-mod-dyn-gw-plain', which does not exist
WARNING: Makefile 'package/feeds/luci/freifunk-mapupdate/Makefile' has a dependency on 'olsrd-mod-nameservice', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-ahcp/Makefile' has a dependency on 'ahcpd', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-olsr/Makefile' has a dependency on 'olsrd', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-olsr/Makefile' has a dependency on 'olsrd-mod-jsoninfo', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-olsr-services/Makefile' has a dependency on 'olsrd', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-olsr-services/Makefile' has a dependency on 'olsrd-mod-nameservice', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-olsr-viz/Makefile' has a dependency on 'olsrd', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-olsr-viz/Makefile' has a dependency on 'olsrd-mod-txtinfo', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-mod-freifunk-community/Makefile' has a dependency on 'olsrd', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-mod-freifunk-community/Makefile' has a dependency on 'olsrd-mod-dyn-gw-plain', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-mod-freifunk-community/Makefile' has a dependency on 'olsrd-mod-jsoninfo', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-mod-freifunk-community/Makefile' has a dependency on 'olsrd-mod-nameservice', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-mod-freifunk-community/Makefile' has a dependency on 'olsrd-mod-watchdog', which does not exist
WARNING: Makefile 'package/feeds/packages/prometheus-node-exporter-lua/Makefile' has a dependency on 'bmx6', which does not exist
WARNING: Makefile 'package/feeds/packages/prometheus-node-exporter-lua/Makefile' has a dependency on 'bmx6-json', which does not exist
WARNING: Makefile 'package/feeds/packages/prometheus-node-exporter-lua/Makefile' has a dependency on 'bmx7', which does not exist
WARNING: Makefile 'package/feeds/packages/prometheus-node-exporter-lua/Makefile' has a dependency on 'bmx7-json', which does not exist
configuration written to .config

*** End of the configuration.
*** Execute 'make' to start the build or try 'make help'.

```

closed 2019-12-20 by coolsnowwolf