coolsnowwolf/lede #2395: 关于版本显示和主机名主机型号修改的问题

bonnyandsky, 2019-12-11:
1：请问固件版本和内核版本为什么一直不变的，并不是R开头的那个版本号
2：请问我要修改主机名和主机型号的话去哪里修改？

bonnyandsky, 2019-12-11:
我看到编译出来的版本号是OpenWrt R9.12.03 。op官方的版本号不是19.0.7 rc2了么。他们之间什么关系？
内核版本现在是4.14.156，然后我发现还有4.9和4.19，有什么区别么？网上搜了下也没找到个所以然

outemancut, 2019-12-11:
版本号你改个20.0.8都可以还不是随便你改的？至于R9.12.03一看就知道是2019年12月03日发布的源码有重大更新所以才以日期做版本号，内核版本除了数字不一样就是稳定不稳定的问题反正内核中加入的功能特性你也没几个用得上 至于4.9跟4.14跟4.19都是长期服务的内核版本 应该还能支持三五年就是了

bonnyandsky, 2019-12-11:
> 版本号你改个20.0.8都可以还不是随便你改的？至于R9.12.03一看就知道是2019年12月03日发布的源码有重大更新所以才以日期做版本号，内核版本除了数字不一样就是稳定不稳定的问题反正内核中加入的功能特性你也没几个用得上 至于4.9跟4.14跟4.19都是长期服务的内核版本 应该还能支持三五年就是了

明白了。感谢回答。所以这个R开头的版本号是L大自定义的和op官方的没什么直接关系

bonnyandsky, 2019-12-11:
请问我要修改主机名和主机型号的话去哪里修改？

eallion, 2019-12-11:
`R9.12.03 `是 dev 版，`19.0.7 rc2`是稳定版。
你用官方的 dev 版自己编译出来，也是显示`R9.12.03`之类的。

outemancut, 2019-12-11:
zzz-default-settings文件58行你随便改就是你要的版本号

bonnyandsky, 2019-12-12:
> zzz-default-settings文件58行你随便改就是你要的版本号

主机名和主机型号呢？哪里改

pepsigold, 2019-12-13:
@bonnyandsky 主机名在系统-基本设置里面改，修改主机名/package/base-files/files/bin下的config_generate中修改

closed 2019-12-15 by coolsnowwolf

Hueysun, 2020-01-26:
请问一下  固件版本信息在哪里修改？