coolsnowwolf/lede #2402: mt7621A 编译报错

brick713, 2019-12-12:
更新了最新的打码还是报错

make[5]: Leaving directory '/home/liyule/lede/target/linux/ramips/image'
Makefile:26: recipe for target 'install' failed
make[4]: *** [install] Error 2
make[4]: Leaving directory '/home/liyule/lede/target/linux/ramips'
Makefile:13: recipe for target 'install' failed
make[3]: *** [install] Error 2
make[3]: Leaving directory '/home/liyule/lede/target/linux'
time: target/linux/install#603.53#22.19#333.01
target/Makefile:23: recipe for target 'target/linux/install' failed
make[2]: *** [target/linux/install] Error 2
make[2]: Leaving directory '/home/liyule/lede'
target/Makefile:19: recipe for target '/home/liyule/lede/staging_dir/target-mipsel_24kc_musl/stamp/.target_install' failed
make[1]: *** [/home/liyule/lede/staging_dir/target-mipsel_24kc_musl/stamp/.target_install] Error 2
make[1]: Leaving directory '/home/liyule/lede'
Build failed - please re-run with -j1 to see the real error message
/home/liyule/lede/include/toplevel.mk:216: recipe for target 'world' failed

wekingchen, 2019-12-12:
Dec 7, 2019 ramips: delete MTK 762x support

可能是上面这条更新的问题，用大雕后面恢复的代码编译会出问题。我的情况是能编译成功，但刷机后无法启动。

目前我的解决办法是，把target\linux\ramips这个文件夹恢复成那个更新之前的版本，编译通过。你可以试试看。

brick713, 2019-12-12:
> Dec 7, 2019 ramips: delete MTK 762x support
> 
> 可能是上面这条更新的问题，用大雕后面恢复的代码编译会出问题。我的情况是能编译成功，但刷机后无法启动。
> 
> 目前我的解决办法是，把target\linux\ramips这个文件夹恢复成那个更新之前的版本，编译通过。你可以试试看。

具体恢复成哪个版本呢？

wekingchen, 2019-12-12:
https://github.com/coolsnowwolf/lede/tree/b0bca4dce08285142d1888e31a5e16f53ba20767

我用的这个版本的ramips覆盖最新的

closed 2019-12-15 by coolsnowwolf