coolsnowwolf/lede #2409: 怎么更改web页面会话超时时间

zjcnew, 2019-12-12:
默认时间我想改成永不超时

pepsigold, 2019-12-13:
`    config core main
            option lang 'zh_cn'
            option mediaurlbase '/luci-static/bootstrap'
            option resourcebase /luci-static/resources
           
    config extern flash_keep
            option uci                 "/etc/config/"
            option dropbear "/etc/dropbear/"
            option openvirtual**        "/etc/openvirtual**/"
            option passwd        "/etc/passwd"
            option opkg                "/etc/opkg.conf"
            option firewall        "/etc/firewall.user"
            option uploads        "/lib/uci/upload/"
           
    config internal languages
           
    config internal sauth
            option sessionpath "/tmp/luci-sessions"
            option sessiontime 3600
           
    config internal ccache
            option enable 1
                   
    config internal themes`
~~~~~~~~~~~~
在这里   option sessiontime 3600

zjcnew, 2019-12-13:
这是哪个文件，在什么位置 /吃惊

zjcnew, 2019-12-13:
找到了 /etc/config/luci，感谢。

garypang13, 2019-12-16:
改了并没有效果,每次打开后台都要登录,

zjcnew, 2019-12-16:
> 改了并没有效果,每次打开后台都要登录,
我重启了整个系统，已经有效果了。

closed 2019-12-24 by coolsnowwolf