coolsnowwolf/lede #2412: 加入luci_app_docker或luci-app-ssr-plus_INCLUDE_Trojan编译失败

i6o6i, 2019-12-12:
我是通过clone friendlyarm的源码并merge lean源码编译的固件
[具体yml](https://github.com/i6o6i/NanoPi-R1S-H5-Openwrt-Autobuild/blob/master/.github/workflows/openwrt-ci.yml)
[.config文件](https://github.com/i6o6i/NanoPi-R1S-H5-Openwrt-Autobuild/blob/master/NanoPi-R1S-H5.config)
前一版本编译成功，[加入docker后的云编译的log在这](https://pipelines.actions.githubusercontent.com/ur1eP8kGaeEVEkoGEyzU20TPxlvkYgfBebpBXmiAKcNTfwdZqJ/_apis/pipelines/1/runs/24/signedlogcontent/3?urlExpires=2019-12-12T14%3A05%3A43.6097242Z&urlSigningMethod=HMACV1&urlSignature=bqVTWJHh9EQopOg%2BK%2BfPFbXqqSgOzc3x%2BjiCUtJFxbc%3D)  [加入Trojan后的log在这](https://pipelines.actions.githubusercontent.com/ur1eP8kGaeEVEkoGEyzU20TPxlvkYgfBebpBXmiAKcNTfwdZqJ/_apis/pipelines/1/runs/25/signedlogcontent/3?urlExpires=2019-12-12T14%3A08%3A20.2005592Z&urlSigningMethod=HMACV1&urlSignature=gTaQL6vvt2x9HALBxiUpX%2BzgcuhytDZMGe6blV9Wu%2Bs%3D)
## Clue
另外编译docker那里的log总是提示
```
WARNING: Makefile 'package/lean/luci-app-docker/Makefile' has a dependency on 'docker-ce'
, which does not exist
```
真的没问题吗？
## 题外话
512M的内存装了这么多插件对Nano Pi R1s是不是有点过分了？

lp123sun, 2019-12-13:
我也是这个提示，我的还有个错误提示：
`/root/openwrt-18.06/build_dir/target-x86_64_musl/ipt2socks/ipt2socks-1.0.2-e6c9b60444bfe2f30830619aacbc67d26ee1015e'
x86_64-openwrt-linux-musl-gcc -std=c99 -Wall -Wextra -O3 -pthread  -s -o ipt2socks ipt2socks.o lrucache.o netutils.o  -l:libuv_a.a
/root/openwrt-18.06/staging_dir/toolchain-x86_64_gcc-7.3.0_musl/lib/gcc/x86_64-openwrt-linux-musl/7.3.0/../../../../x86_64-openwrt-linux-musl/bin/ld: cannot find -l:libuv_a.a
collect2: error: ld returned 1 exit status
make[4]: *** [ipt2socks] Error 1

lp123sun, 2019-12-13:
没有在make menuconfig里面选中luci-app-docker，也会有这个提示：
WARNING: Makefile 'package/lean/luci-app-docker/Makefile' has a dependency on 'docker-ce'
, which does not exist

lp123sun, 2019-12-13:
--   system
--   program_options
CMake Error at /root/openwrt-18.06/staging_dir/host/share/cmake-3.11/Modules/FindPackageHandleStandardArgs.cmake:137 (message):
  Could NOT find OpenSSL, try to set the path to OpenSSL root folder in the
  system variable OPENSSL_ROOT_DIR: Found unsuitable version "1.0.2t", but
  required is at least "1.1.1" (found
  /root/openwrt-18.06/staging_dir/target-x86_64_musl/usr/lib/libcrypto.a)
Call Stack (most recent call first):
  /root/openwrt-18.06/staging_dir/host/share/cmake-3.11/Modules/FindPackageHandleStandardArgs.cmake:376 (_FPHSA_FAILURE_MESSAGE)
  /root/openwrt-18.06/staging_dir/host/share/cmake-3.11/Modules/FindOpenSSL.cmake:379 (find_package_handle_standard_args)
  CMakeLists.txt:46 (find_package)

这2个问题，好像都和trojan的调整有关

coolsnowwolf, 2019-12-13:
make clean

closed 2019-12-14 by i6o6i