coolsnowwolf/lede #2415: OpenWrt 编译 LuCI -> Applications 添加插件应用说明 update -2019.12.8

pepsigold, 2019-12-13:
以下是全部：
-----------------------------------------------------------------------------------------
LuCI ---> Applications ---> luci-app-accesscontrol  #访问时间控制
LuCI ---> Applications ---> luci-app-adblock   #ADB广告过滤
LuCI ---> Applications ---> luci-app-adbyby-plus  #广告屏蔽大师Plus +
LuCI ---> Applications ---> luci-app-adbyby   #广告过滤大师（已弃）
LuCI ---> Applications ---> luci-app-adkill   #广告过滤（已弃）
LuCI ---> Applications ---> luci-app-advanced-reboot  #Linksys高级重启
LuCI ---> Applications ---> luci-app-ahcp  #支持AHCPd
LuCI ---> Applications ---> luci-app-aliddns   #阿里DDNS客户端（已弃，集成至ddns）
LuCI ---> Applications ---> luci-app-amule  #aMule下载工具
LuCI ---> Applications ---> luci-app-aria2 # Aria2下载工具
LuCI ---> Applications ---> luci-app-arpbind  #IP/MAC绑定
LuCI ---> Applications ---> luci-app-asterisk  #支持Asterisk电话服务器
LuCI ---> Applications ---> luci-app-attendedsysupgrade  #固件更新升级相关
LuCI ---> Applications ---> luci-app-autoreboot  #支持计划重启
LuCI ---> Applications ---> luci-app-baidupcs-web  #百度网盘管理   *
LuCI ---> Applications ---> luci-app-bcp38  #BCP38网络入口过滤（不确定）
LuCI ---> Applications ---> luci-app-birdl-ipv4  #对Birdl-ipv6的支持   *
LuCI ---> Applications ---> luci-app-birdl-ipv6  #对Birdl-ipv4的支持   *
LuCI ---> Applications ---> luci-app-bird4   #Bird 4（未知）（已弃）
LuCI ---> Applications ---> luci-app-bird6   #Bird 6（未知）（已弃）
LuCI ---> Applications ---> luci-app-bmx6  #BMX6路由协议
LuCI ---> Applications ---> luci-app-bmx7  #BMX7路由协议
LuCI ---> Applications ---> luci-app-caldav  #联系人（已弃）
LuCI ---> Applications ---> luci-app-cifsd  #网络共享CIFS/SMB服务器   *
LuCI ---> Applications ---> luci-app-cjdns  #加密IPV6网络相关
LuCI ---> Applications ---> luci-app-clamav  #ClamAV杀毒软件
LuCI ---> Applications ---> luci-app-commands   #Shell命令模块
LuCI ---> Applications ---> luci-app-cshark   #CloudShark捕获工具
LuCI ---> Applications ---> luci-app-ddns   #动态域名 DNS（集成阿里DDNS客户端）
LuCI ---> Applications ---> luci-app-diag-core   #core诊断工具
LuCI ---> Applications ---> luci-app-dnscrypt-proxy  #DNSCrypt解决DNS污染
LuCI ---> Applications ---> luci-app-dnsforwarder  #DNSForwarder防DNS污染
LuCI ---> Applications ---> luci-app-dnspod  #DNSPod
LuCI ---> Applications ---> luci-app-docker  #Docker容器   *
LuCI ---> Applications ---> luci-app-dump1090  #民航无线频率（不确定）
LuCI ---> Applications ---> luci-app-dynapoint  #DynaPoint（未知）
LuCI ---> Applications ---> luci-app-e2guardian   #Web内容过滤器
LuCI ---> Applications ---> luci-app-familycloud   #家庭云盘
LuCI ---> Applications ---> luci-app-filetransfer  #文件传输（可web安装ipk包）
LuCI ---> Applications ---> luci-app-firewall   #添加防火墙
LuCI ---> Applications ---> luci-app-flowoffload  #Turbo ACC网络加速（集成FLOW,BBR,NAT,DNS...
LuCI ---> Applications ---> luci-app-freifunk-diagnostics   #freifunk组件 诊断（未知）
LuCI ---> Applications ---> luci-app-freifunk-policyrouting  #freifunk组件 策略路由（未知）
LuCI ---> Applications ---> luci-app-freifunk-widgets  #freifunk组件 索引（未知）
LuCI ---> Applications ---> luci-app-frpc   #内网穿透 Frp
LuCI ---> Applications ---> luci-app-fwknopd  #Firewall Knock Operator服务器
LuCI ---> Applications ---> luci-app-guest-wifi   #WiFi访客网络
LuCI ---> Applications ---> luci-app-gfwlist   #GFW域名列表（已弃）
LuCI ---> Applications ---> luci-app-haproxy-tcp   #HAProxy负载均衡-TCP
LuCI ---> Applications ---> luci-app-hd-idle  #硬盘休眠
LuCI ---> Applications ---> luci-app-hnet  #Homenet Status家庭网络控制协议
LuCI ---> Applications ---> luci-app-ipsec-virtual**d  #virtual**服务器 IPSec
LuCI ---> Applications ---> luci-app-kodexplorer  #KOD可道云私人网盘
LuCI ---> Applications ---> luci-app-kooldns  #virtual**服务器 ddns替代方案（已弃）
LuCI ---> Applications ---> luci-app-koolproxy  #KP去广告（已弃）
LuCI ---> Applications ---> luci-app-lxc   #LXC容器管理
LuCI ---> Applications ---> luci-app-meshwizard #网络设置向导
LuCI ---> Applications ---> luci-app-minidlna   #完全兼容DLNA / UPnP-AV客户端的服务器软件
LuCI ---> Applications ---> luci-app-mjpg-streamer   #兼容Linux-UVC的摄像头程序
LuCI ---> Applications ---> luci-app-mtwifi  #MTWiFi驱动的支持   *
LuCI ---> Applications ---> luci-app-mmc-over-gpio   #添加SD卡操作界面（已弃）
LuCI ---> Applications ---> luci-app-multiwan   #多拨虚拟网卡（已弃，移至syncdial）
LuCI ---> Applications ---> luci-app-mwan   #MWAN负载均衡（已弃）
LuCI ---> Applications ---> luci-app-mwan3   #MWAN3负载均衡
LuCI ---> Applications ---> luci-app-mwan3helper   #MWAN3分流助手
LuCI ---> Applications ---> luci-app-n2n_v2   #N2N内网穿透 N2N v2 virtual**服务
LuCI ---> Applications ---> luci-app-netdata  #Netdata实时监控（图表）   *
LuCI ---> Applications ---> luci-app-nft-qos  #QOS流控 Nftables版
[... 62 lines cut ...]
LuCI ---> Applications ---> luci-app-vnstat   #vnStat网络监控（图表）
LuCI ---> Applications ---> luci-app-virtual**bypass  #virtual** BypassWebUI  绕过virtual**设置
LuCI ---> Applications ---> luci-app-vsftpd  #FTP服务器
LuCI ---> Applications ---> luci-app-watchcat  #断网检测功能与定时重启
LuCI ---> Applications ---> luci-app-webadmin  #Web管理页面设置
LuCI ---> Applications ---> luci-app-webshell  #网页命令行终端（已弃）
LuCI ---> Applications ---> luci-app-wifischedule  #WiFi 计划
LuCI ---> Applications ---> luci-app-wireguard  #virtual**服务器 WireGuard状态
LuCI ---> Applications ---> luci-app-wirele违禁软件egdb  #WiFi无线
LuCI ---> Applications ---> luci-app-wol   #WOL网络唤醒
LuCI ---> Applications ---> luci-app-wrtbwmon  #实时流量监测
LuCI ---> Applications ---> luci-app-xlnetacc  #迅雷快鸟
LuCI ---> Applications ---> luci-app-zerotier  #ZeroTier内网穿透
---------------------------------------------------------------------------------------------------
支持 iPv6：
1、Extra packages  --->  ipv6helper  （选定这个后下面几项自动选择了）
Network  --->  odhcp6c
Network  --->  odhcpd-ipv6only
LuCI  --->  Protocols  --->  luci-proto-ipv6
LuCI  --->  Protocols  --->  luci-proto-ppp

2、打开适用于VMware的VM Tools
Utilities  --->  open-vm-tools

该内容为转载，原帖地址：[请点我](https://www.right.com.cn/forum/thread-344825-1-1.html)

eallion, 2019-12-15:
好是好，可是能不能改下格式啊，h2 看着眼睛累。。。

MxCen, 2019-12-15:
好是好，可是能不能改下格式啊，h2 看着眼睛累。。。 +1

KFERMercer, 2019-12-15:
|  建议   | 使用  |
|  ----  | ----  |
| Markdown  | 表格 |
| 重制  | 下. |

eallion, 2019-12-16:
### 为方便新手编译及学习，特整理一下 applications 相关应用说明。
> 以下是全部：

|APP|名称|释义|
|---|---|---|
|LuCI ---> Applications | luci-app-accesscontrol  |# 访问时间控制	|
|LuCI ---> Applications | luci-app-adblock   |#ADB 广告过滤	|
|LuCI ---> Applications | luci-app-adbyby-plus  |# 广告屏蔽大师 Plus +	|
|LuCI ---> Applications | luci-app-adbyby   |# 广告过滤大师（已弃）	|
|LuCI ---> Applications | luci-app-adkill   |# 广告过滤（已弃）	|
|LuCI ---> Applications | luci-app-advanced-reboot  |#Linksys 高级重启	|
|LuCI ---> Applications | luci-app-ahcp  |# 支持 AHCPd	|
|LuCI ---> Applications | luci-app-aliddns   |# 阿里 DDNS 客户端（已弃，集成至 ddns）	|
|LuCI ---> Applications | luci-app-amule  |#aMule 下载工具	|
|LuCI ---> Applications | luci-app-aria2 |# Aria2 下载工具	|
|LuCI ---> Applications | luci-app-arpbind  |#IP/MAC 绑定	|
|LuCI ---> Applications | luci-app-asterisk  |# 支持 Asterisk 电话服务器	|
|LuCI ---> Applications | luci-app-attendedsysupgrade  |# 固件更新升级相关	|
|LuCI ---> Applications | luci-app-autoreboot  |# 支持计划重启	|
|LuCI ---> Applications | luci-app-baidupcs-web  |# 百度网盘管理   *	|
|LuCI ---> Applications | luci-app-bcp38  |#BCP38 网络入口过滤（不确定）	|
|LuCI ---> Applications | luci-app-birdl-ipv4  |# 对 Birdl-ipv6 的支持   *	|
|LuCI ---> Applications | luci-app-birdl-ipv6  |# 对 Birdl-ipv4 的支持   *	|
|LuCI ---> Applications | luci-app-bird4   |#Bird 4（未知）（已弃）	|
|LuCI ---> Applications | luci-app-bird6   |#Bird 6（未知）（已弃）	|
|LuCI ---> Applications | luci-app-bmx6  |#BMX6 路由协议	|
|LuCI ---> Applications | luci-app-bmx7  |#BMX7 路由协议	|
|LuCI ---> Applications | luci-app-caldav  |# 联系人（已弃）	|
|LuCI ---> Applications | luci-app-cifsd  |# 网络共享 CIFS/SMB 服务器   *	|
|LuCI ---> Applications | luci-app-cjdns  |# 加密 IPV6 网络相关	|
|LuCI ---> Applications | luci-app-clamav  |#ClamAV 杀毒软件	|
|LuCI ---> Applications | luci-app-commands   |#Shell 命令模块	|
|LuCI ---> Applications | luci-app-cshark   |#CloudShark 捕获工具	|
|LuCI ---> Applications | luci-app-ddns   |# 动态域名 DNS（集成阿里 DDNS 客户端）	|
|LuCI ---> Applications | luci-app-diag-core   |#core 诊断工具	|
|LuCI ---> Applications | luci-app-dnscrypt-proxy  |#DNSCrypt 解决 DNS 污染	|
|LuCI ---> Applications | luci-app-dnsforwarder  |#DNSForwarder 防 DNS 污染	|
|LuCI ---> Applications | luci-app-dnspod  |#DNSPod	|
|LuCI ---> Applications | luci-app-docker  |#Docker 容器   *	|
|LuCI ---> Applications | luci-app-dump1090  |# 民航无线频率（不确定）	|
|LuCI ---> Applications | luci-app-dynapoint  |#DynaPoint（未知）	|
|LuCI ---> Applications | luci-app-e2guardian   |#Web 内容过滤器	|
|LuCI ---> Applications | luci-app-familycloud   |# 家庭云盘	|
|LuCI ---> Applications | luci-app-filetransfer  |# 文件传输（可 web 安装 ipk 包）	|
|LuCI ---> Applications | luci-app-firewall   |# 添加防火墙	|
|LuCI ---> Applications | luci-app-flowoffload  |#Turbo ACC 网络加速（集成 FLOW,BBR,NAT,DNS...	|
|LuCI ---> Applications | luci-app-freifunk-diagnostics   |#freifunk 组件 诊断（未知）	|
|LuCI ---> Applications | luci-app-freifunk-policyrouting  |#freifunk 组件 策略路由（未知）	|
|LuCI ---> Applications | luci-app-freifunk-widgets  |#freifunk 组件 索引（未知）	|
|LuCI ---> Applications | luci-app-frpc   |# 内网穿透 Frp	|
|LuCI ---> Applications | luci-app-fwknopd  |#Firewall Knock Operator 服务器	|
|LuCI ---> Applications | luci-app-guest-wifi   |#WiFi 访客网络	|
|LuCI ---> Applications | luci-app-gfwlist   |#GFW 域名列表（已弃）	|
|LuCI ---> Applications | luci-app-haproxy-tcp   |#HAProxy 负载均衡 - TCP	|
|LuCI ---> Applications | luci-app-hd-idle  |# 硬盘休眠	|
|LuCI ---> Applications | luci-app-hnet  |#Homenet Status 家庭网络控制协议	|
|LuCI ---> Applications | luci-app-ipsec-virtual**d  |#virtual** 服务器 IPSec	|
|LuCI ---> Applications | luci-app-kodexplorer  |#KOD 可道云私人网盘	|
|LuCI ---> Applications | luci-app-kooldns  |#virtual** 服务器 ddns 替代方案（已弃）	|
|LuCI ---> Applications | luci-app-koolproxy  |#KP 去广告（已弃）	|
|LuCI ---> Applications | luci-app-lxc   |#LXC 容器管理	|
|LuCI ---> Applications | luci-app-meshwizard |#网络设置向导	|
|LuCI ---> Applications | luci-app-minidlna   |# 完全兼容 DLNA / UPnP-AV 客户端的服务器软件	|
|LuCI ---> Applications | luci-app-mjpg-streamer   |# 兼容 Linux-UVC 的摄像头程序	|
|LuCI ---> Applications | luci-app-mtwifi  |#MTWiFi 驱动的支持   *	|
|LuCI ---> Applications | luci-app-mmc-over-gpio   |# 添加 SD 卡操作界面（已弃）	|
|LuCI ---> Applications | luci-app-multiwan   |# 多拨虚拟网卡（已弃，移至 syncdial）	|
|LuCI ---> Applications | luci-app-mwan   |#MWAN 负载均衡（已弃）	|
|LuCI ---> Applications | luci-app-mwan3   |#MWAN3 负载均衡	|
|LuCI ---> Applications | luci-app-mwan3helper   |#MWAN3 分流助手	|
[... 63 lines cut ...]
|LuCI ---> Applications | luci-app-vnstat   |#vnStat 网络监控（图表）	|
|LuCI ---> Applications | luci-app-virtual**bypass  |#virtual** BypassWebUI  绕过 virtual** 设置	|
|LuCI ---> Applications | luci-app-vsftpd  |#FTP 服务器	|
|LuCI ---> Applications | luci-app-watchcat  |# 断网检测功能与定时重启	|
|LuCI ---> Applications | luci-app-webadmin  |#Web 管理页面设置	|
|LuCI ---> Applications | luci-app-webshell  |# 网页命令行终端（已弃）	|
|LuCI ---> Applications | luci-app-wifischedule  |#WiFi 计划	|
|LuCI ---> Applications | luci-app-wireguard  |#virtual** 服务器 WireGuard 状态	|
|LuCI ---> Applications | luci-app-wirele 违禁软件 egdb  |#WiFi 无线	|
|LuCI ---> Applications | luci-app-wol   |#WOL 网络唤醒	|
|LuCI ---> Applications | luci-app-wrtbwmon  |# 实时流量监测	|
|LuCI ---> Applications | luci-app-xlnetacc  |# 迅雷快鸟	|
|LuCI ---> Applications | luci-app-zerotier  |#ZeroTier 内网穿透	|

支持 iPv6：
1、`Extra packages ---> ipv6helper` （选定这个后下面几项自动选择了）
```
Network ---> odhcp6c
Network ---> odhcpd-ipv6only
LuCI ---> Protocols ---> luci-proto-ipv6
LuCI ---> Protocols ---> luci-proto-ppp
```

2、打开适用于 VMware 的 VM Tools
`Utilities` --->` open-vm-tools`

mrclock, 2019-12-20:
luci-app-meshwizard 这个是怎么玩的呢，我装了这个，从来没在界面里找到过。

closed 2019-12-24 by coolsnowwolf

segadrive, 2020-01-22:
在几位大神的教程下编译完成了，再次谢谢各位大佬

mgsqz, 2020-01-27:
这么好的东西应该写进wiki，强烈建议

liasica, 2020-02-21:
感谢分享

lnjoph, 2020-03-15:
好人一生平安！

shu0802, 2020-04-07:
很遗憾没有与L大同步更新

danshui-git, 2020-07-28:
请问大神们，如果我要把插件改分类，如何改呢？比如luci-app-v2 瑞编译出来默认在V/P/N项上的，我把它改到服务项上可以吗？

HUiTHEME, 2021-02-16:
luci-app-nlbwmon  和  luci-app-wrtbwmon  什么区别？

yoyojacky, 2021-09-01:
请问，这个https://github.com/NateLol/luci-app-oled 怎么添加进去？