coolsnowwolf/lede #2416: 最新的代码编译不通过

GreatGarlic, 2019-12-13:
[image]

sitieqiang, 2019-12-13:
我也遇到了相同的问题 cryptodev 模块中的  ioctl.c 的sys_close 改成ksys_close就可以了  但是请大神来看看  是不是cryptodev 选取的版本应该是过旧 新linux内核已经把sys_close 改成了ksys_close

bonnyandsky, 2019-12-13:
是不是这一步出问题了？
我想我脚本也没改过呀。突然就不行了。是什么问题哦
[image]

bonnyandsky, 2019-12-13:
> sys_close

你说的这文件在哪里？

GreatGarlic, 2019-12-13:
@bonnyandsky 看起来不像是1225行这种问题，因为编译通过的里面也有这个警告

bonnyandsky, 2019-12-13:
> @bonnyandsky 看起来不像是1225行这种问题，因为编译通过的里面也有这个警告
我是说我脚本什么都没改，然后过了2天编译就到第三部就出问题了。搞得我以为那里除问题。重新fork后什么都不改还是编译不通过

GreatGarlic, 2019-12-13:
@bonnyandsky 感觉是源出了问题

sitieqiang, 2019-12-13:
我是再ubuntu19.04上编译的 这上面编译会把这个警告变成错误 现在大神的代码使用的是1.09  再新的1.10版本里已经被修改了内核版本超过4.17的已经被修改了
		if (unlikely(ret)) {
#if (LINUX_VERSION_CODE < KERNEL_VERSION(4, 17, 0))
			sys_close(fd);
#else
			ksys_close(fd);
#endif
			return ret;
		}

sheji-sm, 2019-12-13:
块放弃了   这个编译有同样问题。一直卡着不过

EnnawYang, 2019-12-13:
...... 微软的问题你们瞎改啥呢

GreatGarlic, 2019-12-13:
@EnnawYang 好吧，那就等微软的源修复吧。。。

MxCen, 2019-12-13:
看了下，自己的可以编译呀。

GreatGarlic, 2019-12-13:
@MxCen 大佬你在哪看的。。

zeXtreme, 2019-12-13:
引入P3TERX 的 commit [**Fix bug caused by Microsoft baba.**](https://github.com/P3TERX/Actions-OpenWrt/commit/fcaafa745f37e57176c75f8051c96724d8b76514) 可以解决

GreatGarlic, 2019-12-13:
@zeXtreme 大佬牛逼😀

daydyup, 2019-12-13:
@GreatGarlic 问题解决了吗？

GreatGarlic, 2019-12-13:
@daydyup 还没编译完呢

sheji-sm, 2019-12-13:
咋改的   能共享一下代码不？

GreatGarlic, 2019-12-13:
@sheji-sm 就fork一下L大的代码到自己仓库呀

sheji-sm, 2019-12-13:
@GreatGarlic   奇怪复制了下你的就开始编译了   哈哈哈

GreatGarlic, 2019-12-13:
@sheji-sm 一推送代码github会检测到push事件

bonnyandsky, 2019-12-13:
解决了，好用

closed 2019-12-15 by coolsnowwolf