coolsnowwolf/lede #2440: 上网时间控制插件，只要开启所有设置不能上网。

snakwu, 2019-12-15:
上网时间控制插件，只要开启所有设置不能上网。什么cpu模式也开了，我就连列表不加mac，只要打开上网时间控制，就所有设备上不了网！

coolsnowwolf, 2019-12-15:
并不会 已测试

snakwu, 2019-12-15:
那搞不明白是什么问题，也无日志显示，折腾了一天，对了我网络有v2ray，有双wan，有ipv6，只要打开上网控制开关，所有设备无法上网，但能访问openwrt页面

snakwu, 2019-12-15:
同时也开了dbs转发

outemancut, 2019-12-15:
并没有问题！好用的很

snakwu, 2019-12-15:
我也不明原因

snakwu, 2019-12-15:
对了，我是软路由，虚拟机环境，还接有交换机，有划vlan

coolsnowwolf, 2019-12-16:
不能重现

closed 2019-12-16 by coolsnowwolf

snakwu, 2019-12-16:
是否我编译时少了什么依赖或iptables mod

snakwu, 2019-12-18:
L大,问题是在  iptables -t nat -A PREROUTING -p udp --dport 53 -j REDIRECT --to-ports 53 -m comment --comment "Rule For Control"
  iptables -t nat -A PREROUTING -p tcp --dport 53 -j REDIRECT --to-ports 53 -m comment --comment "Rule For Control"
这两行iptables上面,只要这两个一添加,全部设备不能上网!

chisenin, 2020-07-25:
今天也遇到同样的问题，开了上网时间控制插件，电脑无法获得IP,但是手机可以，控制的手机照样上网，其它设备则不能，手工设置电脑IP,可以访问openwrt设置页面，但无法上网。之前一直没有用过这个插件，一直没有问题。固件是今天编译的，也不知道是否有其它的编译依赖没解决

> 

主机名 | OpenWrt
-- | --
主机型号 | Intel(R) Celeron(R) CPU N3150 @ 1.60GHz : 4 Core 4 Thread
CPU 信息 | 1600.072 MHz +45.0°C (high = +90.0°C, crit = +90.0°C)
固件版本 | OpenWrt R20.7.1 / LuCI Master (git-20.191.36863-eee6bae)
内核版本 | 4.19.123
本地时间 | Sat Jul 25 21:27:52 2020
运行时间 | 0h 37m 21s
平均负载 | 0.64, 0.24, 0.09
CPU 使用率（%） | 3 %

config.seed文件内容

CONFIG_TARGET_x86=y
CONFIG_TARGET_x86_64=y
CONFIG_TARGET_x86_64_Generic=y

# CONFIG_EFI_IMAGES is not set

CONFIG_LIBCURL_COOKIES=y
CONFIG_LIBCURL_FILE=y
CONFIG_LIBCURL_FTP=y
CONFIG_LIBCURL_HTTP=y
CONFIG_LIBCURL_NO_SMB="!"
CONFIG_LIBCURL_OPENSSL=y
CONFIG_LIBCURL_PROXY=y
CONFIG_NGINX_HEADERS_MORE=y
CONFIG_NGINX_HTTP_ACCESS=y
CONFIG_NGINX_HTTP_AUTH_BASIC=y
CONFIG_NGINX_HTTP_AUTOINDEX=y
CONFIG_NGINX_HTTP_BROWSER=y
CONFIG_NGINX_HTTP_CACHE=y
CONFIG_NGINX_HTTP_CHARSET=y
CONFIG_NGINX_HTTP_EMPTY_GIF=y
CONFIG_NGINX_HTTP_FASTCGI=y
CONFIG_NGINX_HTTP_GEO=y
CONFIG_NGINX_HTTP_GZIP=y
CONFIG_NGINX_HTTP_LIMIT_CONN=y
CONFIG_NGINX_HTTP_LIMIT_REQ=y
CONFIG_NGINX_HTTP_MAP=y
CONFIG_NGINX_HTTP_MEMCACHED=y
CONFIG_NGINX_HTTP_PROXY=y
CONFIG_NGINX_HTTP_REFERER=y
CONFIG_NGINX_HTTP_REWRITE=y
CONFIG_NGINX_HTTP_SCGI=y
CONFIG_NGINX_HTTP_SPLIT_CLIENTS=y
CONFIG_NGINX_HTTP_SSI=y
CONFIG_NGINX_HTTP_UPSTREAM_HASH=y
CONFIG_NGINX_HTTP_UPSTREAM_IP_HASH=y
CONFIG_NGINX_HTTP_UPSTREAM_KEEPALIVE=y
CONFIG_NGINX_HTTP_UPSTREAM_LEAST_CONN=y
CONFIG_NGINX_HTTP_USERID=y
CONFIG_NGINX_HTTP_UWSGI=y
CONFIG_NGINX_NAXSI=y
CONFIG_NGINX_PCRE=y
CONFIG_OPENSSL_WITH_COMPRESSION=y
CONFIG_OPENSSL_WITH_DTLS=y
CONFIG_OPENSSL_WITH_EC2M=y
CONFIG_OPENSSL_WITH_NPN=y
CONFIG_PACKAGE_6in4=y
CONFIG_PACKAGE_amd64-microcode=y

# CONFIG_PACKAGE_ath10k-firmware-qca9888 is not set

# CONFIG_PACKAGE_ath10k-firmware-qca988x is not set

[... 195 lines cut ...]
CONFIG_PACKAGE_strongswan-mod-pubkey=y
CONFIG_PACKAGE_strongswan-mod-random=y
CONFIG_PACKAGE_strongswan-mod-sha1=y
CONFIG_PACKAGE_strongswan-mod-socket-default=y
CONFIG_PACKAGE_strongswan-mod-stroke=y
CONFIG_PACKAGE_strongswan-mod-updown=y
CONFIG_PACKAGE_strongswan-mod-x509=y
CONFIG_PACKAGE_strongswan-mod-xauth-generic=y
CONFIG_PACKAGE_strongswan-mod-xcbc=y
CONFIG_PACKAGE_tcpping=y
CONFIG_PACKAGE_wol=y
CONFIG_PACKAGE_zerotier=y

# CONFIG_PCAP_HAS_NETFILTER is not set

# CONFIG_PCAP_HAS_USB is not set

# CONFIG_SIMPLE_OBFS_STATIC_LINK is not set

CONFIG_STRONGSWAN_ROUTING_TABLE="220"
CONFIG_STRONGSWAN_ROUTING_TABLE_PRIO="220"

# CONFIG_ZEROTIER_ENABLE_DEBUG is not set

# CONFIG_ZEROTIER_ENABLE_SELFTEST is not set

chisenin, 2020-07-26:
昨天我以为是控制上网时间插件的问题，今天我把那插件卸载了，用smartdns和passwall时，一样出现在这个问题。初步估计是这两个插件配合有问题。或者是这两个插件和L大的源码兼容性问题。单独用smartdns/passwall,初次设置之后正常，但是只要修改了设置，保存&应用之后就一定会出现这个问题。后面没有再测试下去，折腾的时间太久了。