coolsnowwolf/lede #2446: 自编译的lede，DHCP问题导致重启后无网络。

MRDHR, 2019-12-15:
问题描述：编译的LEDE，安装后的第一次启动并配置完网络无问题。重启软路由后DHCP分配的默认网关和dns服务器会变成本网段的127

拿我本地环境的举例：
软路由 关闭ipv6 dhcp
lan 静态ip：192.168.2.1
第一次启动后获取的网络信息
ip 192.168.2.2
子网掩码 255.255.255.0
默认网关 192.168.2.1
dns服务器 192.168.2.1

重启路由器后获取到的网络信息
ip 192.168.2.2
子网掩码 255.255.255.0
默认网关 192.168.2.127
dns服务器 192.168.2.127

编译时 只增加了organ主题，其他均为默认配置。
试过openwrt官网的版本，只要将dnsmasq替换为dnsmasq-full也会出现相同的问题。网上也搜索过，暂时未找到答案。请大佬解疑。

coolsnowwolf, 2019-12-15:
同一段有其他 DHCP服务器

coolsnowwolf, 2019-12-15:
例如vmware 的

MRDHR, 2019-12-15:
我的电脑，有virtual box，但是虚拟机均未启动并且没有dhcp服务器。遇到这种情况，可以在路由器强行配置dns服务器和默认网关么？

MRDHR, 2019-12-15:
[image]
我尝试下改成这样，下次重启后再来报告情况。

closed 2019-12-16 by coolsnowwolf

MRDHR, 2020-01-14:
通过修改lan的dhcp 勾选强制选项，成功解决了该问题。