coolsnowwolf/lede #2453: 64M内存的MT7621机型出现的内存问题

keke1023, 2019-12-16:
机型是中兴的E8820V2，7621+64M内存，内存确实有点小，但是其他64M内存的机型貌似也没有遇到这个问题
具体症状是负载突然升高，高到5-10都有可能，查看进程主要是kworker，而且可能在一段时间后恢复
使用的功能测试过不同的搭配，多拨+ss（很少出现），多拨+trojan（经常出现），纯trojan（较少出现），纯ss（较少出现）
因为情况出现的不是很规律，所以不清楚具体触发的原因，猜测可能是内存不够导致的，刚运行起来貌似都好好的，负载也很低，内存看起来也剩20多M，但是不知道什么时候就突然卡住了，ssh上去看负载超高
最后贴一下关于这部分的内核日志，烦请各位帮我判断一下问题是出在机子自身内存不够，还是内核问题，还是trojan或者ss问题？前面是13秒正常启动日志，然后就到了6000秒的异常状态
[ 6203.346150] swapper/0: page allocation failure: order:0, mode:0x1080020(GFP_ATOMIC), nodemask=(null)
[ 6203.346605] swapper/3: page allocation failure: order:0
[ 6203.355296] swapper/0 cpuset=/
[ 6203.360513] , mode:0x1080020(GFP_ATOMIC), nodemask=
[ 6203.360534]  mems_allowed=0
[ 6203.363572] (null)
[ 6203.368440] CPU: 0 PID: 0 Comm: swapper/0 Not tainted 4.14.158 #0
[ 6203.371212] swapper/3 cpuset=
[ 6203.373205] Stack :
[ 6203.379271] /
[ 6203.382211]  00000000
[ 6203.384293]  mems_allowed=0
[ 6203.385852]  00000000 01080020 80071de8 805f0000 80588da4 00000000 00000000
[ 6203.397815]         805531c8 83809cb4 805c2e2c 805c29a7 8054dfd0 00000001 83809c58 ac07f587
[ 6203.406147]         00000000 00000000 80740000 00000124 00000029 00000003 00000001 6c756e28
[ 6203.414478]         83809a9c 805d0000 00058c34 00000000 00000000 805f0000 00000000 00000000
[ 6203.422808]         80e6d1c0 805c01c8 01080020 00000000 00000001 802ce230 00000000 80730000
[ 6203.431145]         ...
[ 6203.433585] Call Trace:
[ 6203.436047] [<8000c700>] show_stack+0x58/0x100
[ 6203.440481] [<80489734>] dump_stack+0xa4/0xe0
[ 6203.444836] [<800dcc54>] warn_alloc+0xcc/0x15c
[ 6203.449261] [<800dd988>] __alloc_pages_nodemask+0xbf8/0xd78
[ 6203.454807] [<800ddd4c>] page_frag_alloc+0x14c/0x178
[ 6203.459753] [<8033b5f0>] fe_poll+0x33c/0x804
[ 6203.464010] [<8037ec74>] net_rx_action+0x150/0x30c
[ 6203.468791] [<804a8620>] __do_softirq+0x128/0x2ec
[ 6203.473485] [<8002fc64>] irq_exit+0xac/0xc8
[ 6203.477654] [<802817ec>] plat_irq_dispatch+0xfc/0x138
[ 6203.482682] [<80007608>] except_vec_vi_end+0xb8/0xc4
[ 6203.487623] [<80008fd0>] r4k_wait_irqoff+0x1c/0x24
[ 6203.492407] CPU: 3 PID: 0 Comm: swapper/3 Not tainted 4.14.158 #0
[ 6203.492496] Mem-Info:
[ 6203.498476] Stack : 00000000 00000000 01080020 80071de8 805f0000 80588da4 00000000
[ 6203.500781] active_anon:2369 inactive_anon:58 isolated_anon:0
[ 6203.500781]  active_file:821 inactive_file:1600 isolated_file:12
[ 6203.500781]  unevictable:0 dirty:0 writeback:0 unstable:0
[ 6203.500781]  slab_reclaimable:490 slab_unreclaimable:2791
[ 6203.500781]  mapped:1461 shmem:342 pagetables:115 bounce:0
[ 6203.500781]  free:766 free_pcp:42 free_cma:0
[ 6203.500788]  00000000
[ 6203.508357] Node 0 active_anon:9476kB inactive_anon:232kB active_file:3284kB inactive_file:6400kB unevictable:0kB isolated(anon):0kB isolated(file):48kB mapped:5844kB dirty:0kB writeback:0kB shmem:1368kB writeback_tmp:0kB unstable:0kB all_unreclaimable? no
[ 6203.540403]         805531c8 8380fc94 8383518c 805c29a7 8054dfd0 00000001
[ 6203.542708] Normal free:3064kB min:8192kB low:10240kB high:12288kB active_anon:9476kB inactive_anon:232kB active_file:3284kB inactive_file:6400kB unevictable:0kB writepending:0kB present:65536kB managed:58256kB mlocked:0kB kernel_stack:720kB pagetables:460kB bounce:0kB free_pcp:168kB local_pcp:12kB free_cma:0kB
[ 6203.565246]  8380fc38 ac07f587
[ 6203.565258]         00000000 00000000
[ 6203.572029] lowmem_reserve[]:
[ 6203.599421]  80740000 00000124 30232038 6465746e 312e3420 35312e34
[ 6203.602482]  0
[ 6203.606117]         8380fa7c 805d0000 00078377 00000000 00000000 805f0000
[ 6203.609094]  0
[ 6203.615234]  00000000
[ 6203.616896]  0
[ 6203.623634]  00000000
[ 6203.627535]         00000000 805c01c8 01080020 00000000 00000000
[ 6203.629211] Normal: 
[ 6203.631464]  802ce230
[ 6203.637449] 9*4kB 
[ 6203.639602]  0000000c
[ 6203.641856] (UM) 
[ 6203.643850]  8073000c
[ 6203.646123] 0*8kB 
[ 6203.650268] 2*16kB 
[ 6203.652259]        
[ 6203.652267] (ME) 
[... 350 lines cut ...]
[ 6232.524465] [ 1655]     0  1655      394       32       4       0        0             0 uhttpd
[ 6232.533964] [ 2370]     0  2370      443       34       4       0        0             0 hostapd
[ 6232.542824] [ 2396]     0  2396      443       31       4       0        0             0 hostapd
[ 6232.551680] [ 2813]     0  2813      313       31       5       0        0             0 pppd
[ 6232.560285] [ 2824]     0  2824      313       32       4       0        0             0 pppd
[ 6232.568843] [ 2827]     0  2827      313       32       4       0        0             0 pppd
[ 6232.577386] [ 3424]     0  3424      305       11       3       0        0             0 ntpd
[ 6232.585919] [ 3973]     0  3973      329       36       4       0        0             0 sh
[ 6232.594261] [ 4322]     0  4322      329       36       3       0        0             0 sh
[ 6232.602651] [ 4703]     0  4703      329       36       5       0        0             0 sh
[ 6232.611033] [18607]     0 18607      305        9       3       0        0             0 crond
[ 6232.619673] [18758]     0 18758     2462      918       7       0        0             0 trojan
[ 6232.628403] [18794] 65534 18794      381        0       4       0        0             0 pdnsd
[ 6232.639174] [18809]     0 18809      324       30       4       0        0             0 sh
[ 6232.650049] [18914]   453 18914      852      538       5       0        0             0 dnsmasq
[ 6232.661336] [18926]     0 18926      307       12       5       0        0             0 sh
[ 6232.671084] [24795]     0 24795      335       41       3       0        0             0 sh
[ 6232.679480] [24799]     0 24799      315       20       4       0        0             0 sh
[ 6232.687852] [24829]     0 24829      304        7       4       0        0             0 sleep
[ 6232.696482] [24844]     0 24844      306        9       4       0        0             0 sleep
[ 6232.705075] [24849]     0 24849      306        9       3       0        0             0 sleep
[ 6232.713748] [24850]     0 24850      306        9       4       0        0             0 sleep
[ 6232.722386] Out of memory: Kill process 18758 (trojan) score 61 or sacrifice child
[ 6232.730018] Killed process 18758 (trojan) total-vm:9848kB, anon-rss:3668kB, file-rss:0kB, shmem-rss:4kB
[ 6232.760392] oom_reaper: reaped process 18758 (trojan), now anon-rss:0kB, file-rss:0kB, shmem-rss:0kB

coolsnowwolf, 2019-12-16:
内存被无线和 Trojan 耗尽了，编译时去掉无线，MTK 的驱动虽然很差，但吃内存比 AR71xx 和 高通的 多多了

keke1023, 2019-12-16:
@coolsnowwolf 谢谢大佬亲自解答，那我就还是改下内存再测试，猜测也是内存不够，但是没想到是无线驱动在助攻。。

closed 2019-12-17 by coolsnowwolf