coolsnowwolf/lede #2458: CPU调速降频过于激进导致网络抖动增加

WWWWindGG, 2019-12-16:
### 路由器型号
        竞斗云2(R619AC)
### 遇到问题
路由器使用过程中，发现路由器在较小负载环境下，与路由器有线连接的电脑ping路由器网关地址时延时有较大抖动，具体在日常使用当中体现为游戏中的抖动、小范围跳ping等现象，对日常上网没有明显影响。
### 验证
验证发现路由器在较高负载时ping值和抖动均降低，因此得出结论抖动并非硬件性能导致。编译netdata进固件检查路由器运行情况时发现，路由器在低负载条件下cpu频率被控制在100Mhz以下，运行游戏等的负载量不足以让cpu以更高频率工作，此时ping 出现高延迟抖动现象。
[image]
**_大部分时间下cpu运行在100Mhz以下_**
[image]
**_出现较高延迟与抖动_**
### 临时解决方案
修改cpu调速策略至performance
[image]
此时cpu频率被锁定在716Mhz
[image]
ping即可恢复正常
[image]
### 请求添加功能/修改ondemand mode的最低CPU百分比
修改scaling_governor文件可以可以解决内网到路由器的抖动问题，但cpu长期工作在最高频率容易因为散热产生问题，所以请lean大能在luci里添加手动选择cpu调速模式、设置最小cpu、最大cpu的插件，并且提高默认条件下的最小cpu比值，感谢！

coolsnowwolf, 2019-12-16:
很高兴有人注意到这一点

因为 IPQ40xx 采用 28nm 制作，如果不控制频率，将会带来发热的问题，所以默认我采用了保守的节能策略

/etc/init.d/linksys_recovery 中可以调整这两个数值

echo 35 > /sys/devices/system/cpu/cpufreq/ondemand/up_threshold
echo 10 > /sys/devices/system/cpu/cpufreq/ondemand/sampling_down_factor

coolsnowwolf, 2019-12-16:
你也可以测试下，看哪个参数是最适合的，echo 后面的数字

WWWWindGG, 2019-12-16:
我测试了几次，发现适当调小up_threshold并且调大sampling_down_factor的值可以缓解，但并不能完全解决，而且当up_threshold调至15一下时，ping出现了超过10ms的长延迟，这可能是由于过于频繁触发调速程序导致中断过多导致的。

另外，我通过
`echo performance > /sys/devices/system/cpu/cpufreq/policy0/scaling_governor`
`echo “目标频率” >/sys/devices/system/cpu/cpufreq/policy0/scaling_max_freq`
锁定cpu频率，这个频率可以根据机器的性能需求和散热情况调整，在我这里设置500Mhz时，路由器响应ping延迟<1ns，这应该也算是一种解决方案吧

gbaoye, 2019-12-17:
> 很高兴有人注意到这一点
> 
> 因为 IPQ40xx 采用 28nm 制作，如果不控制频率，将会带来发热的问题，所以默认我采用了保守的节能策略
> 
> /etc/init.d/linksys_recovery 中可以调整这两个数值
> 
> echo 35 > /sys/devices/system/cpu/cpufreq/ondemand/up_threshold
> echo 10 > /sys/devices/system/cpu/cpufreq/ondemand/sampling_down_factor

28nm是非常成熟的工艺，这个发热没多大关系，是散热问题，14nm，7nm，一样会发热，过热

Aes64X, 2019-12-17:
[image]
QCA官网资料40nm ......................... - -

stock169, 2019-12-17:
只要能限制一下CPU最低频率应该就可以搞定

Aes64X, 2019-12-17:
LGA1150有一个cpu策略补丁，你可以试试，他那是4.19的，我看了4.14的补丁路径，那部分都一样可以直接用。

closed 2019-12-30 by coolsnowwolf