coolsnowwolf/lede #2461: V2RAY加密方式选择

star-o0O, 2019-12-16:
V2RAY官方推荐加密方式为auto，大部分机场默认配置为auto。
查文档发现AMD64、ARM64等设备中auto=AES-128-GCM。
众所周知，入门级软路由：雕™N2600，不支持aes且性能较差。
实测AES-128-GCM--------120Mbps（多线程,WebSocket）
CHACHA20-POLY1305----200Mbps（多线程,WebSocket）
目前通过手动修改订阅脚本/usr/share/shadowsocksr/subscribe.sh来修改默认加密方式。
请问能否实现
在不支持AES且性能较差的设备中auto=CHACHA20-POLY1305
在支持AES的设备中auto=AES-128-GCM
以实现在安全/性能中更好的平衡（也许？）

hcym, 2019-12-16:
有大区别吗，新三用机场的128-gcm也还行，和什么c20，gfb没什么大的不同。而且据说只有gcm支持多线程

star-o0O, 2019-12-16:
> 有大区别吗，新三用机场的128-gcm也还行，和什么c20，gfb没什么大的不同。而且据说只有gcm支持多线程

上边不是测了么，按百分比的话差67%，新3是性能弱爆了，差距不明显。

hcym, 2019-12-16:
蜗牛是不支持的，感觉auto和5020u没什么区别，不过我都是油管肉测一下，不计较小的差别

xynu617, 2019-12-16:
如果用的是ws+tls+web，加密方式选none就行了

star-o0O, 2019-12-16:
> 如果用的是ws+tls+web，加密方式选none就行了

就一层WS，不是自建的话，都不会套这么多吧。

star-o0O, 2019-12-16:
> 蜗牛是不支持的，感觉auto和5020u没什么区别，不过我都是油管肉测一下，不计较小的差别

我说前提了呀...N2600究极入门机。性能从账面上看，蜗牛J1900是它的220%+，5020U是他的480%+。我上3990X也是没什么差别的。

closed 2019-12-24 by coolsnowwolf