coolsnowwolf/lede #2471: SmartDNS已经开源了，能否在代码中加入？

tjuyy, 2019-12-16:
SmartDNS已经开源了，能否在代码中加入
https://github.com/pymumu/smartdns
https://github.com/pymumu/luci-app-smartdns

czjesda, 2019-12-17:
你自己把luci-app-smartdns下载后放在package目录里面就好了，不就可以固件编译了么

eallion, 2019-12-17:
试试先。

eallion, 2019-12-17:
```
cd ~/lede/package/lean
git clone https://github.com/pymumu/luci-app-smartdns.git
make menuconfig
# [*]luci-app-smartdns
make package/luci-app-smartdns/compile V=99
```

xiaoluoabc123, 2019-12-17:
> ```
> cd ~/lede/package/lean
> git clone https://github.com/pymumu/luci-app-smartdns.git
> make menuconfig
> # [*]luci-app-smartdns
> make package/luci-app-smartdns/compile V=99
> ```

编译不了各种问题

eallion, 2019-12-17:
> > ```
> > cd ~/lede/package/lean
> > git clone https://github.com/pymumu/luci-app-smartdns.git
> > make menuconfig
> > # [*]luci-app-smartdns
> > make package/luci-app-smartdns/compile V=99
> > ```
> 
> 编译不了各种问题

我按通用教程写的，我也还没试。

EnnawYang, 2019-12-17:
> > > ```
> > > cd ~/lede/package/lean
> > > git clone https://github.com/pymumu/luci-app-smartdns.git
> > > make menuconfig
> > > # [*]luci-app-smartdns
> > > make package/luci-app-smartdns/compile V=99
> > > ```
> > 
> > 
> > 编译不了各种问题
> 
> 我按通用教程写的，我也还没试。
少依赖 当然编不出来

EnnawYang, 2019-12-17:
> > > > ```
> > > > cd ~/lede/package/lean
> > > > git clone https://github.com/pymumu/luci-app-smartdns.git
> > > > make menuconfig
> > > > # [*]luci-app-smartdns
> > > > make package/luci-app-smartdns/compile V=99
> > > > ```
> > > 
> > > 
> > > 编译不了各种问题
> > 
> > 
> > 我按通用教程写的，我也还没试。

少依赖 当然编不出来

outemancut, 2019-12-18:
已经编译可以正常使用！不过openssl1.1.1要降回1.0.2不然会报错，不过应该可以用patches打上补丁支持，不过我不会

ch1993128, 2019-12-18:
直接改makefile编译依赖，删掉不支持库编译居然成功了，还能正常使用。。。。。。

closed 2019-12-18 by coolsnowwolf