coolsnowwolf/lede #2486: node编译出错

hceasy, 2019-12-18:
`../src/node_http2.cc: In constructor 'node::http2::Http2Options::Http2Options(node::Environment*, node::http2::nghttp2_session_type)':
../src/node_http2.cc:148:5: error: 'nghttp2_option_set_max_outbound_ack' was not declared in this scope
     nghttp2_option_set_max_outbound_ack(options_, 10000);
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../src/node_http2.cc:148:5: note: suggested alternative: 'nghttp2_option_set_no_auto_ping_ack'
     nghttp2_option_set_max_outbound_ack(options_, 10000);
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
     nghttp2_option_set_no_auto_ping_ack
node_lib.target.mk:156: recipe for target '/var/www/openwrt/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/node-v8.16.1/out/Release/obj.target/node_lib/src/node_http2.o' failed
make[5]: *** [/var/www/openwrt/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/node-v8.16.1/out/Release/obj.target/node_lib/src/node_http2.o] Error 1
rm 7bce13a782486bb6111801614a3420e769317268.intermediate
Makefile:88: recipe for target 'node' failed
make[4]: *** [node] Error 2
make[4]: Leaving directory '/var/www/openwrt/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/node-v8.16.1'`
这个有人碰到么.

coolsnowwolf, 2019-12-18:
make clean

hceasy, 2019-12-18:
> 
> 
> make clean
这个有试过,不行.
最后删了整个目录重新拉了一个版本,编译通过.

closed 2019-12-18 by hceasy