coolsnowwolf/lede #2489: smartdns编译缺少依赖

gsblay, 2019-12-18:
xyz@ubuntu:~/lede$ make defconfig
WARNING: Makefile 'package/lean/luci-app-smartdns/Makefile' has a dependency on 'smartdns', which does not exist
#
# configuration written to .config

lifetyper, 2019-12-18:
估计lean还没更新完吧，过两小时回来再看。

ch1993128, 2019-12-18:
lean老大真的加上去了嘛。不是吧

4kerSoo, 2019-12-18:
https://github.com/pymumu/smartdns/blob/master/package/openwrt/Makefile
package/lean下新建smartdns/Makefile

outemancut, 2019-12-18:
#
# Copyright (c) 2018-2019 Nick Peng (pymumu@gmail.com)
# This is free software, licensed under the GNU General Public License v3.
#

include $(TOPDIR)/rules.mk

PKG_NAME:=smartdns
PKG_VERSION:=1.2019.28
PKG_RELEASE:=1

PKG_SOURCE_PROTO:=git
PKG_SOURCE_URL:=https://www.github.com/pymumu/smartdns.git
PKG_SOURCE_VERSION:=cc93156ee3af9de89e8e6db72554a3790170ce0b
PKG_MIRROR_HASH:=89199413a3860d8245da6e4c6f171d449b89ec56162579004fa2166a29d20159

PKG_MAINTAINER:=Nick Peng <pymumu@gmail.com>
PKG_LICENSE:=GPL-3.0-or-later

PKG_BUILD_PARALLEL:=1

include $(INCLUDE_DIR)/package.mk

MAKE_ARGS:=CFLAGS="$(TARGET_CFLAGS) -DSMARTDNS_VERION=\\\"$(PKG_VERSION)\\\"" CC=$(TARGET_CC) 
MAKE_PATH:=src

define Package/smartdns
  SECTION:=net
  CATEGORY:=Network
  TITLE:=smartdns server
  DEPENDS:=+libopenssl
  URL:=http://www.github.com/pymumu/smartdns/
endef

define Package/smartdns/description
SmartDNS is a local DNS server which accepts DNS query requests from local network clients,
get DNS query results from multiple upstream DNS servers concurrently, and returns the fastest IP to clients.
Unlike dnsmasq's all-servers, smartdns returns the fastest IP. 
endef

define Package/smartdns/conffiles
/etc/config/smartdns
/etc/smartdns/address.conf
/etc/smartdns/blacklist-ip.conf
/etc/smartdns/custom.conf
endef

define Package/smartdns/install
	$(INSTALL_DIR) $(1)/usr/sbin $(1)/etc/config $(1)/etc/init.d $(1)/etc/smartdns
	$(INSTALL_BIN) $(PKG_BUILD_DIR)/src/smartdns $(1)/usr/sbin/smartdns
	$(INSTALL_BIN) $(PKG_BUILD_DIR)/package/openwrt/files/etc/init.d/smartdns $(1)/etc/init.d/smartdns
	$(INSTALL_CONF) $(PKG_BUILD_DIR)/package/openwrt/address.conf $(1)/etc/smartdns/address.conf
	$(INSTALL_CONF) $(PKG_BUILD_DIR)/package/openwrt/blacklist-ip.conf $(1)/etc/smartdns/blacklist-ip.conf
	$(INSTALL_CONF) $(PKG_BUILD_DIR)/package/openwrt/custom.conf $(1)/etc/smartdns/custom.conf
	$(INSTALL_CONF) $(PKG_BUILD_DIR)/package/openwrt/files/etc/config/smartdns $(1)/etc/config/smartdns
endef

$(eval $(call BuildPackage,smartdns))

gsblay, 2019-12-18:
大雕已经加上了！

closed 2019-12-18 by gsblay