coolsnowwolf/lede #2496: 常用公共DNS

eallion, 2019-12-18:
### 常用公共DNS
> 选择加粗的10个就可以了

|名称|DNS1|DNS2|
|---|---|---|
|**腾讯云**|119.29.29.29 |182.254.116.116|
|**阿里云**|223.5.5.5 |223.6.6.6|
|**114DNS**|114.114.114.114 |114.114.115.115|
|**CNNIC**|1.2.4.8 |210.2.4.8|
|**百度云**|180.76.76.76 ||
|360 DNS派| 101.226.4.6 |218.30.118.6|
|清华大学| 101.6.6.6 |2001:da8::666|
|OneDNS| 117.50.11.11 |117.50.22.22|
|**Google**| 8.8.8.8 |8.8.4.4|
|**OpenDNS**| 208.67.222.222 |208.67.220.220|
|**Cloudflare**| 1.1.1.1 |1.0.0.1|
|**Quad9**| 9.9.9.9 |149.112.112.112|
|**Yandex**| 77.88.8.8 |77.88.8.1|
|Norton| 199.85.126.10 |199.85.127.10|
|Freenom| 80.80.80.80 |80.80.81.81|
|COMODO| 8.26.56.26 |8.20.247.20|
|Level3| 209.244.0.3 |209.244.0.4|
|DNS.Watch| 84.200.69.80 |84.200.70.40|
|Verisign| 64.6.64.6 |64.6.65.6|
|SafeDNS| 195.46.39.39 |195.46.39.40|
|OpenNIC| 185.121.177.177 |169.239.202.202|
|SmartViper| 208.76.50.50 |208.76.51.51|
|Freenom| 80.80.80.80 |80.80.81.81|
|Dyn| 216.146.35.35 |216.146.36.36|
|FreeDNS| 37.235.1.174 |37.235.1.177|
|Alternate DNS| 198.101.242.72 |23.253.163.53|
|UncensoredDNS| 91.239.100.100 |89.233.43.71|
|Neustar| 156.154.70.1 |156.154.71.1|

fefz, 2019-12-18:
为什么要发到issue里？

eallion, 2019-12-18:
> 为什么要发到 issue 里？

就是为了好玩，不然为什么叫 Gayhub 呢。而且我发现这里小白比较多的。

shantsaichan, 2019-12-26:
挺好，有dns over tls和https的列表吗，最近找这种dns

closed 2019-12-30 by coolsnowwolf