coolsnowwolf/lede #2507: 定时切换策略

lix5027, 2019-12-19:
我看也有人有这个issue需求
希望大佬可以添加定时切换科学节点的功能.
企业环境下,多人使用,单一机场流量不够用,大概有3-8个机场,每天夜里4点定时切换,不知道方便加么?
如果不方便写,那就自己用crontab做了

bingo235, 2019-12-19:
负载均衡可否？

coolsnowwolf, 2019-12-19:
其实不需要，你安装一个 HAProxy 的 luci，把那几个节点加入，然后在 XXR Plus 设为 127.0.0.1 的节点即可

sqliuchang, 2019-12-23:
的确 haproxy更适合你 还可以根据不同机场流量的多少按比例分配负载权重 反正我个人校园网10拨 每个出口都指定一条机场线路

closed 2019-12-30 by coolsnowwolf

mailsex, 2020-01-20:
你好，我也需要这个功能，我有2个服务器，一个服务器只有150G流量，一个服务器1T流量，1T的服务器白天速度很好，150G的速度半天晚上都很好，所以能不能增加一个定时切换服务器的功能，晚上自动切换到指定的服务器。早上又切换到流量大的服务器。谢谢。

sqliuchang, 2020-01-21:
> 晚上自动切换到指定的服务器。早上又切换到流量大的服务器。

自己写个计划任务就能实现

mailsex, 2020-01-21:
就是不知道怎么写。没有找到ssr plus的相关文件。

获取 Outlook for iOS<https://aka.ms/o0ukef>

luyunlong666, 2020-03-06:
> > 晚上自动切换到指定的服务器。早上又切换到流量大的服务器。
> 
> 自己写个计划任务就能实现

请问您有模板吗  自己也有这个需求

sqliuchang, 2020-03-06:
> 请问您有模板吗 自己也有这个需求

会用sed的话，就自己用sed修改一下config文件，然后重启服务就行了
不会用sed的话，就把选用不同服务器时候的配置文件分别保存下来，然后在定时任务里面用cp替换当前配置文件，并重启服务

luyunlong666, 2020-03-06:
感谢您的回复～我研究一下

luyunlong666, 2020-03-08:
> 其实不需要，你安装一个 HAProxy 的 luci，把那几个节点加入，然后在 XXR Plus 设为 127.0.0.1 的节点即可

那样的话 线路会不会来回跳   看奈飞尴尬。

samueler2016, 2023-03-06:
> > 其实不需要，你安装一个 HAProxy 的 luci，把那几个节点加入，然后在 XXR Plus 设为 127.0.0.1 的节点即可
> 
> 那样的话 线路会不会来回跳 看奈飞尴尬。

哥们，你后来研究出来了没？😂

luyunlong666, 2023-03-07:
> 

没有 有什么更好的方法吗

luyunlong666, 2023-03-07:
> > > 其实不需要，你安装一个 HAProxy 的 luci，把那几个节点加入，然后在 XXR Plus 设为 127.0.0.1 的节点即可
> > 
> > 
> > 那样的话 线路会不会来回跳 看奈飞尴尬。
> 
> 哥们，你后来研究出来了没？😂

没有  你有什么好方法没