coolsnowwolf/lede #2529: 怎么修改登录IP？

Sevenp-hh, 2019-12-20:


gsblay, 2019-12-20:
我用的是文件大法。
其实就是把你要登陆的IP写入  files\etc\config\network

lphgor, 2019-12-20:
在这个文件[package/lean/default-settings/files/zzz-default-settings](https://github.com/coolsnowwolf/lede/blob/master/package/lean/default-settings/files/zzz-default-settings) 里面添加代码
```sh
uci set network.lan.ipaddr=192.168.1.1
uci commit network
```

Sevenp-hh, 2019-12-26:
> 在这个文件[package/lean/default-settings/files/zzz-default-settings](https://github.com/coolsnowwolf/lede/blob/master/package/lean/default-settings/files/zzz-default-settings) 里面添加代码
> 
> ```shell
> uci set network.lan.ipaddr=192.168.1.1
> uci commit network
> ```

好像不大行

xinpenghan, 2019-12-26:
使用下面三条命令，修改lan默认地址：
uci set network.lan.ipaddr=192.168.10.1
uci commit network
ifup lan

closed 2019-12-30 by coolsnowwolf