coolsnowwolf/lede #2537: ShadowSocksR Plus+ 与端口映射好像有问题！！

66508393, 2019-12-21:
我一开启 ShadowSocksR Plus+ 端口映射就不生效了！我用的是2019-12-20日的版本

caicaicai21, 2019-12-23:
我的正常，是你设置的端口和ShadowSocksR Plus+冲突了吗？！

66508393, 2019-12-23:
我描述一下吧  我是安装在pve虚似机里面的 刚早上重装了最新版本的 开启了 端口映射和ShadowSocksR Plus+ 开始的时候正常用一段时间  下午用着用着就又不行了！ 然后改 ShadowSocksR Plus+的端口结果也是不行 出问题之后 只要打开 ShadowSocksR Plus+就端口映射不了！ 刚刚试了一下 把  ShadowSocksR Plus+  下面的  需要代理的端口  由  所有端口   改为 仅常用端口  这样子就暂时正常 还需要测试一下！！

coolsnowwolf, 2019-12-23:
更新一下 国内IP段 数据库然后重启SSR Plus 就正常了

caicaicai21, 2019-12-24:
> 更新一下 国内IP段 数据库然后重启SSR Plus 就正常了

大佬，好像更新 国内IP段 会一直处于 "正在更新，请稍后..."，是规则无法下载吗？

chtx56789, 2019-12-24:
> 我描述一下吧 我是安装在pve虚似机里面的 刚早上重装了最新版本的 开启了 端口映射和ShadowSocksR Plus+ 开始的时候正常用一段时间 下午用着用着就又不行了！ 然后改 ShadowSocksR Plus+的端口结果也是不行 出问题之后 只要打开 ShadowSocksR Plus+就端口映射不了！ 刚刚试了一下 把 ShadowSocksR Plus+ 下面的 需要代理的端口 由 所有端口 改为 仅常用端口 这样子就暂时正常 还需要测试一下！！

有同样情况，固件是l大的OpenWrt R9.12.3 for K2P 集成低延迟转发补丁（B站神油）闭源驱动，之前弄半天以为是被墙内墙了，因为刚启动会正常一小段时间，然后就连不进设备，看了这段改成仅常用端口才让外头正常连进来。

coolsnowwolf, 2019-12-24:
同样的问题

closed 2019-12-30 by coolsnowwolf