coolsnowwolf/lede #2542: 制作了一个Github Actions + Docker Hub项目以支持增量编译，加速在线编译过程

tete1030, 2019-12-22:
使用github actions编译openwrt非常好用，只是编译过程不能保存下来供下次使用，导致每次小修改都要重新编译，非常耗时。

为了缩短时间，我们可以把actions和docker配合起来，将编译进度保存到docker hub。由于github actions网速很快，docker image的拉取和提交相对不会花费很长时间，这样我们就不必每次修改配置完全重新编译固件。

https://github.com/tete1030/openwrt-fastbuild-actions

近期花费了大量时间完善这个项目，不敢独享。目前相对稳定，base builder的创建也支持docker layer cache。如果大家使用过程遇到任何问题，欢迎提交issue帮助完善！

抱歉，由于精力有限，一部分介绍仅有英文版本。如果各位需要我会尽快完善README

garypang13, 2019-12-22:
太好了,希望尽快完善好中文readme,看英文太累了,

hcym, 2019-12-22:
感谢

bonnyandsky, 2019-12-25:
对小白来说还是太复杂了，每次编译2小时我还是可以等的，对于测试人员来说应该有用，可以大大的节约时间。

tete1030, 2019-12-25:
谢谢建议。一开始因为是自用，所以想做得灵活多变一些，结果就是学习曲线略陡。我正在把主要的步骤自动化，让步骤简单点

adaaaaaa, 2019-12-25:
> 学习曲线略陡

不是小白-friendly。。。🤪🤪🤪🤪🤪🤪

tete1030, 2019-12-25:
> > 学习曲线略陡
> 
> 不是小白-friendly。。。🤪🤪🤪🤪🤪🤪

刚刚更新让它friendly了一些，把前两种模式合并了，不需要理解具体实现细节就可以使用，可以merge或者重新fork试试
另外现在支持在commit message中指定编译job和选项，具体请看README

kang-mk, 2019-12-26:
好复杂 我还是简单粗暴吧  一次两个小时也不算很久了

tete1030, 2019-12-26:
> 好复杂 我还是简单粗暴吧 一次两个小时也不算很久了

嗯，每个人需求不同。主要是我自己每次编译只是为了添加package module，或者apply patch，这个时候等两个小时有些长。

我最开始是编译放在vultr里，每次编译restore snapshot，完成了再创建snapshot。蛮麻烦的，因此有了这个

eallion, 2019-12-28:
我电脑也不算好，每次编译只要10分钟左右，怎么要几个小时的？
试试
```
make download
make -j8 V=s
```
-j 后面跟上 cpu 核心数乘以2

tete1030, 2019-12-29:
> 我电脑也不算好，每次编译只要10分钟左右，怎么要几个小时的？
> 试试
> 
> ```
> make download
> make -j8 V=s
> ```
> 
> -j 后面跟上 cpu 核心数乘以2

你说的就是本地的增量编译啊。这里讨论的是使用Github Actions的在线增量编译。

closed 2019-12-30 by coolsnowwolf