coolsnowwolf/lede #2543: 不建议用SmartDNS，科学上网速度忽快忽慢，你们呢？用着怎么样？

xinpenghan, 2019-12-22:
SmartDNS和Turbo ACC有冲突，就算关闭Turbo ACC，使用SmartDNS科学上网体验也不好，速度忽快忽慢，最后还是用回原来的，你们呢？用着怎么样？

miaoxinwei, 2019-12-22:
SmartDNS用来做国内加速还行 主要有缓存

hcym, 2019-12-22:
号称移动加速，用了没什么感觉，还有那个ADguard home，正常使用快的还是快慢的还是慢，运行不正常就惨了，根本不知道哪个环节出叉。国内环境还是精确分流王道，其它花里胡哨的都是中医中药

zcluo, 2019-12-22:
开了dnsmasq上游ssr plus就状态里国外状态断开改转发53正常

发自我的iPhone

miaoxinwei, 2019-12-22:
> 开了dnsmasq上游ssr plus就状态里国外状态偏移改转发53正常发自我的iPhone
> […](#)
> 在2019年12月22日，下午10：12，lico ***@***.***>设置：SmartDNS使用做国内加速还行主要有缓存—您收到此消息是因为您已订阅到这个线程。直接回复此电子邮件，在GitHub上查看或取消订阅。

设置端口5335, 然后配置成dnsmasq上游  ssr plus选5335dns 最后需要重启一下lan口

guochan2019, 2019-12-22:
目前使用V2还算正常，但经常会导致网页页面加载不全。使用TJ的话，刚完全不通，估计还是有些问题，希望L大能完善，毕竟多了一个选择，支持L大

bob771025, 2019-12-22:
希望可以導入CoreDNS
CoreDNS支援國內外分流
可以設定國內域名走國內DNS,其他走UDP,TCP or DoT DNS 可以自行靈活設定

keivenliao, 2019-12-23:
我测试 Smartdns+ trojan 很快啊。

pV-nRT, 2019-12-23:
> 希望可以引入CoreDNS
> CoreDNS支持内部分流
> 可以设置内部域名走国内DNS，其他走UDP，TCP或DoT DNS可以自行灵活设置

smartdns有第二服务器，可以dns分流，具体看smartdns的参数介绍

coolsnowwolf, 2019-12-23:
加入 SmartDNS 其实主要是用来恶心那些 Up 主 和 小白们的

hcym, 2019-12-23:
大雕太坏了，gg要缩水五cm

closed 2019-12-24 by coolsnowwolf

ilss, 2020-04-16:
联通 电信 没什么正面效果，SSR+ 直接 pdsnd 也反映刚快。	       

可能铁通 广电那些残废宽带会有正面效果。

lswang6, 2020-04-20:
> 我测试 Smartdns+ trojan 很快啊。
> smartdns 配置好的dns 服务器很重要。
> [image]

请问你这个是哪个插件的界面呢?