coolsnowwolf/lede #2545: X64编译失败，恳请大佬帮忙

royalwei, 2019-12-22:
[image]
谢谢！！！！

royalwei, 2019-12-22:
Applying ./patches/000-printk.patch using plaintext: 
patching file xt_FULLCONENAT.c
cp -fpR ./files/Makefile /root/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2018-12-15-d4daedd0/
touch /root/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2018-12-15-d4daedd0/.prepared_a0e4ae0943bcba73fe92c037fa6fb661_6664517399ebbbc92a37c5bb081b5c53
rm -f /root/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2018-12-15-d4daedd0/.configured_*
rm -f /root/lede/staging_dir/target-x86_64_musl/stamp/.openwrt-fullconenat_installed
(cd /root/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2018-12-15-d4daedd0/./; if [ -x ./configure ]; then find /root/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2018-12-15-d4daedd0/ -name config.guess | xargs -r chmod u+w; find /root/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2018-12-15-d4daedd0/ -name config.guess | xargs -r -n1 cp --remove-destination /root/lede/scripts/config.guess; find /root/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2018-12-15-d4daedd0/ -name config.sub | xargs -r chmod u+w; find /root/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2018-12-15-d4daedd0/ -name config.sub | xargs -r -n1 cp --remove-destination /root/lede/scripts/config.sub; AR="x86_64-openwrt-linux-musl-gcc-ar" AS="x86_64-openwrt-linux-musl-gcc -c -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap/root/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2018-12-15-d4daedd0:fullconenat-2018-12-15-d4daedd0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" LD=x86_64-openwrt-linux-musl-ld NM="x86_64-openwrt-linux-musl-gcc-nm" CC="x86_64-openwrt-linux-musl-gcc" GCC="x86_64-openwrt-linux-musl-gcc" CXX="x86_64-openwrt-linux-musl-g++" RANLIB="x86_64-openwrt-linux-musl-gcc-ranlib" STRIP=x86_64-openwrt-linux-musl-strip OBJCOPY=x86_64-openwrt-linux-musl-objcopy OBJDUMP=x86_64-openwrt-linux-musl-objdump SIZE=x86_64-openwrt-linux-musl-size CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap/root/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2018-12-15-d4daedd0:fullconenat-2018-12-15-d4daedd0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " CXXFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap/root/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2018-12-15-d4daedd0:fullconenat-2018-12-15-d4daedd0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " CPPFLAGS="-I/root/lede/staging_dir/target-x86_64_musl/usr/include -I/root/lede/staging_dir/target-x86_64_musl/include -I/root/lede/staging_dir/toolchain-x86_64_gcc-7.5.0_musl/usr/include -I/root/lede/staging_dir/toolchain-x86_64_gcc-7.5.0_musl/include/fortify -I/root/lede/staging_dir/toolchain-x86_64_gcc-7.5.0_musl/include " LDFLAGS="-L/root/lede/staging_dir/target-x86_64_musl/usr/lib -L/root/lede/staging_dir/target-x86_64_musl/lib -L/root/lede/staging_dir/toolchain-x86_64_gcc-7.5.0_musl/usr/lib -L/root/lede/staging_dir/toolchain-x86_64_gcc-7.5.0_musl/lib -znow -zrelro "   ./configure --target=x86_64-openwrt-linux --host=x86_64-openwrt-linux --build=x86_64-pc-linux-gnu --program-prefix="" --program-suffix="" --prefix=/usr --exec-prefix=/usr --bindir=/usr/bin --sbindir=/usr/sbin --libexecdir=/usr/lib --sysconfdir=/etc --datadir=/usr/share --localstatedir=/var --mandir=/usr/man --infodir=/usr/info --disable-nls  ; fi; )
touch /root/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2018-12-15-d4daedd0/.configured_68b329da9893e34099c7d8ad5cb9c940
rm -f /root/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2018-12-15-d4daedd0/.built
touch /root/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2018-12-15-d4daedd0/.built_check
make -j1 -C "/root/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-4.19.88" CROSS_COMPILE="x86_64-openwrt-linux-musl-" ARCH="x86" SUBDIRS="/root/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2018-12-15-d4daedd0" EXTRA_CFLAGS="" modules
make[4]: Entering directory '/root/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-4.19.88'
make[5]: *** No rule to make target 'tools/objtool/objtool', needed by '/root/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2018-12-15-d4daedd0/xt_FULLCONENAT.o'.  Stop.
Makefile:1523: recipe for target '_module_/root/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2018-12-15-d4daedd0' failed
make[4]: *** [_module_/root/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2018-12-15-d4daedd0] Error 2
make[4]: Leaving directory '/root/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-4.19.88'
Makefile:64: recipe for target '/root/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2018-12-15-d4daedd0/.built' failed
make[3]: *** [/root/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2018-12-15-d4daedd0/.built] Error 2
make[3]: Leaving directory '/root/lede/package/lean/openwrt-fullconenat'
time: package/lean/openwrt-fullconenat/compile#0.62#0.28#0.88
package/Makefile:107: recipe for target 'package/lean/openwrt-fullconenat/compile' failed
make[2]: *** [package/lean/openwrt-fullconenat/compile] Error 2
make[2]: Leaving directory '/root/lede'
package/Makefile:103: recipe for target '/root/lede/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/root/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/root/lede'
/root/lede/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2

pepsigold, 2019-12-22:
samba36和samba4只能选一种

royalwei, 2019-12-22:
我没有都选

tieba123, 2019-12-23:
我们一样的报错，编译硬路由都没问题，软路由编译和你报错一样

royalwei, 2019-12-23:
> 
> 
> 我们一样的报错，编译硬路由都没问题，软路由编译和你报错一样

老铁，问题解决
 命令行输入 sudo apt-get update ，然后输入
sudo apt-get -y install build-essential asciidoc binutils bzip2 gawk gettext git libncurses5-dev libz-dev patch unzip zlib1g-dev lib32gcc1 libc6-dev-i386 subversion flex uglifyjs git-core gcc-multilib p7zip p7zip-full msmtp libssl-dev texinfo libglib2.0-dev xmlto qemu-utils upx libelf-dev autoconf automake libtool autopoint device-tree-compiler
是依赖库没更新

tieba123, 2019-12-23:
> 老铁，问题解决
> 命令行输入 sudo apt-get update ，然后输入
> sudo apt-get -y install build-essential asciidoc binutils bzip2 gawk gettext git libncurses5-dev libz-dev patch unzip zlib1g-dev lib32gcc1 libc6-dev-i386 [...]
我的不是这个问题，我的都是最新了，无线我也没勾选，也替换了openwrt官方源码

royalwei, 2019-12-23:
>

兄嘚，我已经编译出来了，确实是这个问题，不行删了git clone下吧

tieba123, 2019-12-24:
> 兄嘚，我已经编译出来了，确实是这个问题，不行删了git clone下吧

不知道为啥还是不行。我用稳定版的源码，最新的lede的没问题

coolsnowwolf, 2019-12-24:
No rule to make target 'tools/objtool/objtool', needed by '/root/lede/build_dir/target

缺 libelf-dev， apt-get 装上吧

closed 2019-12-24 by coolsnowwolf