coolsnowwolf/lede #2548: SmartDNS科学部分，V2正常，TJ不通，烦请L大疏通

guochan2019, 2019-12-22:
新加的SmartDNS设置后，国内部分完全正常，科学部分，V2也基本正常，但TJ死活不通（设置PDNSD能正常解析）。现在主力是TJ，烦请L大关注解决，感激不尽！

keivenliao, 2019-12-23:
我个人的没有问题。你看一下
[image]
和这里

[image]

guochan2019, 2019-12-23:
> 我个人的没有问题。你看一下
> [image]
> 和这里
> 
> [image]
设置没问题，我说的是在使用Torjan的情况下不通，你这个配置也用的tj?

keivenliao, 2019-12-23:
> > 我个人的没有问题。你看一下
> > [image]
> > 和这里
> > [image]
> > 设置没问题，我说的是在使用Torjan的情况下不通，你这个配置也用的tj?你那个重定向都没设置？你科学那部分方便截图看下吗？

是的，我也用的TJ ，速度基本跑满，而且很稳定。

guochan2019, 2019-12-23:
> > > 我个人的没有问题。你看一下
> > > [image]
> > > 和这里
> > > [image]
> > > 设置没问题，我说的是在使用Torjan的情况下不通，你这个配置也用的tj?你那个重定向都没设置？你科学那部分方便截图看下吗？
> 
> 是的，我也用的TJ ，速度基本跑满，而且很稳定。
> [image]

按你的设置后确实通了，不过状态里连通性检测还是不通，实际上是通了

keivenliao, 2019-12-23:
> > > > 我个人的没有问题。你看一下
> > > > [image]
> > > > 和这里
> > > > [image]
> > > > 设置没问题，我说的是在使用Torjan的情况下不通，你这个配置也用的tj?你那个重定向都没设置？你科学那部分方便截图看下吗？
> > 
> > 
> > 是的，我也用的TJ ，速度基本跑满，而且很稳定。
> > [image]
> 
> 按你的设置后确实通了，不过状态里连通性检测还是不通，实际上是通了

哪个可能是之前的dns地址有污染导致的， 你可以重新启动一下smartdns 和 ssrp 或者直接重启一次。另外smartdns 配置的地址很重要，你看看这里。
https://github.com/coolsnowwolf/lede/issues/2543#issuecomment-568323594

guochan2019, 2019-12-23:
> > > > > 我个人的没有问题。你看一下
> > > > > [image]
> > > > > 和这里
> > > > > [image]
> > > > > 设置没问题，我说的是在使用Torjan的情况下不通，你这个配置也用的tj?你那个重定向都没设置？你科学那部分方便截图看下吗？
> > > 
> > > 
> > > 是的，我也用的TJ ，速度基本跑满，而且很稳定。
> > > [image]
> > 
> > 
> > 按你的设置后确实通了，不过状态里连通性检测还是不通，实际上是通了
> 
> 哪个可能是之前的dns地址有污染导致的， 你可以重新启动一下smartdns 和 ssrp 或者直接重启路由一次。另外smartdns 配置的地址很重要，你看看这里。
> [#2543 (comment)](https://github.com/coolsnowwolf/lede/issues/2543#issuecomment-568323594)

好的，感谢兄弟指导

guochan2019, 2019-12-23:
感谢楼上兄台指导，经过测试，还是有这样那样的问题，也可能是我的TJ有问题，不过同样的设置V2却很正常

coolsnowwolf, 2019-12-23:
加入 SmartDNS 其实主要是用来恶心那些 Up 主 和 小白们的。 原理自己看 issue置顶

closed 2019-12-23 by coolsnowwolf