coolsnowwolf/lede #2551: 加入 SmartDNS 其实主要是用来恶心那些 Up 主 和 小白们的

coolsnowwolf, 2019-12-23:
1. 固件 DNS 的逻辑是， 国外域名会通过 SSR Plus 内部的 PDNSD 会走 SSR 的转发工具（SS/SSR/V2 等）建立的隧道到 VPS 节点，然后由 VPS 通过 TCP DNS 查询 8.8.8.8 等 BGP 服务器，获得离 VPS 节点最优的 CDN DNS 查询记录，这样你的梯子才是速度最快的

2. 国内域名直接缓存 ISP 或者 114 的结果，过滤掉错误的和被污染的IP记录，并缓存在内存中。

3. 你本地路由器通过 SmartDNS 查询国外域名，肯定返回的是“你本地路由器的最优CDN节点DNS记录”， 而不是 “VPS节点的最优DNS记录”。 这样你访问国外域名的记录肯定是几乎最慢的

那些 Up 主（例如 Mike 之类的傻逼），为了流量和点击率，DNS 的原理都没搞清楚，带节奏骗小白们。

工具本身没问题（SmartDNS 通过设置也能走隧道），是某些人的脑子问题而已。

82kg, 2019-12-23:
很多up主 水的一笔

guochan2019, 2019-12-23:
现在看明白了，感谢L大回复

hcym, 2019-12-23:
简单就是美，还是挺大雕

keivenliao, 2019-12-23:
我个人这么理解对否？
1、如果通过SSRP的PDNSD去 8.8.8.8 查询 dns地址，那么返回的应该是离VPS最近的服务器地址？
2、如果1成立，查询国内dns的地址时，是受SSRP的访问控制的，用gfwlist或者绕过国内IP查询？
3、过滤掉错误的和污染的记录是SSRP内置的功能?
4、如果在SSRP设置了绕过国内IP的情况下，smartdns 去访问国外IP时，例如8.8.8.8是否也会通过VPS？

hineH, 2019-12-23:
强烈建议内置dnscrypt-proxy2，(地址: https://github.com/DNSCrypt/dnscrypt-proxy )作为ssrp自带的dns方案取代默认的pdnsd。固件自带的luci-app-dnscrypt-proxy是一个非常老的版本，几乎废无法使用。

每次新升级以后都要先开ssrp内置pdnsd模式（因为不开代理根本没法下）然后用opkg安装dnscrypt-proxy2的ipk包（某些架构下还会出错，需要手动下载binary替换），然后配置好改ssrp为5335模式再重启。

只要做好分流，个人觉得smartDNS这种恨不得全世界都知道你在访问什么网站的多线查询是没有必要的，而加密解析则是至关重要，并且也是未来ESNI安全性的基石。且不说隐私问题，明文dns查询极易被代理提供商的一些莫名其妙的过滤规则误杀（如查询含有torrent关键字的域名）

个人采用的分流方案：
1. ssrp使用大陆IP白名单+5335作为dns解析的方式
2. dnscyrpt-proxy2配置监听在5335，仅使用DoH协议向cloudflare/adguard服务器查询（这些https的查询请求全部会被ssrp代理转发），并屏蔽ipv6的结果（ssrp能否支持同时连接v4 和v6两台代理服务器并分别处理不同流量呢）
3. iptables劫持所有目标为53端口的流量到路由器的53端口（这个默认就有了）
4. 53端口使用默认的dnsmasq作为dns服务，上游设置为127.0.0.1#5335，配置中添加`conf-dir=/etc/dnsmasq.d` 并在/etc/dnsmasq.d/中放入这个 https://github.com/felixonmars/dnsmasq-china-list/blob/master/accelerated-domains.china.conf 文件（并通过脚本自动更新）

这样实现了凡是大陆的和在大陆有cdn的以及返回AAAA记录的域名全部直连查询并直连访问，其余的全部走代理查询，且结果在大陆以外的走代理访问（IP+域名的双白名单机制）

doumeki, 2019-12-23:
> 我个人这么理解对否？
> 1、如果通过SSRP的PDNSD去 8.8.8.8 查询 dns地址，那么返回的应该是离VPS最近的服务器地址？
> 2、如果1成立，查询国内dns的地址时，是受SSRP的访问控制的，用gfwlist或者绕过国内IP查询？
> 3、过滤掉错误的和污染的记录是SSRP内置的功能?
> 4、如果在SSRP设置了绕过国内IP的情况下，smartdns 去访问国外IP时，例如8.8.8.8是否也会通过VPS？

同问，还就就是我试了下如果使用5335端口而非在下面选8.8.4.4之类的，进程中就没有pdnsd的进程了，也上不了外国网，也不知道是哪里改错了还是原本就是这样的。

keivenliao, 2019-12-23:
> > 我个人这么理解对否？
> > 1、如果通过SSRP的PDNSD去 8.8.8.8 查询 dns地址，那么返回的应该是离VPS最近的服务器地址？
> > 2、如果1成立，查询国内dns的地址时，是受SSRP的访问控制的，用gfwlist或者绕过国内IP查询？
> > 3、过滤掉错误的和污染的记录是SSRP内置的功能?
> > 4、如果在SSRP设置了绕过国内IP的情况下，smartdns 去访问国外IP时，例如8.8.8.8是否也会通过VPS？
> 
> 同问，还就就是我试了下如果使用5335端口而非在下面选8.8.4.4之类的，进程中就没有pdnsd的进程了，也上不了外国网，也不知道是哪里改错了还是原本就是这样的。

你5335 是smartdns 吗？

garypang13, 2019-12-23:
> 你5335 是smartdns 吗？

> 1. 固件 DNS 的逻辑是， 国外域名会通过 SSR Plus 内部的 PDNSD 会走 SSR 的转发工具（SS/SSR/V2 等）建立的隧道到 VPS 节点，然后由 VPS 通过 TCP DNS 查询 8.8.8.8 等 BGP 服务器，获得离 VPS 节点最优的 CDN DNS 查询记录，这样你的梯子才是速度最快的
> 2. 国内域名直接缓存 ISP 或者 114 [...]

如果SSRP选本地5335,然后把smartdns设置为5335,smartdns访问国外域名的时候会走隧道到VPS吗?

shantsaichan, 2019-12-23:
那些up完全就是骗小白，瞎配置，博眼球。
分享我用smartdns的方法：
1. 配置smartdns上游，国内用udp，国外用tls，https。
2.国外dns配置group组为foreign。
3.使用nameserver域名分流，下载gfw list转换为smartdns格式，nameserver指定gfw list分流走国外foreign组
4.对一些ip不通的域名，将请求转发到第二dns，分流使用smartdns的第二dns解析，第二dns关闭测速，就进解析ip，在配合ipset透明度请求网页 

smartdns支持tls，https，还有分流，少装了好几个插件。

ip黑名单还不知到怎么用，可能不需要。

keivenliao, 2019-12-23:
讲真，我觉得其实真不需要这么复杂，配几个国内的防污染DNS服务器速度就已经很快了。

doumeki, 2019-12-23:
> 你5335 是smartdns 吗？

不是，ssrp 自带就有是这个选项，是pdnsd的吧，但是我改了连pdnsd进程都没了

doumeki, 2019-12-23:
> 讲真，我觉得其实真不需要这么复杂，配几个国内的防污染DNS服务器速度就已经很快了，例如中科大，清华等。 可以去这里看看。https://dns.iui.im/#noad

问题不是啥dns,按大雕说法是能转发dns请求到vps,这才是重点

garypang13, 2019-12-23:
SSRP的转发只针对gfwlist里的域名,即使选择的是绕过大陆IP模式,大量国外有CDN的网站其实还是走的本地解析.

Bnosora, 2019-12-23:
请问局域网内同时存在白名单、黑名单和直连设备的时候，ssrplus内置的dns会根据每个设备的连接方式决定dns的查询方式吗？（白名单设备的dns也走白名单，黑名单设备的dns走黑名单，直连设备dns也直连）
还是说只按照全局设定决定所有设备的dns查询方式？（全局设定白名单，不管黑白直连设备的dns都走白名单）

coolsnowwolf, 2019-12-23:
还是要考虑 VPS CDN 的就近性DNS 选择问题

junwu0608, 2019-12-23:
干的漂亮。

eallion, 2019-12-23:
是的。
如果用这个固件，用 SSRplus 的话，就用 Turbo ACC DNS 加速就好了。
这几天用 OpenClash + SmartDNS 被坑惨了。当然分开来这两个都是好东西。

stztao, 2019-12-23:
东西是集成度挺高，最近通过Up们的推广，用的人一多问题层出不穷，后面再火个Trojan模式，感觉作者自己都理不开头绪处理内部bug了

xnxy2012, 2019-12-24:
挺大雕，另外问下各位，最近两天在编译的时候老师出现warning，说xxxxx、xxxxx目录或者文件不存在，不晓得是为啥子

Myoko, 2019-12-24:
感谢L大！对smartdns 还不太了解 学习一下如何设置

cnekol, 2019-12-24:
绝大多数up说的全是废话，以及，错话。。。粗略看大概是——全讲操作，不讲逻辑（知识）的。而且讲的操作还是建立在错误的逻辑（知识）上。

lietxia, 2019-12-24:
为啥说smartdns没啥用
1、DNS在干嘛？
【原理】DNS的作用就是你访问域名的时候，返回一个ip，比如 访问www.baidu.com 返回183.232.231.172
【无用理由】就是说：只有你同时+大量访问不同域名的东西的时候，才会受到
DNS查询太慢的影响。【那么你会同时访问超多不同域名的东西吗？】

2、多DNS查询
【原理】多个DNS一起查询
【无用理由】但实际上返回速度最快的【一直是那一个】

3、DNS防污染
【原理】一些服务器商给的DNS，会让你的HTTP访问广告网站
【无用理由】
1）只要你不用服务器商提供的DNS，一般公众DNS是直接有防污染的。
2）特别的上网方式访问海外，插件自身就用代理去查询DNS，也不会有海外污染。

4、DNS缓存
【原理】就是把查询DNS的结果暂存起来，下次查询直接调用。
【无用理由】这个大雕插件的Turbo ACC 早就有了。

5、广告过滤
【原理】把广告域名指向一个无法访问的地址，达到屏蔽效果
【无用理由】去广告范围很有限，效果也不如adbyby之类专门的去广告插件效果好

garypang13, 2019-12-24:
> 为啥说smartdns没啥用
> 1、DNS在干嘛？
> 【原理】DNS的作用就是你访问域名的时候，返回一个ip，比如 访问www.baidu.com 返回183.232.231.172
> 【无用理由】就是说：只有你同时+大量访问不同域名的东西的时候，才会受到
> DNS查询太慢的影响。【那么你会同时访问超多不同域名的东西吗？】
>
> 2、多DNS查询
> 【原理】多个DNS一起查询 [...]

胡言乱语,都不值得反驳

greennyreborn, 2019-12-24:
所以是这个东西我加了，但是确实没什么用的意思 XD

qiboshi19, 2019-12-24:
`那些 Up 主（例如 Mike 之类的傻逼）`太能装 看过一个拆星际蜗牛的视频 没见过这么能装的 直接过滤他的一切视屏

immyonlystar, 2019-12-24:
说的太对了，那几个傻逼UP，一个MIKE，一个金水（频道名字还叫JS神技能，最傻逼），一个什么三金，一个比一个傻逼，一个比一个屌丝。内容错误连篇，而且都废话超级多，两分钟能说清楚的事儿，非要扯上二三十分钟。恶心透了。smartdns装了试用了下又卸载了，按那些SB UP的设置方法外网根本不通，换回系统自带的一切正常。

chtx56789, 2019-12-25:
SmartDNS 目前有个优点ipv6环境下，强制只回报ipv4 地址。https://www.mivm.cn/openwrt-connection-reset/

f3404923, 2019-12-25:
没人能拿出点数据来比较吗，没有数据来瞎bb也没啥意义

hcym, 2019-12-25:
比较一下，自己做主吧，我是喜欢简单，不喜欢复杂。ss和v2一直都是基本的tcp，也用的蛮好

pV-nRT, 2019-12-25:
我认为固件dns的逻辑是：
（1）使用GFW模式时（开启GFWlist查询，没有开启国内IP库查询）：发起域名解析请求-判断该域名在GFWlist中？-是，使用pdnsd进行tcp隧道解析，返回结果来自于海外vps的ip，并使用隧道访问（否，使用路由器本机国内dns解析，返回结果来自于国内dns，并使用国内直连，无论该结果是否为大陆ip，因为GFW模式没有开启国内IP库查询）
（2）使用绕过大陆模式时（开启GFWlist查询，同时打开国内IP库查询）：发起域名解析请求-判断该域名在GFWlist中？-是，使用pdnsd进行tcp隧道查询，返回结果来自于海外vps的ip，并使用隧道访问（否，使用路由器本机国内dns解析，返回结果来自于国内dns，此时与GFW模式的区别是，该处会再一次进行国内IP库的查询，返回结果属于国内，则使用直连访问，返回结果属于非国内，则使用隧道访问）

不知道这样理解对不对

pV-nRT, 2019-12-25:
还有一点疑惑：
本地路由器通过smartdns或是其他dns插件，如果添加了海外dns，则需要在XXRplus的“访问控制”-“wan ip访问控制”-“强制走代理的wan ip”中添加海外dns的ip地址，如果是使用https-dns，还要把对应的https网址填入“强制走代理的域名”中。

这样的作法是否正确？

410054336, 2019-12-25:
所谓的up主，就借此处代码，骗小白买自己的机场和那些垃圾硬件

hfutxqd, 2019-12-25:
> 强烈建议内置dnscrypt-proxy2，(地址: https://github.com/DNSCrypt/dnscrypt-proxy )作为ssrp自带的dns方案取代默认的pdnsd。固件自带的luci-app-dnscrypt-proxy是一个非常老的版本，几乎废无法使用。 [...]

我的packages feed里有dnscrypt-proxy2

ylwnet, 2019-12-25:
Mike  这个傻逼 早晚举报了，为了赚钱 卖机场，各种恶心人的做法，
各种忽悠 说是别人的机场 就TM 自己建立的忽悠小白们
 当婊子立牌坊 发点什么 都被说是广告。
真TM傻逼。他的信息我都搜集好了，已经提交 一些部门举报这傻逼。

ylwnet, 2019-12-25:
另外 这个傻逼 P技术没有 ，都是先抄 现卖  就为了卖机场赚钱。早晚抓紧去。这傻逼 咋不去死。
另外绝对支持大雕

realsteam, 2019-12-26:
好多视频主都是这样的，设置通了以为就可以教学出视频了，很多误导人的地方原理逻辑自己都是糊涂的。关键是有些视频作者喜欢把标题弄的耸人听闻，唬人的一屁的标题

hineH, 2019-12-26:
> 我的packages feed里有dnscrypt-proxy2

@hfutxqd   请问下路径？

coolsnowwolf, 2019-12-26:
> 还有一点疑惑：
> 本地路由器通过smartdns或是其他dns插件，如果添加了海外dns，则需要在XXRplus的“访问控制”-“wan ip访问控制”-“强制走代理的wan ip”中添加海外dns的ip地址，如果是使用https-dns，还要把对应的https网址填入“强制走代理的域名”中。
> 
> 这样的作法是否能使smartdns和其他dns插件正确走隧道？（在plus使用5335端口进行dns解析，同时smartdns或其他dns插件已经正确重定向5335时）

确实是可以这样的

82kg, 2019-12-26:
> 我认为固件dns的逻辑是：
> （1）使用GFW模式时（开启GFWlist查询，没有开启国内IP库查询）：发起域名解析请求-判断该域名在GFWlist中？-是，使用pdnsd进行tcp隧道解析，返回结果来自于海外vps的ip，并使用隧道访问（否，使用路由器本机国内dns解析，返回结果来自于国内dns，并使用国内直连，无论该结果是否为大陆ip，因为GFW模式没有开启国内IP库查询） [...]

对

hyird, 2019-12-26:
SmartDNS没问题，某些人的脑子有问题

hfutxqd, 2019-12-26:
代理+DoT/DoH(smartdns, adguardhome等都能实现)有个好处是机场无法收集到你访问的域名信息

hfutxqd, 2019-12-26:
> @hfutxqd 请问下路径？

https://github.com/hfutxqd/packages/tree/master/net/dnscrypt-proxy2
https://github.com/hfutxqd/luci/tree/master/applications/luci-app-dnscrypt-proxy

ihuaer, 2019-12-27:
没有你开源的软件，也就没有现在openwrt遍地开花

qpzr, 2019-12-27:
那些半桶水的  和 那些看了有图逼艾斯比主子说马上听风就是雨的狗腿子 太多了。连DNS是什么 SMATDNS的原理 和作用是什么都不知道  动机就是很明显的往红杏出墙的方向带，意图很明显的
我就随便说了一句自己玩低调点，直接被回怼 “你没用过，你不明白 感受那是太明显了”  哎 我就呵呵了，那帮人想 忽如一夜春风来，万里城墙全倒塌 想疯了吧。。。
搞到现在这个局势 raw.githubusercontent.com 都已经 半墙和连接已重置的状态了  真是害人不浅

aglent, 2019-12-30:
> 
> 
> > 还有一点疑惑：
> > 本地路由器通过smartdns或是其他dns插件，如果添加了海外dns，则需要在XXRplus的“访问控制”-“wan ip访问控制”-“强制走代理的wan ip”中添加海外dns的ip地址，如果是使用https-dns，还要把对应的https网址填入“强制走代理的域名”中。
> > 这样的作法是否能使smartdns和其他dns插件正确走隧道？（在plus使用5335端口进行dns解析，同时smartdns或其他dns插件已经正确重定向5335时）
> 
> 确实是可以这样的

大师：
我在固件里SSR用大陆IP白名单+5335端口DNS服务
Smartdns添加国内和国外DNS节点 本地端口6053 同时开启第二DNS服务端口5335 这个端口专用国外的DNS tls模式
固件dnsmasq的DHCP/DNS设置了转发到smartdns
请问这样配置下
1、域名请求是先到smartdns查询 获得的IP根据大陆白名单判断是否走代理节点隧道？
  还是先到SSR去匹配gfwlist名单 在gfwlist名单里走代理隧道 不在名单里域名再发送到smartdns查询 结果ip继续用SSR大陆白名单判断是否走代理节点隧道？
2、smartdns配置的国外tls dns，如果他们的IP都没有强制走SSR代理，那么查询出来的国外IP只是根据本地优化，而不是针对SSR里的代理节点优化，是这样理解吗？
3、上面的smartdns有无必要设置成2个服务端口，其中一个5335针对SSR？
还是整个smartdns就一个5335端口 为SSR服务？
对SSR来讲有区别吗？
谢谢！

wclmgcd, 2019-12-31:
> > > 还有一点疑惑：
> > > 本地路由器通过smartdns或是其他dns插件，如果添加了海外dns，则需要在XXRplus的“访问控制”-“wan ip访问控制”-“强制走代理的wan ip”中添加海外的https网址填入“强制走代理的域名”中。
> > > 这样的作法是否能使smartdns和其他dns插件正确走隧道？（在plus使用5335端口进行dns解析，同时smartdns或其他dns插件已经正确重定向5335时）
> > 
> > 
> > 确实是可以这样的
> 
小白一个，smardns还是有用吧，我是移动网，移动劫持53端口，导致有时候有的网页打不开，于是大雕固件采用gfw模式，被墙的域名走GFW并转发smardns的5335端口，并设置DT H的谷歌，cf,并让这两个IP走X XP，对于翻墙的域名有加速，其余的域名走6053,用中科大TCP, 和红鱼，鹅厂加速

aglent, 2020-01-01:
> 强烈建议内置dnscrypt-proxy2，(地址: https://github.com/DNSCrypt/dnscrypt-proxy )作为ssrp自带的dns方案取代默认的pdnsd。固件自带的luci-app-dnscrypt-proxy是一个非常老的版本，几乎废无法使用。 [...]

今天更新编译 发现luci里dnscrypt变为dnscrypt-proxy2了
真的是已经集成进去了？

ErickG233, 2020-01-02:
> 今天更新编译 发现luci里dnscrypt变为dnscrypt-proxy2了
> 真的是已经集成进去了？

刚才更新了下，没看到。。。

aglent, 2020-01-02:
> 刚才更新了下，没看到。。。

仔细看了下 原来我的package目录下有一份dnscrypt-proxy2，应该是之前git clone下来的
固件是没有集成进去

aglent, 2020-01-02:
根据上面几位大师的分析 尝试smartdns和SSR配合使用 效果还可以
SmartDNS 本地端口6053 添加国内和国外DNS（tls） 把国外DNS分为一组 开启第二DNS服务器 端口设为5335 跳过cache 国内dns开启过滤黑名单IP
chinadns-ng 本地端口5353 国内外dns分别指向smartdns的6053和5335
SSR使用绕过大陆IP模式 用5335的DNS服务 强制wan ip走代理那里添加国外dns IP
dnsmasq转发到chinadns-ng
经测试 国内域名获取得IP都是延迟最低的
国外域名无论是特殊的还是正常的 获得的IP对代理节点优化
国外正常域名在国内有IP的 获取对应国内IP 如intel.com

winxpp, 2020-01-07:
> 很多up主 水的一笔

那个麦克 满嘴开航天飞机 搞的神秘兮兮的

KevinZonda, 2020-01-09:
> 讲真，我觉得其实真不需要这么复杂，配几个国内的防污染DNS服务器速度就已经很快了，例如中科大，清华等。 可以去这里看看。https://dns.iui.im/#noad

国内运营商是直接拦截 53端口的udp DNS查询协议，然后抢答。基本上，你使用任何标准端口的自建服务，也会被拦截

keivenliao, 2020-01-10:
> > 讲真，我觉得其实真不需要这么复杂，配几个国内的防污染DNS服务器速度就已经很快了，例如中科大，清华等。 可以去这里看看。https://dns.iui.im/#noad
> 
> 国内运营商是直接拦截 53端口的udp DNS查询协议，然后抢答。基本上，你使用任何标准端口的自建服务，也会被拦截

这年头谁还用UDP啊
[image]

ntgeralt, 2020-01-13:
@doumeki  中科大的还好；清华的DNS，你不是学校内部的会面临间接性抽风。主页解释是所有解释权归他们所有。
@keivenliao 长宽已实现劫持53 tcp

josephzjw, 2020-01-18:
> SSRP的DNS转发只针对gfwlist里的域名,即使选择的是绕过大陆IP模式,大量国外有CDN的网站其实还是走的本地解析.如果选择绕过大陆IP模式,那么DNS也应该所有国外域名都转发到代理VPS才对.

我也发现这个现象了，我明明使用了全局模式，但是有些域名还是无法访问，感觉即使是全局模式好像也不是所有 DNS 查询走服务器的 DNS 解析。相同情况下我开 SS 客户端设置成全局模式，就可以访问。

norikohxb, 2020-01-22:
支持lean大，非常感谢lean在这里说出了这些实话，其实你的v2ray已经把你的域名通过你的vps的ip地址直接联系谷歌的dns了，还需要劳什子smartdns去加上一堆dns。还有国内的也是，我厦门的dns就很好用了。lean的解释真的是一针见血

leokkk, 2020-02-06:
装了smartdns，搞了半天才发现了问题，原来L大早已看穿一切。

sandianyue, 2020-02-06:
马一下 都是大佬方案 容我慢慢试😂

OUZYcn, 2020-02-09:
> 马一下 都是大佬方案 容我慢慢试😂

我最近的偷懒方式，主要是因为DNSCrypt-proxy2没有好用的luci界面：
1. 下载eSir最新编译的SPP小包（必要设定自行搞点）
2. 启用AdGuard home，配置为dnsmasq的上游。（因为AdGuard也是集成了DNSCrypt-proxy支持DoH的版本，也可以填入sdns://格式地址。
3. ssrp 使用绕开国内ip，5335作为dns解析。
舒坦。

PS：sdns，依据相应速度，建议如下
sdns://AgcAAAAAAAAACjQ1LjkwLjI4LjAgPhoaD2xT8-l6SS1XCEtbmAcFnuBXqxUFh2_YP9o9uDgOZG5zLm5leHRkbnMuaW8PL2Ruc2NyeXB0LXByb3h5
sdns://AgcAAAAAAAAABzEuMC4wLjEAEmRucy5jbG91ZGZsYXJlLmNvbQovZG5zLXF1ZXJ5
https://dns.quad9.net/dns-query

OUZYcn, 2020-02-09:
> 强烈建议内置dnscrypt-proxy2，(地址: https://github.com/DNSCrypt/dnscrypt-proxy )作为ssrp自带的dns方案取代默认的pdnsd。固件自带的luci-app-dnscrypt-proxy是一个非常老的版本，几乎废无法使用。 [...]

请教：“accelerated-domains.china” 这个怎么配置才会生效？我目前的dnsmasq:53 上游是 adguard 5335 DoH通过SSR plus去1.1.1.1，但是发现不管是国内，国外都走了SSRP通道。

[170 more replies]

closed 2020-07-04 by coolsnowwolf