coolsnowwolf/lede #2554: SSR Plus 选绕过大陆IP模式,但DNS转发还是用的gfwlist

garypang13, 2019-12-23:
导致大量国外有CDN的网站流量走的代理,但是IP却是本地DNS解析的结果.造成访问缓慢.

sqliuchang, 2019-12-23:
总不能做个国内域名表 或者 国外域名表 吧，这不现实啊

Bnosora, 2019-12-23:
> 总不能做个国内域名表 或者 国外域名表 吧，这不现实啊

有国内cdn表和ip表，绕过大陆ip模式或者叫白名单模式用的就是这东西。

eallion, 2019-12-23:
在`状态`标签下更新一下列表？

sqliuchang, 2019-12-23:
> 有国内cdn表和ip表，绕过大陆ip模式或者叫白名单模式用的就是这东西。
但那是ip段啊，cdn和国内ip段是长期稳定的，并且ip数量是有限的，再配合CIDR 记法，可以有效地分流。
域名做不到这一点，首先域名是无穷多的，无论是国内还是国外，并且没有域名CIDR这种东西。其次要想做到国外域名走代理隧道DNS解析，就必须有规则判定这个域名是国内还是国外，也就需要维护一套国内域名表或国外域名表。这不现实，最多维护gfwlist+国外topxxxx，而这样的话完全可以自己把国外top网站加入黑名单或者在dnsmasq中将其设置为gfwlist并使用代理解析DNS，最后再把ss设置为gfw模式就好了。了。

coolsnowwolf, 2019-12-24:
其实有个情况： 如果一个全球域名既有国内的 aws 节点，也有国际的，那应该算国内域名还是国外域名呢？

coolsnowwolf, 2019-12-24:
例如 apple.com

Bnosora, 2019-12-28:
> 但那是ip段啊，cdn和国内ip段是长期稳定的，并且ip数量是有限的，再配合CIDR 记法，可以有效地分流。 [...]

没注意ip表对解析没帮助是我疏忽了，对不起。但隔壁v2ray插件的cdn表是域名表，所以那插件提供了dns解析优先度的选项（国内优先or国外优先），cdn表（域名表）来源如下：

https://raw.githubusercontent.com/hq450/fancyss/master/rules/cdn.txt

closed 2019-12-30 by coolsnowwolf