coolsnowwolf/lede #2557: openwrt官方源码添加L大的SSR-plus插件，首次开启，GFW模式，要等很长一段时间才可以正常使用

xiaoxiang8002, 2019-12-23:
第一次开启后，无法DNS服务器无法正常解析，等一段时间后才可以正常使用，请问是哪里设置的问题？

Mon Dec 23 17:52:40 2019 daemon.info dnsmasq[1718]: read /etc/hosts - 4 addresses
Mon Dec 23 17:52:40 2019 daemon.info dnsmasq[1718]: read /tmp/hosts/odhcpd - 4 addresses
Mon Dec 23 17:52:40 2019 daemon.info dnsmasq[1718]: read /tmp/hosts/dhcp.cfg01411c - 2 addresses
Mon Dec 23 17:52:40 2019 daemon.info dnsmasq-dhcp[1718]: read /etc/ethers - 0 addresses
Mon Dec 23 17:53:56 2019 kern.info /usr/bin/ssr-redir[2999]: initializing ciphers... none
Mon Dec 23 17:53:56 2019 kern.info /usr/bin/ssr-redir[3001]: initializing ciphers... none
Mon Dec 23 17:53:56 2019 kern.info /usr/bin/ssr-redir[3003]: initializing ciphers... none
Mon Dec 23 17:53:56 2019 kern.info /usr/bin/ssr-redir[3005]: initializing ciphers... none
Mon Dec 23 17:53:56 2019 daemon.info pdnsd[3026]: pdnsd-1.2.9b-par starting.
Mon Dec 23 17:53:56 2019 daemon.err pdnsd[3026]: Could not listen on tcp socket: Not a socket
Mon Dec 23 17:53:56 2019 daemon.err pdnsd[3026]: error in UDP recv: Not a socket
Mon Dec 23 17:53:56 2019 daemon.err pdnsd[3026]: error in UDP recv: Not a socket
Mon Dec 23 17:53:56 2019 daemon.err pdnsd[3026]: error in UDP recv: Not a socket
Mon Dec 23 17:53:56 2019 daemon.err pdnsd[3026]: error in UDP recv: Not a socket
Mon Dec 23 17:53:56 2019 daemon.err pdnsd[3026]: error in UDP recv: Not a socket
Mon Dec 23 17:53:56 2019 daemon.err pdnsd[3026]: error in UDP recv: Not a socket
Mon Dec 23 17:53:56 2019 daemon.err pdnsd[3026]: error in UDP recv: Not a socket
Mon Dec 23 17:53:56 2019 daemon.err pdnsd[3026]: error in UDP recv: Not a socket
Mon Dec 23 17:53:56 2019 daemon.err pdnsd[3026]: error in UDP recv: Not a socket
Mon Dec 23 17:53:56 2019 daemon.err pdnsd[3026]: error in UDP recv: Not a socket
Mon Dec 23 17:53:57 2019 daemon.info dnsmasq[1718]: exiting on receipt of SIGTERM
Mon Dec 23 17:53:58 2019 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Mon Dec 23 17:53:58 2019 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Mon Dec 23 17:53:59 2019 kern.info /usr/bin/ssr-redir[3003]: tcp port reuse enabled
Mon Dec 23 17:53:59 2019 kern.info /usr/bin/ssr-redir[3003]: listening at 0.0.0.0:1234
Mon Dec 23 17:53:59 2019 kern.info /usr/bin/ssr-redir[3003]: running from root user
Mon Dec 23 17:53:59 2019 kern.info /usr/bin/ssr-redir[3005]: tcp port reuse enabled
Mon Dec 23 17:53:59 2019 kern.info /usr/bin/ssr-redir[3005]: listening at 0.0.0.0:1234

coolsnowwolf, 2019-12-23:
这个你要问官方了

9l, 2019-12-25:
你先开全局，然后手动更新gfw列表，国内域名ip啥的，更新完再换成gfw模式

xinpenghan, 2019-12-27:
dns无法解析和SSR-plus应该没关系。你用命令清理一下电脑dns缓存或者断一下网，然后在试

closed 2019-12-28 by coolsnowwolf