coolsnowwolf/lede #2570: 求大神支招！！编译的时候怎么加入Samba4呢？一直提示编译错误

xinpenghan, 2019-12-24:
编译配置文件，加入samba4，已去掉samba，还是提示错误
CONFIG_PACKAGE_luci-app-samba4=y
# CONFIG_PACKAGE_uci-app-samba is not set

miaoxinwei, 2019-12-24:
# CONFIG_PACKAGE_autosamba is not set

xinpenghan, 2019-12-25:
> ```
> # CONFIG_PACKAGE_autosamba is not set
> # CONFIG_PACKAGE_luci-app-samba is not set
> CONFIG_PACKAGE_luci-app-samba4=y
> CONFIG_SAMBA4_SERVER_NETBIOS=y
> ```
> 
> 用samba4传文件，用miniDLNA看视频，我是这么用的

谢谢你，在问一下，samba4要比samba速度要快吗？

miaoxinwei, 2019-12-25:
> > ```
> > # CONFIG_PACKAGE_autosamba is not set
> > # CONFIG_PACKAGE_luci-app-samba is not set
> > CONFIG_PACKAGE_luci-app-samba4=y
> > CONFIG_SAMBA4_SERVER_NETBIOS=y
> > ```
> > 
> > 
> > 用samba4传文件，用miniDLNA看视频，我是这么用的
> 
> 谢谢你，在问一下，samba4要比samba速度要快吗？

那就不知道了，我一直在用samba4

xinpenghan, 2019-12-25:
> > > ```
> > > # CONFIG_PACKAGE_autosamba is not set
> > > # CONFIG_PACKAGE_luci-app-samba is not set
> > > CONFIG_PACKAGE_luci-app-samba4=y
> > > CONFIG_SAMBA4_SERVER_NETBIOS=y
> > > ```
> > > 
> > > 
> > > 用samba4传文件，用miniDLNA看视频，我是这么用的
> > 
> > 
> > 谢谢你，在问一下，samba4要比samba速度要快吗？
> 
> 那就不知道了，我一直在用samba4

您的samba4 设置模版可以复制给我吗？我用的默认设置，mac连接不上
[global]
	netbios name = |NAME| 
	interfaces = |INTERFACES|
	server string = |DESCRIPTION|
	unix charset = |CHARSET|
	workgroup = |WORKGROUP|

	## This global parameter allows the Samba admin to limit what interfaces on a machine will serve SMB requests.
	bind interfaces only = yes
	
	## time for inactive connections to-be closed in minutes
	deadtime = 15
	
	## disable core dumps
	enable core files = no
	
	## set security (auto, user, domain, ads)
	security = user
	
	## This parameter controls whether a remote client is allowed or required to use SMB encryption.
	## It has different effects depending on whether the connection uses SMB1 or SMB2 and newer:
    ## If the connection uses SMB1, then this option controls the use of a Samba-specific extension to the SMB protocol introduced in Samba 3.2 that makes use of the Unix extensions.
	## If the connection uses SMB2 or newer, then this option controls the use of the SMB-level encryption that is supported in SMB version 3.0 and above and available in Windows 8 and newer. 
	##(default/auto,desired,required,off)
	#smb encrypt = default
	
	## set invalid users
	invalid users = root
	
	## map unknow users to guest
	map to guest = Bad User

	## allow client access to accounts that have null passwords. 
	null passwords = yes
	
	## The old plaintext passdb backend. Some Samba features will not work if this passdb backend is used. (NOTE: enabled for size reasons)
	## (tdbsam,smbpasswd,ldapsam)
	passdb backend = smbpasswd
	
	## Set location of smbpasswd ('smbd -b' will show default compiled location)
	#smb passwd file = /etc/samba/smbpasswd 
	
	## LAN/WAN options (IPTOS_LOWDELAY TCP_NODELAY) WAN (IPTOS_THROUGHPUT)
	socket options = IPTOS_LOWDELAY TCP_NODELAY
	
	## lower CPU useage if supported
	use sendfile = yes
	
	## samba will behave as previous versions of Samba would and will fail the lock request immediately if the lock range cannot be obtained.
	#blocking locks = No
	
	## disable loading of all printcap printers by default (iprint, cups, lpstat)
	load printers = No
	printcap name = /dev/null

	## Enabling this parameter will disable Samba's support for the SPOOLSS set of MS-RPC's.
	disable spoolss = yes
	
	## This parameters controls how printer status information is interpreted on your system.
	## (BSD, AIX, LPRNG, PLP, SYSV, HPUX, QNX, SOFTQ)
	printing = bsd
	
	## Disable that nmbd is acting as a WINS server for unknow netbios names
	#dns proxy = No
	
	## win/unix user mapping backend
	#idmap config * : backend = tdb
	
	## Allows the server name that is advertised through MDNS to be set to the hostname rather than the Samba NETBIOS name.
	## This allows an administrator to make Samba registered MDNS records match the case of the hostname rather than being in all capitals.
	## (netbios, mdns)
	mdns name = mdns
	
	## Clients that only support netbios won't be able to see your samba server when netbios support is disabled.
	#disable netbios = Yes
	
	## Setting this value to no will cause nmbd never to become a local master browser.
	#local master = no

	## (auto, yes) If this is set to yes, on startup, nmbd will force an election, and it will have a slight advantage in winning the election. It is recommended that this parameter is used in conjunction with domain master = yes, so that nmbd can guarantee becoming a domain master. 
	#preferred master = yes

	## (445 139) Specifies which ports the server should listen on for SMB traffic.
	## 139 is netbios/nmbd
	#smb ports = 445 139
	
	## This is a list of files and directories that are neither visible nor accessible.
	## Each entry in the list must be separated by a '/', which allows spaces to be included in the entry. '*' and '?' can be used to specify multiple files or directories as in DOS wildcards.
	veto files = /Thumbs.db/.DS_Store/._.DS_Store/.apdisk/
	
	## If a directory that is to be deleted contains nothing but veto files this deletion will fail unless you also set the delete veto files parameter to yes.
	delete veto files = yes
	
################ Filesystem and creation rules ################
	## reported filesystem type (NTFS,Samba,FAT)
	#fstype = FAT
	
	## Allows a user who has write access to the file (by whatever means, including an ACL permission) to modify the permissions (including ACL) on it.
	#dos filemode = Yes
	
	## file/dir creating rules
	#create mask = 0666
	#directory mask = 0777
	#force group = root
	#force user = root
	#inherit owner = windows and unix
################################################################

miaoxinwei, 2019-12-25:
> 您的samba4 设置模版可以复制给我吗？我用的默认设置，mac连接不上
> [global]
> netbios name = |NAME|
> interfaces = |INTERFACES|
> server string = |DESCRIPTION|
> unix charset = |CHARSET|
> workgroup = |WORKGROUP|
>
> ```
> ## This global [...]

```
## set invalid users
invalid users = root
```
改成
```

## set invalid users
# invalid users = root
```

xinpenghan, 2019-12-25:
> ```
> ## set invalid users
> invalid users = root
> ```
>
> 改成
>
> ```
>
> ## set invalid users
> # invalid users = root
> ```

谢谢，解决了，困扰好久的编译问题

miaoxinwei, 2019-12-25:
不客气

closed 2019-12-30 by coolsnowwolf