coolsnowwolf/lede #2576: 编译时出错，求助！！

nono8273, 2019-12-25:
cc1: some warnings being treated as errors
scripts/Makefile.build:303: recipe for target '/home/openwrt/lede/build_dir/target-aarch64_cortex-a53_musl/linux-brcm2708_bcm2710/cryptodev-linux-1.9.git-2017-10-04/ioctl.o' failed
make[6]: *** [/home/openwrt/lede/build_dir/target-aarch64_cortex-a53_musl/linux-brcm2708_bcm2710/cryptodev-linux-1.9.git-2017-10-04/ioctl.o] Error 1
Makefile:1523: recipe for target '_module_/home/openwrt/lede/build_dir/target-aarch64_cortex-a53_musl/linux-brcm2708_bcm2710/cryptodev-linux-1.9.git-2017-10-04' failed
make[5]: *** [_module_/home/openwrt/lede/build_dir/target-aarch64_cortex-a53_musl/linux-brcm2708_bcm2710/cryptodev-linux-1.9.git-2017-10-04] Error 2
make[5]: Leaving directory '/home/openwrt/lede/build_dir/target-aarch64_cortex-a53_musl/linux-brcm2708_bcm2710/linux-4.19.88'
Makefile:27: recipe for target 'build' failed
make[4]: *** [build] Error 2
make[4]: Leaving directory '/home/openwrt/lede/build_dir/target-aarch64_cortex-a53_musl/linux-brcm2708_bcm2710/cryptodev-linux-1.9.git-2017-10-04'
Makefile:74: recipe for target '/home/openwrt/lede/build_dir/target-aarch64_cortex-a53_musl/linux-brcm2708_bcm2710/cryptodev-linux-1.9.git-2017-10-04/.built' failed
make[3]: *** [/home/openwrt/lede/build_dir/target-aarch64_cortex-a53_musl/linux-brcm2708_bcm2710/cryptodev-linux-1.9.git-2017-10-04/.built] Error 2
make[3]: Leaving directory '/home/openwrt/lede/feeds/packages/utils/cryptodev-linux'
time: package/feeds/packages/cryptodev-linux/compile#0.96#0.56#2.04
package/Makefile:107: recipe for target 'package/feeds/packages/cryptodev-linux/compile' failed
make[2]: *** [package/feeds/packages/cryptodev-linux/compile] Error 2
make[2]: Leaving directory '/home/openwrt/lede'
package/Makefile:103: recipe for target '/home/openwrt/lede/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile' failed
make[1]: *** [/home/openwrt/lede/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/openwrt/lede'
/home/openwrt/lede/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2

yf2000plus, 2019-12-26:
最新内核和cryptodev的冲突
https://patchwork.openembedded.org/patch/151867/

closed 2019-12-30 by coolsnowwolf

whlsxl, 2020-01-12:
```
./scripts/feeds clean
./scripts/feeds update -a 
./scripts/feeds install -a
```