coolsnowwolf/lede #2577: IPSec VPN 服务器 bug反馈

bonnyandsky, 2019-12-25:
网络环境1：
       主路由刷LEDE固件，并在固件里运行v2ray和ipsec vpn服务器
              结果：外网手机连上ipsec后可以正常使用v2服务，也可以访问接在主路由器下面的群晖nas的web控制页面。

网络环境2：
       主路由不开启任何服务，只是做pptp和ipsec的端口映射。
       主路由下面接了一台群晖nas，在群晖nas里用Virtual Machine Manager虚拟安装lede并运行（当作旁路由使用），虚拟机的lede运行v2ray和ipsec vpn和pptp vpn服务。
             结果：外网使用手机通过pptp连接上去后可以使用v2服务器，可以访问lede及nas和主路由的web控制页面。但是使用ipsec连接上去后只能使用v2服务和访问lede的控制页面，局域网内其他设备的web控制页面无法打开。

PS:2个环境下的lede里面的ipsec设置相同，ipsec vpn客户端地址均和网络的ip段是同一个，vpn客户端的dns服务器均为202.96.209.6（电信的有效dns服务器）

coolsnowwolf, 2019-12-25:
不再支持lede，lede 死了很久了

bonnyandsky, 2019-12-26:
> 不再支持lede，lede 死了很久了

是说ipsec不再支持lede的意思？那可有其他替代的vpn服务器插件么？

bonnyandsky, 2019-12-26:
可能我表述不清楚，我说的lede就是你的源代码编译出来的固件。

coolsnowwolf, 2019-12-26:
ah esp 协议的穿透问题

bonnyandsky, 2019-12-26:
> ah esp 协议的穿透问题

请问能解决么？完全不懂……另外我最近想利用家里的2个tplink的路由器做无缝漫游，一个是wr710n（这个看有配置文件）另外一个是703n（没有配置文件）请问能否帮忙添加

coolsnowwolf, 2019-12-29:
需要上游的路由器支持 IPSEC VPN 透传，属于 ALG 的功能

bonnyandsky, 2019-12-29:
> 需要上游的路由器支持 IPSEC VPN 透传，属于 ALG 的功能

LEDE支持么？我稍后用家里的网络测试下，家里的主路由用的也是lede

closed 2019-12-30 by coolsnowwolf

bonnyandsky, 2020-01-03:
> 需要上游的路由器支持 IPSEC VPN 透传，属于 ALG 的功能

还是不太懂，我在上级路由中已经打开了udp500，1701，4500端口的映射功能，我也查了下你说的ipsec透传的关键字，还是不太明白应该怎么设置才可以正常使用，还请指导。谢谢！