coolsnowwolf/lede #2581: 动态DNS插件里阿里云ddns解析出错

lqhorochi, 2019-12-25:
动态DNS插件里阿里云ddns解析出错
ddns.AliDDNS.check_interval='5'
ddns.AliDDNS.domain='kennethlai.bid'
ddns.AliDDNS.enabled='1'
ddns.AliDDNS.force_interval='24'
ddns.AliDDNS.lookup_host='kennethlai.bid'
ddns.AliDDNS.password='***PW***'
ddns.AliDDNS.service_name='aliyun.com'
ddns.AliDDNS.username='LTAI3jyw'
ddns.AliDDNS=service
 140318       : verbose mode  : 0 - run normal, NO console output
 140318       : check interval: 300 seconds
 140318       : force interval: 86400 seconds
 140318       : retry interval: 60 seconds
 140318       : retry counter : 0 times
 140318       : No old process
 140318       : last update: 2019-12-25 13:55
 140318       : Detect registered/public IP
 140318       : #> /usr/bin/nslookup kennethlai.bid  >/var/run/ddns/AliDDNS.dat 2>/var/run/ddns/AliDDNS.err
 140319       : Registered IP '116.20.230.10
113.99.218.149
113.70.215.225' detected
 140319  info : Starting main loop at 2019-12-25 14:03
 140319       : Detect local IP on 'network'
 140319       : Local IP '113.99.218.14' detected on network 'wan'
 140319       : Update needed - L: '113.99.218.14' <> R: '116.20.230.10
113.99.218.149
113.70.215.225'
 140319       : parsing script '/usr/lib/ddns/update_aliyun_com.sh'
 140319       : #> /usr/bin/wget-ssl -nv -t 1 -O /var/run/ddns/AliDDNS.dat -o /var/run/ddns/AliDDNS.err --no-proxy 'http://alidns.aliyuncs.com/?Action=DescribeSubDomainRecords&SubDomain=kennethlai.bid&Format=JSON&Version=2015-01-09&AccessKeyId=LTAI3jyw&SignatureMethod=HMAC-SHA1&Timestamp=2019-12-25T06%3A03%3A19Z&SignatureVersion=1.0&SignatureNonce=3e836e71-b23d-42e0-b6c3-1c6ff9f06a5c&Signature=yZyTyY03sdnvvZQ9nI%2FvEgqVNUc%3D'
 140320       : 获取到解析记录: {"PageNumber":1,"TotalCount":3,"PageSize":20,"RequestId":"99278DD0-FBF1-4CD0-88B4-BEA2557CE79E","DomainRecords":{"Record":[{"RR":"@","Status":"ENABLE","Value":"113.70.215.225","Weight":1,"RecordId":"3599731555375104","Type":"A","DomainName":"kennethlai.bid","Locked":false,"Line":"default","TTL":600},{"RR":"@","Status":"ENABLE","Value":"116.20.230.10","Weight":1,"RecordId":"3538990390481920","Type":"A","DomainName":"kennethlai.bid","Locked":false,"Line":"default","TTL":600},{"RR":"@","Status":"ENABLE","Value":"113.99.218.149","Weight":1,"RecordId":"3256124717487104","Type":"A","DomainName":"kennethlai.bid","Locked":false,"Line":"default","TTL":600}]}}
 140321       : 获得解析记录ID: 3599731555375104, 类型: A
 140321       : 地址需要修改
 140324  note : PID '16184' terminated by 'SIGTERM' at 2019-12-25 14:03
 140327       : ************ ************** ************** **************
 140327  note : PID '22875' started at 2019-12-25 14:03
 140327       : ddns version  : 2.7.8-1
 140328       : uci configuration:
ddns.AliDDNS.check_interval='5'
ddns.AliDDNS.domain='kennethlai.bid'
ddns.AliDDNS.enabled='1'
ddns.AliDDNS.force_interval='24'
ddns.AliDDNS.lookup_host='kennethlai.bid'
ddns.AliDDNS.password='***PW***'
ddns.AliDDNS.service_name='aliyun.com'
ddns.AliDDNS.username='LTAI3jyw'
ddns.AliDDNS=service
 140328       : verbose mode  : 0 - run normal, NO console output
 140328       : check interval: 300 seconds
 140328       : force interval: 86400 seconds
 140328       : retry interval: 60 seconds
 140328       : retry counter : 0 times
 140329       : No old process
 140329       : last update: 2019-12-25 13:55
 140329       : Detect registered/public IP
 140329       : #> /usr/bin/nslookup kennethlai.bid  >/var/run/ddns/AliDDNS.dat 2>/var/run/ddns/AliDDNS.err
 140329       : Registered IP '113.70.215.225
116.20.230.10
113.99.218.149' detected
 140329  info : Starting main loop at 2019-12-25 14:03
 140329       : Detect local IP on 'network'
 140329       : Local IP '113.99.218.14' detected on network 'wan'
 140329       : Update needed - L: '113.99.218.14' <> R: '113.70.215.225
116.20.230.10
113.99.218.149'
 140329       : parsing script '/usr/lib/ddns/update_aliyun_com.sh'
 140330       : #> /usr/bin/wget-ssl -nv -t 1 -O /var/run/ddns/AliDDNS.dat -o /var/run/ddns/AliDDNS.err --no-proxy 'http://alidns.aliyuncs.com/?Action=DescribeSubDomainRecords&SubDomain=kennethlai.bid&Format=JSON&Version=2015-01-09&AccessKeyId=LTAI3jyw&SignatureMethod=HMAC-SHA1&Timestamp=2019-12-25T06%3A03%3A30Z&SignatureVersion=1.0&SignatureNonce=ffc5f345-c0d1-46e2-b5a9-987e60645f74&Signature=lKoY%2FA5ruCHqE5m7kGLs585Ejnk%3D'
 140330       : 获取到解析记录: {"PageNumber":1,"TotalCount":3,"PageSize":20,"RequestId":"2C4B9474-C2E9-44F8-80A3-0639C1F3D7D2","DomainRecords":{"Record":[{"RR":"@","Status":"ENABLE","Value":"113.70.215.225","Weight":1,"RecordId":"3599731555375104","Type":"A","DomainName":"kennethlai.bid","Locked":false,"Line":"default","TTL":600},{"RR":"@","Status":"ENABLE","Value":"116.20.230.10","Weight":1,"RecordId":"3538990390481920","Type":"A","DomainName":"kennethlai.bid","Locked":false,"Line":"default","TTL":600},{"RR":"@","Status":"ENABLE","Value":"113.99.218.149","Weight":1,"RecordId":"3256124717487104","Type":"A","DomainName":"kennethlai.bid","Locked":false,"Line":"default","TTL":600}]}}
 140332       : 获得解析记录ID: 3599731555375104, 类型: A
 140332       : 地址需要修改
[... 156 lines cut ...]
 170839       : retry counter : 0 times
 170839       : No old process
 170839       : last update: 2019-12-25 13:55
 170839       : Detect registered/public IP
 170840       : #> /usr/bin/nslookup kennethlai.bid  >/var/run/ddns/AliDDNS.dat 2>/var/run/ddns/AliDDNS.err
 170840       : Registered IP '116.20.228.117
113.99.218.149
116.20.230.10
113.70.215.225' detected
 170840  info : Starting main loop at 2019-12-25 17:08
 170840       : Detect local IP on 'network'
 170840       : Local IP '113.99.218.14' detected on network 'wan'
 170840       : Update needed - L: '113.99.218.14' <> R: '116.20.228.117
113.99.218.149
116.20.230.10
113.70.215.225'
 170840       : parsing script '/usr/lib/ddns/update_aliyun_com.sh'
 170841       : #> /usr/bin/wget-ssl -nv -t 1 -O /var/run/ddns/AliDDNS.dat -o /var/run/ddns/AliDDNS.err --no-proxy 'http://alidns.aliyuncs.com/?Action=DescribeSubDomainRecords&SubDomain=kennethlai.bid&Format=JSON&Version=2015-01-09&AccessKeyId=LTAI3jyw&SignatureMethod=HMAC-SHA1&Timestamp=2019-12-25T09%3A08%3A40Z&SignatureVersion=1.0&SignatureNonce=3fc223f7-ddcc-4cfb-9f3f-a4f5b8f0363f&Signature=np%2Bl2UoJIdeouitIEZZS%2BYVqNwg%3D'
 170841       : 获取到解析记录: {"PageNumber":1,"TotalCount":0,"PageSize":20,"RequestId":"F46C1DA6-B0EC-4864-A671-822A7CC12C2B","DomainRecords":{"Record":[]}}
 170842       : 解析记录不存在
 170845       : #> /usr/bin/wget-ssl -nv -t 1 -O /var/run/ddns/AliDDNS.dat -o /var/run/ddns/AliDDNS.err --no-proxy 'http://alidns.aliyuncs.com/?Action=AddDomainRecord&DomainName=bid&RR=kennethlai&Type=A&Value=113.99.218.14&Format=JSON&Version=2015-01-09&AccessKeyId=LTAI3jyw&SignatureMethod=HMAC-SHA1&Timestamp=2019-12-25T09%3A08%3A45Z&SignatureVersion=1.0&SignatureNonce=1104bd38-54a0-42f5-9fbb-cf1e10a30253&Signature=OVGs8SYi%2FfT5GfzFbBMWaYk%2FOw%3D%3D'
 170846 ERROR : wget 错误代码: '8'
 170846       : http://alidns.aliyuncs.com/?Action=AddDomainRecord&DomainName=bid&RR=kennethlai&Type=A&Value=113.99.218.14&Format=JSON&Version=2015-01-09&AccessKeyId=LTAI3jyw&SignatureMethod=HMAC-SHA1&Timestamp=2019-12-25T09%3A08%3A45Z&SignatureVersion=1.0&SignatureNonce=1104bd38-54a0-42f5-9fbb-cf1e10a30253&Signature=OVGs8SYi%2FfT5GfzFbBMWaYk%2FOw%3D%3D:
2019-12-25 17:08:46 ERROR 400: Bad Request.
 170846  WARN : 传输失败 - 1/0 在 60 秒后重试

junwu0608, 2019-12-26:
你还不如上个设置界面看看

closed 2019-12-30 by coolsnowwolf

hb1707, 2020-07-07:
阿里云那边要给dns的修改权限

bruceyan303, 2023-01-10:
同问