coolsnowwolf/lede #2582: gfwlist更新不是最新

GDZmax, 2019-12-25:
用ssr+更新的GFWlist是五千两百多条，但是github里的GFWlist项目上是五千七百多条

snakwu, 2019-12-25:
还有国内ip列表也无法更新，n久的问题了

vlanbin, 2019-12-26:
> 还有国内ip列表也无法更新，n久的问题了，能用就行

国内IP列表可以正常更新

hcym, 2019-12-26:
以前有一阵是git被屏蔽的，后来编译时直接加好就没用过这个更新功能，就是广告规则麻烦一点

lphgor, 2019-12-26:
因为本地更新的时候过滤了一部分

https://github.com/coolsnowwolf/lede/blob/aa71afabfa45a7afafef50c7288e174d0ba1ecad/package/lean/luci-app-ssr-plus/root/usr/bin/ssr-gfw#L6-L14

caicaicai21, 2019-12-26:
原CDN文件有63000+行
[image]
下载、处理成国内ip段文件最快也需要2分半的时间(Github Action上运行到结束所需时间)
[image]

snakwu, 2019-12-26:
更新慢和不更新其实是这两个源地址问题
http://ftp.apnic.net/apnic/stats/apnic/delegated-apnic-latest
https://cdn.jsdelivr.net/gh/gfwlist/gfwlist/gfwlist.txt
如果出错,重试三次,就会不再更新,加进了强制走代理也是如此
ps:话说不知道Lean大,会不会考虑更换源地址

closed 2019-12-30 by coolsnowwolf