coolsnowwolf/lede #2585: 关于SmartDNS的添加设置DNS设置问题

Myoko, 2019-12-25:
关于SmartDNS的添加设置DNS设置问题！我想一下 上游的DNS 如下设置的吗？对于这个部分不太明白该如何设置为好。
[image]

shantsaichan, 2019-12-26:
之前发过的：

配置smartdns上游，国内用udp，国外用tls，https。

国外dns配置group组为foreign。

使用nameserver域名分流，下载gfw list转换为smartdns格式，nameserver指定gfw list分流走国外foreign组。

对一些ip不通的域名，将请求转发到第二dns，分流使用smartdns的第二dns解析，第二dns关闭测速，就进解析ip，在配合ipset透明度请求网页

如果不想搞分流，可以用黑白名单：
那个黑名单可以配置假ip范围，白名单可以配置chn-router，然后在应用到国内dns上

Myoko, 2019-12-26:
感谢，另外问一下 RP 好像不能改本地的端口 只能用5335 想改一个 结果发现XXRP 这边仅识别5335
[image]

klever1988, 2019-12-26:
置顶帖为啥没人看？
https://github.com/coolsnowwolf/lede/issues/2551

closed 2019-12-26 by Myoko