coolsnowwolf/lede #2588: ssr

bonnyandsky, 2019-12-26:


snakwu, 2019-12-26:
.github.com

lphgor, 2019-12-26:
看自带的列表，应该直接填一级域名就好了。
https://github.com/coolsnowwolf/lede/blob/master/package/lean/luci-app-ssr-plus/root/etc/config/gfw.list

bonnyandsky, 2019-12-26:
> 看自带的列表，应该直接填一级域名就好了。
> https://github.com/coolsnowwolf/lede/blob/master/package/lean/luci-app-ssr-plus/root/etc/config/gfw.list

看了下github默认已经在列表里面了，所以应该会自动走代理喽？可是我下载编译好的固件还是很慢，开全局代理就快了。

lphgor, 2019-12-26:
> 看了下github默认已经在列表里面了，所以应该会自动走代理喽？可是我下载编译好的固件还是很慢，开全局代理就快了。

从github下载，需要添加这个域名：
`githubusercontent.com`

bonnyandsky, 2019-12-26:
> githubusercontent.com

我刚才用一个没用的节点设置了代理。然后哦用baidu做测试，结果没效果

hcym, 2019-12-26:
白名单有效，黑的没用过不知道，应该差不多吧

xinpenghan, 2019-12-26:
我一般用访问控制设置强制走代理的域名

xinpenghan, 2019-12-26:
试试这个配置
github.com
gstatic.com
avatars0.githubusercontent.com
avatars1.githubusercontent.com
avatars2.githubusercontent.com
avatars3.githubusercontent.com
githubapp.com
www.ipip.net
ipip.net
ggpht.com
googleapis.com
githubusercontent.com
googletagmanager.com
googlevideo.com
gstatic.com
ytimg.com
youtube.com
youtube-nocookie.com
youtubeeducation.com
youtu.be
yt.be

bonnyandsky, 2019-12-27:
> 试试这个配置，我正好更新一下
> github.com
> gstatic.com
> avatars0.githubusercontent.com
> avatars1.githubusercontent.com
> avatars2.githubusercontent.com
> avatars3.githubusercontent.com
> actions.githubusercontent.com
> github.githubassets.com
> api.github.com
> gist.github.com
> help.github.com
> github-cloud.s3.amazonaws.com
> user-images.githubusercontent.com
> live.github.com
> githubapp.com
> githubassets.com
> schema.org
> [www.ipip.net](http://www.ipip.net)
> ipip.net
> ggpht.com
> googleapis.com
> githubusercontent.com
> googletagmanager.com
> googlevideo.com
> gstatic.com
> ytimg.com
> youtube.com
> youtube-nocookie.com
> youtubeeducation.com
> youtu.be
> yt.be

修改在那个文件

xinpenghan, 2019-12-27:
> 修改在那个文件

在ShadowSocksR Plus+ 里找到访问控制添加强制走代理的域名，将下面这些添加进去
刚新调整了一下，gfw模式下用github应该没问题了
github.com
gstatic.com
github.githubassets.com
translate.googleapis.com
avatars0.githubusercontent.com
avatars1.githubusercontent.com
avatars2.githubusercontent.com
avatars3.githubusercontent.com
actions.githubusercontent.com
user-images.githubusercontent.com
api.github.com
gist.github.com
help.github.com
github-cloud.s3.amazonaws.com
live.github.com
githubapp.com
githubassets.com
schema.org
www.ipip.net
ipip.net
ggpht.com
googleapis.com
googletagmanager.com
googlevideo.com
gstatic.com
ytimg.com
youtube.com
youtube-nocookie.com
youtubeeducation.com
youtu.be
yt.be

ManaStuDent, 2019-12-27:
不知道有没有人遇到过添加白名单了，Chrome浏览器无效，Safari 火狐有效。

closed 2019-12-30 by coolsnowwolf

qq280948982, 2020-03-19:
我也这个问题，在Black domain list里填写了想要强制使用代理的域名，但是迟终没有走代理

maodou38, 2020-06-29:
没什么用啊 根本就  我强制走代理写了个baidu.com 还是走直连