coolsnowwolf/lede #2589: 最新代码rpi3 b+编译失败

aesirteam, 2019-12-26:
Applying /home/dev/lede/target/linux/brcm2708/patches-4.19/950-0131-firmware-raspberrypi-Notify-firmware-of-a-reboot.patch using plaintext: 
patching file drivers/firmware/raspberrypi.c
Hunk #1 FAILED at 15.
1 out of 3 hunks FAILED -- saving rejects to file drivers/firmware/raspberrypi.c.rej
Patch failed!  Please fix /home/dev/lede/target/linux/brcm2708/patches-4.19/950-0131-firmware-raspberrypi-Notify-firmware-of-a-reboot.patch!
make[4]: *** [/home/dev/lede/build_dir/target-aarch64_cortex-a53_musl/linux-brcm2708_bcm2710/linux-4.19.91/.prepared_4c92efec677d2e9b139ee3c1ecc07ca7] Error 1
make[4]: Leaving directory `/home/dev/lede/target/linux/brcm2708'
make[3]: *** [compile] Error 2
make[3]: Leaving directory `/home/dev/lede/target/linux'
time: target/linux/compile#15.15#6.27#17.73
make[2]: *** [target/linux/compile] Error 2
make[2]: Leaving directory `/home/dev/lede'
make[1]: *** [/home/dev/lede/staging_dir/target-aarch64_cortex-a53_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory `/home/dev/lede'
make: *** [world] Error 2

hcym, 2019-12-26:
是的，没看码，直接改armv8过了，用派的惨了

SharerMax, 2019-12-27:
树莓派4B 有同样的问题

xmapst, 2019-12-27:
怎么解决？？？？

kawon7317, 2019-12-27:
4B一样，怎么解决？？？？

xmapst, 2019-12-28:
@aesirteam

cxjava, 2019-12-28:
把这个`950-0131-firmware-raspberrypi-Notify-firmware-of-a-reboot.patch`删除，或者移动到其他地方就可以了。

最新的4.19.1X的`drivers/firmware`下面没有`raspberry.c`这个文件了。所以`应用patch会失败`，是导致编译失败的原因。

你这个例子，可以移动到 `/home/dev` 下面：
```
mv /home/dev/lede/target/linux/brcm2708/patches-4.19/950-0131-firmware-raspberrypi-Notify-firmware-of-a-reboot.patch /home/dev
```

kawon7317, 2019-12-30:
> 把这个`950-0131-firmware-raspberrypi-Notify-firmware-of-a-reboot.patch`删除，或者移动到其他地方就可以了。
>
> 最新的4.19.1X的`drivers/firmware`下面没有`raspberry.c`这个文件了。所以`应用patch会失败`，是导致编译失败的原因。
>
> 你这个例子，可以移动到 `/home/dev` 下面： [...]
谢谢，已经过了

closed 2019-12-30 by coolsnowwolf