coolsnowwolf/lede #2598: aria2启动有问题

MxCen, 2019-12-27:
aria2插件装好后，一直显示收集状态，不能访问WBE页面，估计是没开启服务。勾选启动，点击应用保存也一直这样，没有任何反应。

MxCen, 2019-12-27:
[luci-app-aria2.zip](https://github.com/coolsnowwolf/lede/files/4004291/luci-app-aria2.zip)
官方最新的feeds的包没有问题，烦请L大帮助更新下，谢谢！

georip, 2019-12-27:
肯定是你设置有问题。我这边非常正常。

chuds001, 2019-12-28:
我的aria2也是一直显示收集状态。 @georip 设置要注意些什么？

yatounoneko, 2019-12-29:
请你提供一下启动时的日志
另外如没有设置aria2的默认储存路径将无法启动

MxCen, 2019-12-30:
Mon Dec 30 21:07:44 2019 daemon.info aria2: Instance "main" disabled.
Mon Dec 30 21:09:43 2019 kern.err /usr/bin/ssr-redir[22992]: server recv: Connection reset by peer
Mon Dec 30 21:09:54 2019 daemon.info aria2: Aria2 will run with user 'aria2'.
Mon Dec 30 21:09:54 2019 daemon.info aria2: Please make sure user 'aria2' has write access to download dir: /tmp
Mon Dec 30 21:10:00 2019 cron.info crond[22792]: USER root pid 27178 cmd /usr/share/unblockneteasemusic/log_check.sh
Mon Dec 30 21:10:21 2019 daemon.info aria2: Aria2 will run with user 'aria2'.
Mon Dec 30 21:10:21 2019 daemon.info aria2: Please make sure user 'aria2' has write access to download dir: /tmp
Mon Dec 30 21:11:31 2019 daemon.info aria2: Aria2 will run with user 'aria2'.
Mon Dec 30 21:11:31 2019 daemon.info aria2: Please make sure user 'aria2' has write access to download dir: /mnt/sda1/aria2
Mon Dec 30 21:11:38 2019 daemon.info aria2: Aria2 will run with user 'aria2'.
Mon Dec 30 21:11:38 2019 daemon.info aria2: Please make sure user 'aria2' has write access to download dir: /mnt/sda1/aria2

yatounoneko, 2019-12-30:
请检查你的aria2的下载路径是否有错（或有没有权限）

closed 2019-12-30 by coolsnowwolf