coolsnowwolf/lede #2602: 编译错误请求支援 Cannot satisfy the following dependencies for ddns-scripts_aliyun

ziqqqq, 2019-12-27:
ollected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for ddns-scripts_aliyun:
 * 	openssl-util * 
 * opkg_install_cmd: Cannot install package ddns-scripts_aliyun.
package/Makefile:65: recipe for target 'package/install' failed
make[2]: *** [package/install] Error 255
make[2]: Leaving directory '/home/z/lede'
package/Makefile:104: recipe for target '/home/z/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_install' failed
make[1]: *** [/home/z/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/z/lede'
/home/z/lede/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2

ziqqqq, 2019-12-27:
搞好了，封贴～～～

dongfangy2011, 2019-12-28:
> 搞好了，封贴～～～

你是怎么弄的，我也是出现和你同样的问题，不小的怎么回事

closed 2019-12-29 by coolsnowwolf

zhipengyan, 2021-05-09:
安装下 openssl-util 就行了