coolsnowwolf/lede #2617: 树莓派4B 64位系统做单臂路由，adbyby plus +不能去广告

cxjava, 2019-12-28:
树莓派4B
已经安装了树莓派爱好者的64位系统
docker运行自编译的LEDE(`Powered by LuCI Master (git-19.338.43082-9e87e69) / OpenWrt R9.12.03`).

想设置树莓派LEDE为单臂路由，开启广告过滤。

主路由IP：192.168.123.1，树莓派LEDE IP为：192.168.123.100。主路由的Lan口连树莓派Lan口，树莓派没有启用wifi。
主路由中已经设置网关和dns为192.168.123.100.
树莓派LEDE中已经设置网关和dns为192.168.123.1，并且已经关闭DHCP。
主路由下面的设备网关已经指向了树莓派的地址

树莓派LEDE开启了ss之后，主路由下面的设备能访问youtube。**说明单臂路由的设置是没有问题的。**

但是树莓派LEDE开启adbyby plus +之后（已经关闭ss），主路由下面的设备并不能去广告，比如去cnbeta的广告，爱qiyi广告。

**是不是在树莓派4B 64位系统下面，树莓派4b里面的LEDE做单臂路由的情况下，adbyby并不能够去广告？**

其中参考设置：
- [在Docker 中运行 OpenWrt 旁路网关](https://mlapp.cn/376.html)
- [自编译 OpenWrt R9.11.9 固件，支持 Raspberry Pi 2B/3B/3B+/4B](https://mlapp.cn/369.html)

[image]

hcym, 2019-12-28:
没弄好，看看hanwckf博客吧，用的n1和K2p毛子，什么ip网关dnsdnx都不用，就是设置好vlan标记，一切正常，广告代理还有kms激活都正常，和普通路由一模一样

cxjava, 2019-12-28:
> 没弄好，看看hanwckf博客吧，用的n1和K2p毛子，什么ip网关dnsdnx都不用，就是设置好vlan标记，一切正常，广告代理还有kms激活都正常，和普通路由一模一样

有博客地址吗？找了一圈没有找到hanwckf的博客，我目前主路由就是刷的他的固件。

closed 2019-12-30 by coolsnowwolf